zephyrproject-rtos/zephyr #83323: Linking error while using DMA for OCTOSPI with STM32H573 discovery board (STM32H573i_dk)

HemantFargade-eaton, 2024-12-23:
**Describe the bug**

Linking error for DMA for OCTOSPI with STM32H573 Discovery board (STM32H573i_dk) 
I am trying to enable DMA for Currently supported OCTOSPI flash driver on Discovery board.
Compatible for QSPI driver = "st,stm32-ospi"
Compatible for flash nor driver = "st,stm32-ospi-nor"

**To Reproduce**

Zephyr version: 3.6.0
Added below dmas property in overlay file to activate DMA in OCTOSPI node

&octospi1 {
		pinctrl-0 = <&octospi1_io0_pb1 &octospi1_io1_pd12
					&octospi1_clk_pf10 &octospi1_ncs_pg6>;
		/* request 57 for XSPI1 */
		dmas = <&gpdma1 4 57 STM32_DMA_PERIPH_TX
		&gpdma1 5 57 STM32_DMA_PERIPH_RX>;
		dma-names = "tx", "rx";
		mx25lm51245: ospi-nor-flash@90000000 {
		spi-bus-width = <OSPI_SPI_MODE>;
		data-rate = <OSPI_STR_TRANSFER>;
	};
};

prj.conf : 

CONFIG_STDOUT_CONSOLE=y
CONFIG_FLASH=y
CONFIG_SPI=y

step to compile code: west build -b stm32h573i_dk -p

**Expected behavior**

Compliable with DMA.

**Impact**

showstopper

**Logs and console output**

Getting linking errors: 
/zephyr/drivers/flash/flash_stm32_ospi.c:1979:41: error: 'XSPI_HandleTypeDef' has no member named 'hdma'; did you mean 'hdmatx'?
 1979 |         __HAL_LINKDMA(&dev_data->hospi, hdma, hdma);
      |                                         ^~~~
 1979 |         __HAL_LINKDMA(&dev_data->hospi, hdma, hdma);

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain : zephyr-sdk-0.16.5-1
 - Commit SHA : 468eb56cf242eedba62006ee758700ee6148763f

**Additional context**

marwaiehm-st, 2024-12-26:
Hi @HemantFargade-eaton, 
Thanks for raising this issue, but please note that v3.6.0 is coming to end of life shortly (https://github.com/zephyrproject-rtos/zephyr/wiki/Release-Management). 
Would you mind testing on a more recent Zephyr version v3.7.0 for instance, which is a Long Term Support version ?

FRASTM, 2025-02-05:
That's right, the  PR https://github.com/zephyrproject-rtos/zephyr/pull/70371/  which makes the stm32h573i_dk disco kit compatible with the XSPI node (no more OSPI bindings)    was introduced after the V 3.6.0 

Could you use the V 3.7.0 LTS instead ?

else (but no preferred)  the backport of the PR #70371 is still possible with an stm32h573i_dk overlay as follows:
```
&xspi1 {
	pinctrl-0 = <&octospi1_io0_pb1 &octospi1_io1_pd12
		&octospi1_clk_pf10 &octospi1_ncs_pg6>;
	/* request 57 for XSPI1 */
	dmas = <&gpdma1 4 57 STM32_DMA_PERIPH_TX
	&gpdma1 5 57 STM32_DMA_PERIPH_RX>;
	dma-names = "tx", "rx";

	mx25lm51245: ospi-nor-flash@90000000 {
		spi-bus-width = <XSPI_SPI_MODE>;
		data-rate = <XSPI_STR_TRANSFER>;
	};
};
```

closed 2025-04-21 by github-actions[bot]