zephyrproject-rtos/zephyr #83333: imx8mp-pinctrl handles drive-strength and slew-rate incorrectly

voltamund, 2024-12-23:
**Describe the bug**

Target platform: MIMX8MM6 Cortex M4
I think with the changes in https://github.com/zephyrproject-rtos/zephyr/pull/75083 `drive-strength` and `slew-rate` are not correctly handled anymore.

`drive-strength = "x6"` should result in `11x` but it actually results in `011`. Something similar also happens for `slew-rate`. Before this change the enum values were also incorrect (not according to the nxp reference manual) but somehow resulted in the correct register values, i.e. `40-ohm` resulted in `11x`.

**Expected behavior**

Device tree values for `drive-strength` and `slew-rate` result in correct register values.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.8
 - Zephyr Version 3.7.0

mmahadevan108, 2025-02-18:
cc @JiafeiPan , can you help take a look at this issue.

JiafeiPan, 2025-02-25:
Created PR: https://github.com/zephyrproject-rtos/zephyr/pull/86273 to fix this issue.

mmahadevan108, 2025-02-26:
@voltamund , can you check the PR fix.

closed 2025-02-26 by fabiobaltieri