zephyrproject-rtos/zephyr #83334: Why does sys_csrand_get hang in UART IRQ handler?

zhang-wenchao, 2024-12-23:
```
void cs_random(void *dst, size_t len)
{
	sys_csrand_get(dst, len);
}

void app_uart_rx_handler(const struct device *dev, void *user_data)
{
	uint8_t fifo_buf[FIFO_SIZE];
	int rx_len;

	if (uart_irq_update(dev) && uart_irq_rx_ready(dev)) {
		rx_len = uart_fifo_read(dev, fifo_buf, sizeof(fifo_buf));
		if (rx_len > 0) {
			// Use cs_random hang
		}
	}
}

int app_uart_irq_register()
{
	if (!device_is_ready(DEV_CONSOLE)) {
		return -1;
	}
	uart_irq_callback_user_data_set(DEV_CONSOLE, app_uart_rx_handler, NULL);
	uart_irq_rx_enable(DEV_CONSOLE);

	return 0;
}
```

zhang-wenchao, 2024-12-23:
https://github.com/zephyrproject-rtos/zephyr/pull/17736

I saw a discussion about irq and I waited for 10 minutes and it didn't return. I think it's wrong.

nordicjm, 2024-12-23:
Probably because it's using a device and needs to lock IRQs which it cannot do, you should use a work handler, all IRQs should be as minimal, small and fast as possible and definitely not calling a slow operation like generating cryptographically secure random data

zhang-wenchao, 2024-12-23:
I will close this question, I create a buffer of random numbers at startup.

closed 2024-12-23 by zhang-wenchao