zephyrproject-rtos/zephyr #83373: openocd-esp32 v0.12 does not work with `west debug`

flisboac, 2024-12-25:
**Describe the bug**

The most recent releases of OpenOCD-ESP32 include a "v" in their version identifiers, when issuing an `openocd --version`.

The openocd implementation of `west debug`, etc., verifies the OpenOCD version before proceeding, but neither does it verify that a match has happened, nor does it contemplate the very common practice of adding a `v` prefix to a version number.

The affected code is here:

https://github.com/zephyrproject-rtos/zephyr/blob/dd37c60d5a4cc4beaf4adb577c4d0b7c775d642f/scripts/west_commands/runners/openocd.py#L225

```py
version_match = re.search(r"Open On-Chip Debugger (\d+.\d+.\d+)", out)
version = version_match.group(1).split('.')
```

**To Reproduce**

1. Install OpenOCD-ESP32 somewhere in your system.
2. Execute (for example): `west build -b esp32_devkitc_wroom/esp32/procpu /workspace/app -p --build-dir /workspace/app/build/esp32_devkitc_wroom_procpu --sysbuild -- -DOPENOCD=/opt/openocd-esp32/bin/openocd -DOPENOCD_DEFAULT_PATH=/opt/openocd-esp32/share/openocd/scripts`
3. Execute: `cd /workspace/app/build/esp32_devkitc_wroom_procpu && west debugserver` or equivalent

**Expected behavior**

Should work.

**Impact**

Cannot debug ESP32 boards with OpenOCD.

**Workarounds**

Right now, I can just edit `openocd.py` to contemplate the `v` prefix (optionally), like so:

```py
version_match = re.search(r"Open On-Chip Debugger [vV]?(\d+.\d+.\d+)", out)
version = version_match.group(1).split('.')
```

Note the `[vV]?`, perhaps it could be the a final fix for this problem. The quantifier `?` guarantees it won't affect other OpenOCD builds that don't include the `v` version prefix.

Another option is to just create a script to intercept calls to `openocd`, and replaces `openocd --version` results with a text that omits the `v` prefix, e.g.:

```sh
#!/bin/sh

OPENOCD="${OPENOCD_ESP32:-"${OPENOCD:-"/opt/openocd-esp32/bin/openocd"}"}"

if [ "$1" = "--version" ]; then
  STR="$("${OPENOCD}" --version)"
  EXITCODE="$?"
  if [ "$EXITCODE" -eq 0 ]; then
    printf '%s' "${STR}" | sed 's/^Open On-Chip Debugger v/Open On-Chip Debugger /'
  fi
  exit "${EXITCODE}"
fi

exec "${OPENOCD}" "$@"
```

**Logs and console output**
```shell
$ west debugserver
-- west debugserver: rebuilding
[0/6] Performing build step for 'app'
[6/6] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
           FLASH:      438960 B    4194240 B     10.47%
     iram0_0_seg:        101 KB       224 KB     45.09%
     dram0_0_seg:       47588 B     140452 B     33.88%
  dram0_shm0_seg:          0 GB         2 KB      0.00%
  dram0_sem0_seg:          0 GB          8 B      0.00%
     dram0_1_seg:          0 GB         0 GB
     irom0_0_seg:      242416 B      4092 KB      5.79%
     drom0_0_seg:         64 KB      4092 KB      1.56%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from /workspace/app/build/esp32_devkitc_wroom_procpu/app/zephyr/zephyr.elf for board: esp32_devkitc_wroom
[1/6] Performing build step for 'mcuboot'
ninja: no work to do.
[6/6] Completed 'mcuboot'
-- west debugserver: using runner openocd
Traceback (most recent call last):
  File "/usr/local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/west/app/main.py", line 1176, in main
    app.run(argv or sys.argv[1:])
  File "/usr/local/lib/python3.10/dist-packages/west/app/main.py", line 255, in run
    self.run_command(argv, early_args)
  File "/usr/local/lib/python3.10/dist-packages/west/app/main.py", line 561, in run_command
    self.run_extension(args.command, argv)
  File "/usr/local/lib/python3.10/dist-packages/west/app/main.py", line 716, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/usr/local/lib/python3.10/dist-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/workspace/ext/zephyr/scripts/west_commands/debug.py", line 58, in do_run
    do_run_common(self, my_args, runner_args)
  File "/workspace/ext/zephyr/scripts/west_commands/run_common.py", line 314, in do_run_common
    prev_runner = do_run_common_image(command, user_args, user_runner_args, used_cmds,
  File "/workspace/ext/zephyr/scripts/west_commands/run_common.py", line 489, in do_run_common_image
    runner.run(command_name)
  File "/workspace/ext/zephyr/scripts/west_commands/runners/core.py", line 715, in run
    self.do_run(command, **kwargs)
  File "/workspace/ext/zephyr/scripts/west_commands/runners/openocd.py", line 256, in do_run
    self.do_debugserver(**kwargs)
  File "/workspace/ext/zephyr/scripts/west_commands/runners/openocd.py", line 389, in do_debugserver
    if self.thread_info_enabled and self.supports_thread_info():
  File "/workspace/ext/zephyr/scripts/west_commands/runners/openocd.py", line 230, in supports_thread_info
    (major, minor, rev) = self.read_version()
  File "/workspace/ext/zephyr/scripts/west_commands/runners/openocd.py", line 224, in read_version
    version = version_match.group(1).split('.')
AttributeError: 'NoneType' object has no attribute 'group'
```

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04 (Linux 6.12.6, x86_64)
 - Toolchain: Zephyr SDK 0.16.8
 - Zephyr Release: v4.0.0 (commit: 8469084dfae85f854555f0607f2c838dad097235)
 
**Additional context**

Target: esp32_devkitc_wroom/esp32/procpu

rruuaanng, 2024-12-25:
(Maybe we should move that issue to west repo.) @kartben

kartben, 2024-12-25:
> (Maybe we should move that issue to west repo.) @kartben 

Why? west commands live in zephyr proper

pdgendt, 2024-12-25:
I think modifying the regex to have an optional v is a proper solution, feel free to open a PR.

fabiobaltieri, 2025-01-07:
@flisboac could you open a PR with the fix? the contribution guideline is here: https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines

EricNRS, 2025-01-07:
I ran into this last night, I will upload a fix in a few minutes.

EricNRS, 2025-01-07:
@flisboac - just read the thread and I see that you proposed doing `[vV]?` whereas my submitted patch that I did before I saw this thread just uses `v?`.  Have you a seen a capital `V` used or did you just add that to be safe?

flisboac, 2025-01-08:
> @flisboac - just read the thread and I see that you proposed doing `[vV]?` whereas my submitted patch that I did before I saw this thread just uses `v?`. Have you a seen a capital `V` used or did you just add that to be safe?

Just to be safe, essentially. The fork already doesn't follow its upstream, so I suppose it's better to consider the possibility of having a `V` as well.

PR #83667 (2025-01-10): west: runners: add option 'v' prefix to OpenOCD version string
The Espressif branch of OpenOCD uses the version string "v0.12.0" instead of "0.12.0" causing the version regex match to fail.

Add an optional 'v' prefix to avoid the failure.

Fixes: #83373
commit 3d42a1cf (2025-01-10): west: runners: add option 'v' prefix to OpenOCD version string

The Espressif branch of OpenOCD uses the version string "v0.12.0" instead
of "0.12.0" causing the version regex match to fail.

Add an optional 'v' prefix to avoid the failure.

Fixes: #83373
closed 2025-01-10 by kartben