zephyrproject-rtos/zephyr #83387: bt_id_create() address is written to flash

xudongzheng, 2024-12-26:
**Describe the bug**

According to `bt_id_create()` documentation:

> The identity will also not be written to flash in case a pre-defined address is provided, since in such a situation the app clearly has some place it got the address from and will be able to repeat the procedure on every power cycle, i.e. it would be redundant to also store the information in flash.

If I call `bt_id_create()` with a hard-coded random static address, it seems to be written to flash. This is confirmed by resetting the device, after which the hard-coded address is already there.

Following the `bt_id_create()` to https://github.com/zephyrproject-rtos/zephyr/blob/535c40d6d69caa302da3cc7dba4ed255c34aea94/subsys/bluetooth/host/id.c#L1328

It doesn't seem like there's any logic to skip writing to flash if the address is specified explicitly.

Assuming I'm interpreting the documentation correctly, it's not clear whether this is a case of inaccurate documentation or missing functionality.

**To Reproduce**

The following code is used to create a second identity (index 1) after BT is initialized. Project has `CONFIG_BT_ID_MAX=2`.

```
bt_addr_le_t addr = {
	.type = BT_ADDR_LE_RANDOM,
	.a.val = {0, 1, 2, 3, 4, 5},
};
BT_ADDR_SET_STATIC(&addr.a);
int profile = bt_id_create(&addr, NULL);
if (profile < 0) {
	LOG_ERR("failed to create profile (err %d)", profile);
	return profile;
}
LOG_INF("default profile %d created", profile);
```

**Expected behavior**

I expect the second identity slot to be vacant.

**Impact**

I have to call `bt_id_get()` and manually delete the second identity first.

xudongzheng, 2024-12-26:
Upon further investigation, it seems like the identities stored in settings are ignored if I call `bt_id_create()` prior to `bt_enable()` (essentially setting `BT_ID_DEFAULT` explicitly). This allows me to subsequently create identity 1 with `bt_id_create()` without deleting anything.

However even though the `bt/id` settings isn't read when I do this, it's still updated each time I make a change.

jhedberg, 2025-01-18:
@xudongzheng it's a bit unclear to me what kind of change (if anything) you'd like to see in the implementation. Perhaps best would be if you could post a pull request with what you think the code should instead be doing?

closed 2025-04-03 by github-actions[bot]