zephyrproject-rtos/zephyr #83392: STM32 USB CDC ACM: Plug/Unplug and Baudrate Issues with usbd_next_prj.conf

marwaiehm-st, 2024-12-26:
Here the result of [PR](https://github.com/zephyrproject-rtos/zephyr/pull/80820) tests using samples/subsys/usb/cdc_acm with -DCONF_FILE=usbd_next_prj.conf and sent a lorem ipsum buffer of 1024 bytes several times.

In any case it fixes the assert issue reported here: https://github.com/zephyrproject-rtos/zephyr/issues/61464.

[image]

**Patch**
https://github.com/zephyrproject-rtos/zephyr/issues/76250 subsys/usb/device_next/class/usbd_cdc_acm.c
```c
@@ -458,11 +458,14 @@ static int usbd_cdc_acm_ctd(struct usbd_class_data *const c_data,
 		if (setup->wLength != len) {
 			errno = -ENOTSUP;
 			return 0;
 		}

-		memcpy(&data->line_coding, buf->data, len);
+		if (buf->data) {
+			printk("SET_LINE_CODING buf->data not null\n");
+			memcpy(&data->line_coding, buf->data, len);
+		}
 		cdc_acm_update_uart_cfg(data);
 		usbd_msg_pub_device(uds_ctx, USBD_MSG_CDC_ACM_LINE_CODING, dev);
 		return 0;

 	case SET_CONTROL_LINE_STATE:
``` 

dts updated
```shell
+
+zephyr_udc0: &usb {
+	pinctrl-0 = <&usb_dm_pa11 &usb_dp_pa12>;
+	pinctrl-names = "default";
+	status = "okay";
+};
``` 

- **nucleo_f303re**

Wrong Baudrate value reported: <inf> cdc_acm_echo: Baudrate 536885760.
With [patch](https://github.com/zephyrproject-rtos/zephyr/pull/80820#workaround-1) value is fixed (115200).
Seems same issue than https://github.com/zephyrproject-rtos/zephyr/issues/76250, no crash because the buf->data has a value but not a correct one.
Old stack also report a fixed baudrate.
[Assertion](https://github.com/zephyrproject-rtos/zephyr/pull/80820#assert-enabled) if CONFIG_ASSERT=y
[Plug/unplug](https://github.com/zephyrproject-rtos/zephyr/pull/80820#plugunplug) issue

- **Assert enabled**

log
```c
*** Booting Zephyr OS build v4.0.0-1636-gddeb51a89cfd ***
[00:00:00.001,000] <err> usbd_core: Failed to set default address after bus reset:
[00:00:00.001,000] <inf> cdc_acm_echo: USBD message: Bus reset:
[00:00:00.001,000] <inf> cdc_acm_echo: USBD message: Stack error:
[00:00:00.001,000] <inf> cdc_acm_echo: USB device support enabled:
[00:00:00.001,000] <inf> cdc_acm_echo: Wait for DTR:
[00:00:00.004,000] <inf> cdc_acm_echo: USBD message: Device suspended:
[00:00:00.112,000] <inf> cdc_acm_echo: USBD message: Device resumed:
[00:00:00.167,000] <inf> cdc_acm_echo: USBD message: Bus reset:
[00:00:00.281,000] <inf> cdc_acm_echo: USBD message: Bus reset:
[00:00:00.362,000] <inf> cdc_acm_echo: USBD message: New device configuration:
ASSERTION FAIL [net_buf_simple_tailroom(buf) >= len] @ WEST_TOPDIR/zephyr/lib/net_buf/buf_simple.c:62
[00:00:00.363,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000003e  r2/a3:  0x00000005:
[00:00:00.363,000] <err> os: r3/a4:  0x00000004 r12/ip:  0x20003160 r14/lr:  0x08003b5d:
[00:00:00.363,000] <err> os:  xpsr:  0x01000000:
[00:00:00.363,000] <err> os: Faulting instruction address (r15/pc): 0x0800d228:
[00:00:00.363,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0:
[00:00:00.363,000] <err> os: Current thread: 0x20000f60 (unknown):
[00:00:00.444,000] <err> os: Halting system:
``` 

- **Plug/unplug**

Important

On linux, second plug produce this issue (re-enumeration):
On Windows, first plug (enumeration), means CDC port never created.

log
```c
[00:00:07.187,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.187,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.187,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.188,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.188,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.188,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.188,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.188,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.188,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.188,000] <err> udc: Failed to allocate net_buf 255
[00:00:07.188,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.188,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.188,000] <err> udc: Failed to allocate net_buf 255
[00:00:07.188,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.188,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.189,000] <err> udc: Failed to allocate net_buf 255
[00:00:07.189,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.189,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.189,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.189,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.189,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.189,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.189,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.189,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.190,000] <err> udc: Failed to allocate net_buf 2
[00:00:07.190,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.190,000] <err> usbd_ch9: Malformed setup packet
[00:00:07.193,000] <err> udc: Failed to allocate net_buf 0
[00:00:07.193,000] <err> usbd_ch9: Buffer for data|status is missing
[00:00:07.193,000] <err> usbd_ch9: Malformed setup packet
```

marwaiehm-st, 2024-12-31:
I reproduced the problem with the STM32H573I-DK. 
I tried to investigate, but it is still not clear why the buf->data is always null. However, when I use the proposed workaround and set the value of CONFIG_UDC_BUF_COUNT to the maximum (256), I no longer experience the plug/unplug issue on Linux  (I haven't tested it on Windows).
Add this line to the file "prj.conf" of the CDC ACM Sample:
```c
CONFIG_UDC_BUF_COUNT=256
```

ndrs-pst, 2025-01-16:
@marwaiehm-st, this seems to be a workaround for now.
My assumption is that one of the `k_fifo_put` calls from `udc_stm32.c` was not handled by `udc_common.c`, or it shouldn't have been placed in the first place (since it will not be handled).

```
static int usbd_ctrl_feed_dout(const struct device *dev, const size_t length)
{
	struct udc_stm32_data *priv = udc_get_private(dev);
	struct udc_ep_config *cfg = udc_get_ep_cfg(dev, USB_CONTROL_EP_OUT);
	struct net_buf *buf;

	buf = k_fifo_get(&cfg->fifo, K_NO_WAIT);
	if (buf != NULL) {
		net_buf_destroy(buf);
	}

	buf = udc_ctrl_alloc(dev, USB_CONTROL_EP_OUT, length);
	if (buf == NULL) {
		return -ENOMEM;
	}

	k_fifo_put(&cfg->fifo, buf);

	HAL_PCD_EP_Receive(&priv->pcd, cfg->addr, buf->data, buf->size);

	return 0;
}
```

marwaiehm-st, 2025-01-22:
Thank you @ndrs-pst 
I tested the modifications on the nucleo_wb55rg and stm32h573i_dk boards, and they work well.

ndrs-pst, 2025-01-23:
Another proposal: Since the STM32H562 peripheral definition is `USB_DRD_FS`, it would be more appropriate to distinguish this using `DT_DRV_COMPAT`. WDYT?

```
	if (ep == USB_CONTROL_EP_IN && len > 0) {
		/* Wait for an empty package from the host.
		 * This also flushes the TX FIFO to the host.
		 */
		if (DT_HAS_COMPAT_STATUS_OKAY(st_stm32_usb)) {
			udc_stm32_flush_tx_fifo(dev);
		} else {
			usbd_ctrl_feed_dout(dev, 0);
		}
	}
```

marwaiehm-st, 2025-01-23:
Thank you for the suggestion. I agree that using DT_DRV_COMPAT to distinguish the peripheral definition for the STM32H562 as USB_DRD_FS is a good approach.This change will enhance the readability and maintainability of the code.

erwango, 2025-02-03:
@marwaiehm-st Can we close this point ?

marwaiehm-st, 2025-02-03:
@Erwan, yes we can close it. The issue resolved https://github.com/zephyrproject-rtos/zephyr/commit/8983a610a39a553009c69f8be1a4eea8db434d1f.

closed 2025-02-03 by erwango