zephyrproject-rtos/zephyr #83396: update zephyr/lvgl branch in accordance with lvgl/lvgl

glory-man, 2024-12-26:
Current `west.yml` uses revision 2b498e6f36d6b82ae1da12c8b7742e318624ecf5 of `zephyr/lvgl` branch, which is merged with v8.4 branch of parent repo `lvgl/lvgl`. But parent v8.4 branch then included some important commits - from May to December. Is it possible to merge last commits into current zephyr/lvgl branch?

faxe1008, 2025-01-05:
Hey @glory-man,
9.2 was just merged in the module master. From now on the module repo will track the upstream master. I have no plans for updating the `zephyr` branch to a 8.4.x version. If you want to have some of the patches applied maybe you want to look into @pdgendt excellent latest addition of the west patch subcommand.

If the reason you need those patches still apply towards 9.2 feel free to add them to the description :^) so we can work towards resolving them upstream.

pdgendt, 2025-01-05:
> If you want to have some of the patches applied maybe you want to look into @pdgendt excellent latest addition of the west patch subcommand.

Credits go to @cfriedt

faxe1008, 2025-03-12:
Closing this one as 8.X most likely will not see another release.

closed 2025-03-12 by faxe1008