zephyrproject-rtos/zephyr #83417: retention: retention_api structure is not used

rkel, 2024-12-27:
**Is your enhancement proposal related to a problem? Please describe.**
Retention subsystem uses driver-like construct with API. I was trying to add second layer of retention system using double retention devices below - the reason for that is to be able to recover even if WDT hit us during CRC calculation. I wanted just to use retention_api and define my own subsystem and was expecting the retention_write to use retention_api->write. But it is not the case. The retention functions assumes that retention device is always of the type defined in retention.c file. And it never uses the defined API structure.

**Describe the solution you'd like**
The retention functions in the retention.h header should just call api - like in any other driver.

**Describe alternatives you've considered**
Alternatively I can define my own driver type - but then I am losing the consistency.
In such a case, not to mislead the user, there is totally no sense in defining  retention_api structure inside retention.c file. We can spare few bytes of unused memory.

**Additional context**
If the driver / subsystem defines API structure I am expecting it to be used and the driver implementation can be alternated.

nordicjm, 2024-12-27:
Retention is built on top of retained memory, it is a software expansion of the hardware driver retained memory, it is not designed nor will it be updated to support running retention on top of retention, if you need 2 retention areas then they can be defined separately

nordicjm, 2024-12-27:
And I'm not really sure about the title here, the struct is used as can be seen in `subsys/retention/bootmode.c` and `subsys/retention/blinfo_mcuboot.c`, it's even used in a file in MCUboot