zephyrproject-rtos/zephyr #83448: compiler warning when i3c device and CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y

mertekren, 2024-12-30:
**Describe the bug**

Following compiler warning occurs when I3C device with 9 byte of reg address to dts, and DCONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y.
_C:/repos/ADI-zephyr/zephyr/arch/arm/core/mpu/arm_mpu.c: In function 'z_arm_mpu_init':_
_C:/repos/ADI-zephyr/zephyr/arch/arm/core/mpu/arm_mpu.c:481:8: warning: integer constant is too large for its type_
  _481 |         (!DT_NULL_PAGE_DETECT_NODE_EXIST)_

- I3C device requires 9 byte of address according to i3c-device.yaml 
https://git.byronlathi.com/bslathi19/zephyr/-/blob/main/dts/bindings/i3c/i3c-device.yaml?ref_type=heads

**To Reproduce**

1. Build a board with M33 with DCONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y with a tests/*:
example:
west build -b mimxrt685_evk/mimxrt685s/cm33 .\tests\drivers\uart\uart_basic_api\ -p always -DCONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y
2. See the compiler warning:
_warning: integer constant is too large for its type
  481 |         (!DT_NULL_PAGE_DETECT_NODE_EXIST)_
  
**Expected behavior**

Compiler warning shall not occur.
**Impact**

CI/CD requires -Werror option, thus this fails CI/CD procedure in out-of-tree zephyr pipeline.
**Logs and console output**

**Environment (please complete the following information):**
 - Zephyr 4.00+ (macro DT_NULL_PAGE_DETECT_NODE_EXIST added [here](https://github.com/zephyrproject-rtos/zephyr/commit/f2e115cca374516cb9e2a5845d1e3b77b53a18c9#diff-c7c1d65647254c45a9882753a577c36e14758fd0fd77f689504cea5cc03d4a65R44) )
 - OS: all
 - Toolchain Zephyr SDK
 - Commit SHA or Version used

wearyzen, 2025-01-30:
There are 2 issues being discussed here:

----------------------------------------------------------------
#### 1st is the one reported by @mertekren which is this:

```
/zephyrproject/zephyr/arch/arm/core/mpu/arm_mpu.c:481:8: warning: integer constant is too large for its type
  481 |         (!DT_NULL_PAGE_DETECT_NODE_EXIST)
      | ^       ~~~~~~~~~~~~~~~~~~~~~
```

You see this warning because of the address of wm8904 in devicetree is too large:
```
	audio_codec: wm8904@1a0000000000000000 {
		compatible = "wolfson,wm8904";
		reg = <0x1a 0 0>;
                ...
	};

```
which results in this:
`#define DT_N_S_soc_S_peripheral_50000000_S_i3c_36000_S_wm8904_1a0000000000000000_REG_IDX_0_VAL_ADDRESS 479615345916448342016 /* 0x1a0000000000000000 */`

Since the `*nodeid_IDX_0_VAL_ADDRESS` has a large value, the `DT_FOREACH_STATUS_OKAY_NODE_VARGS()` used in `DT_NULL_PAGE_DETECT_NODE_EXIST` throws the compiler warning.

Initially, this didn't look like a DT macro issue to me 'coz `DT_REG_ADDR()` inside any `DT_FOREACH_*()` macro will throw the same compiler warning for this large value if used in any other code not just the one used in discussion. But then I found that similar the concern was raised, and the answer [here](https://github.com/zephyrproject-rtos/zephyr/pull/78445#issuecomment-2356320938) was that the large address is correct. So, I would like to kindly get feedback from @decsny and @rettichschnidi on how to address this. Is this an issue with the DT tooling?

---------------------------------------------------------------
##### The 2nd issue that @dcpleung mentioned is w.r.t. below:

```
/zephyrproject/zephyr/arch/arm/core/mpu/arm_mpu.c: In function 'z_arm_mpu_init':
/zephyrproject/zephyr/arch/arm/core/mpu/arm_mpu.c:483:9: note: '#pragma message: Null-Pointer exception detection cannot be configured on un-mapped flash areas'
  483 | #pragma message "Null-Pointer exception detection cannot be configured on un-mapped flash areas"
      |         ^~~~~~~

```

[This ](https://github.com/zephyrproject-rtos/zephyr/commit/f2e115cca374516cb9e2a5845d1e3b77b53a18c9) commit provided a solution to fix this. If the board wants to have mpu config for null point dereference then they can define a node as mentioned in the commit.
@mertekren, I would suggest creating another issue for this 2nd one and assigning it to the respective board owner.

mbolivar, 2025-01-31:
@wearyzen if the address is correct, and the relevant `#address-cells` property is set correctly, then I'm not sure what the problem with the DT tooling could be. Are you sure the `reg` property in the devicetree is being used correctly in the context in the code where it appears?

Please see see "2.3.5 #address-cells and #size-cells" in the DT spec for more info on the bit allocation.

I'm confused because I only see one binding: https://docs.zephyrproject.org/latest/build/dts/api/bindings/audio/wolfson%2Cwm8904.html#dtbinding-wolfson-wm8904

and that's not for an I3C device, but an I2C device. The `reg` property should therefore only have 1 cell according to the standard I2C device binding

decsny, 2025-01-31:
> and that's not for an I3C device, but an I2C device. The `reg` property should therefore only have 1 cell according to the standard I2C device binding

well, i2c devices can be used with i3c controllers, so it can be on an i2c bus in DT but a child of an i3c controller node, which has #address-cells = <3>, that's what's going on here

wearyzen, 2025-01-31:
@mbolivar, I am not familiar with the audio code using the `reg` property. I used DT_REG_ADDR inside DT_FOREACH_STATUS_OKAY_NODE_VARGS in a completely different `MPU` related code and was reported of this warning here.
[This](https://github.com/zephyrproject-rtos/zephyr/pull/78445) PR has some details why they made the address change.

As I mentioned any other code using DT_REG_ADDR inside DT_FOREACH_STATUS_OKAY_NODE_VARGS should get this warning for these **2** boards but for my specific case I think I found a workaround with [this PR](https://github.com/zephyrproject-rtos/zephyr/pull/84997). Would be really helpful to know your thoughts since you and @decsny have more experience with the Zephyr macro magic than I do 😃.

decsny, 2025-01-31:
> Would be really helpful to know your thoughts since you and [@decsny](https://github.com/decsny) have more experience with the Zephyr macro magic than I do 😃.

I commented on the PR, I think you had a better idea than I did above, it's much more direct

mbolivar, 2025-01-31:
> well, i2c devices can be used with i3c controllers, so it can be on an i2c bus in DT but a child of an i3c controller node, which has #address-cells = <3>, that's what's going on here

PR #84997 (2025-02-06): arch: arm: fix mpu compiler warnings
What is the change?
 - Use a macro to search for "zephyr, memory-attr" in only those nodes that have the property "zephyr, memory-regions".
 - Fixes #83448.

Why is this needed?
 - Using `DT_REG_ADDR` inside `DT_FOREACH_STATUS_OKAY_NODE_VARGS` can lead to compiler warnings if a devicetree node has an addresses larger than 64bits. An example of such node is wm8904 i3c with 92 bit encoding addresses `audio_codec: wm8904@1a0000000000000000`. We can avoid this warning since this is not applicable for nodes that have the attribute "zephyr,memory-regions".
commit ce60dd2a (2025-02-06): arch: arm: fix mpu compiler warnings

What is the change?
 - Use a macro that search for "zephyr, memory-attr" in only
   those nodes that have the property "zephyr, memory-regions".
 - Fixes #83448.

Why is this needed?
 - Using `DT_REG_ADDR` inside `DT_FOREACH_STATUS_OKAY_NODE_VARGS` can
   lead to compiler warnings if a devicetree node has an addresses
   larger than 64bits. An example of such node is wm8904 i3c with 92
   bit encoding addresses `audio_codec: wm8904@1a0000000000000000`.
   We can avoid this warning since this is not applicable for nodes
   that have the attribute "zephyr,me [... cut]
closed 2025-02-06 by kartben