zephyrproject-rtos/zephyr #83453: Lorawan semtech radio init: Conditional jump or move depends on uninitialised value(s)

aescolar, 2024-12-30:
**Describe the bug**
During the lorawan driver init, there is multiple if whose condition depends on un-initialized variables.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. `cmake -GNinja -DBOARD=native_sim ../tests/subsys/lorawan/frag_decoder/ -DCONFIG_LORAWAN_FRAG_TRANSPORT_DECODER_SEMTECH=y`
3. ninja
4. valgrind zephyr/zephyr.exe 
5. See valgrind erros

**Expected behavior**
No valgrind errors. No uninitialized variables used in `ifs`.

**Impact**
Unknown

**Logs and console output**
8 warnings like (in different radio.c lines)
```
====    at 0x8052207: RadioAddRegisterToRetentionList (radio.c:1126)
====    by 0x80522D6: RadioInit (radio.c:533)
====    by 0x804EE9E: sx12xx_init (sx12xx_common.c:392)
====    by 0x804F258: sx126x_lora_init (sx126x.c:457)
====    by 0x8055A3D: do_device_init (init.c:311)
``` 

**Environment (please complete the following information):**

 - OS: Ubuntu Linux 24.04
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 14.2
 - Commit SHA or Version used: today's main da91dbdd528194c68e7bfea0029541f4335ed9b6

**Additional context**

JordanYates, 2024-12-31:
My understanding is that this is due to the `native_sim` SPI implementation essentially being a no OP. This does not affect actual hardware, nor these tests since they are testing higher level APIs.
[image]
This could be resolved by initialising the `buffer` variable in `RadioAddRegisterToRetentionList`, but this is in `loramac-node`. The upstream repo is in maintenance mode and this is not really a bug with the code, just a consequence of us not having an actual SPI driver in init.

aescolar, 2025-01-02:
thanks @JordanYates . I close the bug then as a no issue.

closed 2025-01-02 by aescolar