zephyrproject-rtos/zephyr #83465: Too small heap size cause the program to crash

Rrooach, 2024-12-31:
**Describe the bug**

Hi, when I use Zephyr to conduct operations below:
```c
int main(void)
{
	printf("Hello World! %s\n", CONFIG_BOARD_TARGET);
        mbox_msg_receive_and_append_to_queue (0x2, 0x0, 0x8a9);
	printf("finish \n");
	return 0;
}

void mbox_msg_receive_and_append_to_queue(int heap_size, int msg_size, int queue_value) {
    uint8_t heap_mem[heap_size];
    struct k_heap my_heap;
    struct k_mbox_msg msg;

    // Initialize the heap with the allocated memory
    k_heap_init(&my_heap, heap_mem, heap_size);

    // Allocate memory for the received message
    msg.tx_data = k_heap_alloc(&my_heap, msg_size, K_NO_WAIT);

}
```
I noticed if the heap size is too small, the program might crash at k_heap_alloc. But if I set the heap size to 0x200, the program can run without error.

I am not sure if there is a bug or not since the parameters are correct and valid but, indeed, too small.

**To Reproduce**
modify the samples/hello_world
```bash
west build -p always -b qemu_x86 samples/hello_world
qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae -machine q35 -kernel zephyr.elf -nographic
```

and the program keep restart
 

**Expected behavior**
The program output finish

**Logs and console output**

The following are gdb outputs:
```bash
996	void mbox_msg_receive_and_append_to_queue(int heap_size, int msg_size, int queue_value) {
(gdb) n
997	    uint8_t heap_mem[heap_size];
(gdb) n
1002	    k_heap_init(&my_heap, heap_mem, heap_size);
(gdb) n
1005	    msg.tx_data = k_heap_alloc(&my_heap, msg_size, K_NO_WAIT);
(gdb) s
0x00104b7b in k_heap_alloc (heap=0x11ffc4 <z_main_stack+860>, bytes=0, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/kheap.c:112
112		void *ret = k_heap_aligned_alloc(heap, sizeof(void *), bytes, timeout);
(gdb) s
0x00104a9b in k_heap_aligned_alloc (heap=0x11ffc4 <z_main_stack+860>, align=4, bytes=0, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/kheap.c:67
67	{
(gdb) s
sys_timepoint_calc (timeout=...) at /root/zephyr/zephyrproject/zephyr/kernel/timeout.c:297
297		if (K_TIMEOUT_EQ(timeout, K_FOREVER)) {
(gdb) s
sys_timepoint_calc (timeout=...) at /root/zephyr/zephyrproject/zephyr/kernel/timeout.c:299
299		} else if (K_TIMEOUT_EQ(timeout, K_NO_WAIT)) {
(gdb) n
311		return timepoint;
(gdb) bt
#0  sys_timepoint_calc (timeout=...) at /root/zephyr/zephyrproject/zephyr/kernel/timeout.c:311
#1  0x00104ac2 in k_heap_aligned_alloc (heap=0x11ffc4 <z_main_stack+860>, align=4, bytes=0, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/kheap.c:68
#2  0x00104b93 in k_heap_alloc (heap=<optimized out>, bytes=<optimized out>, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/kheap.c:112
#3  0x001002ae in mbox_msg_receive_and_append_to_queue (heap_size=<optimized out>,
    msg_size=<optimized out>, queue_value=<optimized out>)
    at /root/zephyr/zephyrproject/zephyr/samples/hello_world/src/main.c:18
#4  0x00000000 in ?? ()
(gdb) n
k_heap_aligned_alloc (heap=0x11ffc4 <z_main_stack+860>, align=4, bytes=0, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/kheap.c:71
71		k_spinlock_key_t key = k_spin_lock(&heap->lock);
(gdb) n
99			key = k_spin_lock(&heap->lock);
(gdb) n
80			ret = sys_heap_aligned_alloc(&heap->heap, align, bytes);
(gdb) n
```

**Environment (please complete the following information):**

- OS: Ubuntu 22.04 LTS
- Toolchain Zephyr SDK
- Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/143b14bb4fb779437fdeb8369e42fbb81db4a5fd

thedjnK, 2024-12-31:
> I noticed if the heap size is too small, the program might crash at k_heap_alloc. But if I set the heap size to 0x200, the program can run without error.
> 
> I am not sure if there is a bug or not since the parameters are correct and valid but, indeed, too small.

Yeah you need to set a heap size that is correct for your code, what exactly is the issue you are reporting here?

Rrooach, 2024-12-31:
> Yeah you need to set a heap size that is correct for your code, what exactly is the issue you are reporting here? Also you're trying to allocate a heap of size 0

My problem is if this is sort of bug in k_heap_alloc, where failure to check such case cause the program to crash

rruuaanng, 2025-01-02:
@Rrooach In other words, you want it to check if the bytes are 0 and handle that accordingly. Right?

Rrooach, 2025-01-02:
> In other words, you want it to check if the bytes are 0 and handle that accordingly. Right?

Yes, I think this function might need to include some checks to ensure that the arguments passed to it are valid. For example, cases where the requested heap size is too small or set to zero. Such validation could help avoid unnecessary crashes and make the function more robust and user-friendly.

Best regards,

rruuaanng, 2025-01-02:
It looks like the problem is with `k_heap_init`. But sorry, I'm not very familiar with this, when `heap_size` is 30, it also happens the same problem, I guess the minimum size of bytes needs to be greater than the alignment size or a base size

good
```
K_HEAP_DEFINE(my_heap, 0);

void mbox_msg_receive_and_append_to_queue(int msg_size, int queue_value) {
        struct k_mbox_msg msg;
        msg.tx_data = k_heap_alloc(&my_heap, msg_size, K_NO_WAIT);
}

int main(void)
{
	printf("Hello World! %s\n", CONFIG_BOARD_TARGET);
        mbox_msg_receive_and_append_to_queue(0, 0x8a9);
	printf("finish \n");
	return 0;
}
```
output
```
Booting from ROM..
*** Booting Zephyr OS build v4.0.0-2792-gee250a11ce50 ***
Hello World! qemu_x86/atom
finish 
```

amoham57, 2025-01-12:
I tried to reproduce this with on QEMU (cortex m3) 
[image] it seemingly jumped into a fault handler

peter-mitsis, 2025-01-30:
I see this issue as being one of a misconfiguration. The Zephyr kernel does provide the means to detect this misconfiguration in k_heap_init(), and it does detect it when CONFIG_ASSERT is enabled. It is just that the test did not take advantage of the means to detect it. Closing this.

closed 2025-01-30 by peter-mitsis