zephyrproject-rtos/zephyr #83480: Websocket shell backend segfault on closing connection

mrodgers-witekio, 2025-01-01:
**Describe the bug**

When using the websocket shell backend on the `native_sim` platform, closing the websocket connection causes a segfault.

**To Reproduce**
To build:
```
west build \
	-b native_sim \
	samples/net/sockets/echo_server -- \
	-DCONFIG_NET_SHELL=y \
	-DCONFIG_NET_LOG=y \
	-DCONFIG_NET_SOCKETS_SERVICE_STACK_SIZE=2048 \
	-DOVERLAY_CONFIG="overlay-ws-console.conf"
```

Then run the sample, navigate to http://192.0.2.1 in web browser (works in Firefox but not Chrome, see https://github.com/zephyrproject-rtos/zephyr/pull/83477). Once websocket shell connection is established, click the close button on the web page.

The segfault seems to arise from the `LOG_DBG` call inside `ws_recv`. I tried also enabling `CONFIG_LOG_MODE_DEFERRED` since it seemed plausible that logging inside a log backend would otherwise cause some issues, but this had the same result.

It definitely seems to be something logging-related, since disabling logging entirely on this backend (`CONFIG_SHELL_WEBSOCKET_INIT_LOG_LEVEL_NONE=y`) means that the websocket connection can be closed and re-opened with no issues.

**Expected behavior**
No segfault

**Impact**
Relatively minor, shell backend is still usable if it doesn't have logging enabled.

**Logs and console output**

Backtrace from running sample in GDB:

```
Thread 11 "net_socket_serv" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xf38fbac0 (LWP 337311)]
ws_recv (zvfs_pollfd=<optimised out>, ws=<optimised out>) at /home/matt/projects/zephyrproject/zephyr/subsys/shell/backends/shell_websocket.c:136
136			LOG_DBG("Websocket client closed connection");
(gdb) bt
#0  ws_recv (zvfs_pollfd=<optimised out>, ws=<optimised out>)
    at /home/matt/projects/zephyrproject/zephyr/subsys/shell/backends/shell_websocket.c:136
#1  ws_server_cb (evt=0xf38fb1fc)
    at /home/matt/projects/zephyrproject/zephyr/subsys/shell/backends/shell_websocket.c:186
#2  0x08056749 in net_socket_service_callback (
    pev=0x809bd40 <__z_net_socket_service_websocket_server>)
    at /home/matt/projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_service.c:127
#3  0x080569fe in call_work (event=<optimised out>, pev=<optimised out>)
    at /home/matt/projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_service.c:147
#4  trigger_work (pev=<optimised out>)
    at /home/matt/projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_service.c:170
#5  socket_service_thread ()
    at /home/matt/projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets_service.c:260
#6  0x0804b2b5 in z_thread_entry (entry=0x8056778 <socket_service_thread>, p1=0x0, p2=0x0, p3=0x0)
    at /home/matt/projects/zephyrproject/zephyr/lib/os/thread_entry.c:48
#7  0x08051e16 in posix_arch_thread_entry (pa_thread_status=0x80af1e8 <service_thread_stack+2024>)
    at /home/matt/projects/zephyrproject/zephyr/arch/posix/core/thread.c:96
#8  0x0808377c in nct_thread_starter (arg_el=0x80ba5b0)
    at /home/matt/projects/zephyrproject/zephyr/scripts/native_simulator//common/src/nct.c:366
#9  0xf7ce8c01 in ?? () from /lib/i386-linux-gnu/libc.so.6
#10 0xf7d8572c in ?? () from /lib/i386-linux-gnu/libc.so.6
```

**Environment (please complete the following information):**

Linux, Zephyr SDK, commit SHA `0ac9a6c5125`

rlubos, 2025-01-03:
Thanks for the very detailed report. I've spent some time today debugging this, I won't say that I fully understand the macrology behind the logger/shell modules, but it seems the segfault was caused by misconfiguration of the logger instance for the websocket shell backend.

This single-line change helps with the crash:
```patch
diff --git a/subsys/shell/backends/shell_websocket.c b/subsys/shell/backends/shell_websocket.c
index 62b8dc9a5f8..a05a9025151 100644
--- a/subsys/shell/backends/shell_websocket.c
+++ b/subsys/shell/backends/shell_websocket.c
@@ -16,7 +16,7 @@
 #include <zephyr/shell/shell_websocket.h>
 #include <zephyr/logging/log_backend_ws.h>
 
-LOG_MODULE_REGISTER(shell_websocket, CONFIG_SHELL_WEBSOCKET_INIT_LOG_LEVEL);
+LOG_MODULE_REGISTER(shell_websocket, CONFIG_SHELL_WEBSOCKET_LOG_LEVEL);
 
 #define WEBSOCKET_LINE_SIZE CONFIG_SHELL_WEBSOCKET_LINE_BUF_SIZE
 #define WEBSOCKET_TIMEOUT   CONFIG_SHELL_WEBSOCKET_SEND_TIMEOUT
```

If I understand correctly `CONFIG_SHELL_WEBSOCKET_INIT_LOG_LEVEL` defines the default log level that the websocket shell **backend** supports, and not the log level for the logger instance for the backend (I hope you follow). The former defaults to `5`, which is not a valid log level, and the logger instance inside `shell_websocket.c` ended up misconfigured. From what I've seen, the macrology behind the logging subsys did not set some dynamic log context pointer used by runtime filtering (a side note, disabling `CONFIG_LOG_RUNTIME_FILTERING` also works around the issue) and any log attempt ended up in a crash. W/ proper log level set, all seems to work, even with debug logging enabled.

**However**, while the above is true for the immediate log mode, there still seems to be some problem with the deferred mode... There no crash anymore, but sometimes the application hangs, most likely some thread (logger?) starts busy looping. Need to investigate this further so I'm not sending a PR yet.

mrodgers-witekio, 2025-01-06:
I did a bit more digging into the issues with deferred logging after applying your patch.

I think the problem is related to the fact that we have both a shell backend and a log backend trying to send logs to the same websocket.

The following patch seems to make deferred logging work again:

```
diff --git a/subsys/shell/backends/shell_websocket.c b/subsys/shell/backends/shell_websocket.c
index 62b8dc9a5f8..bef8d0f6e7e 100644
--- a/subsys/shell/backends/shell_websocket.c
+++ b/subsys/shell/backends/shell_websocket.c
@@ -16,7 +16,7 @@
 #include <zephyr/shell/shell_websocket.h>
 #include <zephyr/logging/log_backend_ws.h>
 
-LOG_MODULE_REGISTER(shell_websocket, CONFIG_SHELL_WEBSOCKET_INIT_LOG_LEVEL);
+LOG_MODULE_REGISTER(shell_websocket, CONFIG_SHELL_WEBSOCKET_LOG_LEVEL);
 
 #define WEBSOCKET_LINE_SIZE CONFIG_SHELL_WEBSOCKET_LINE_BUF_SIZE
 #define WEBSOCKET_TIMEOUT   CONFIG_SHELL_WEBSOCKET_SEND_TIMEOUT
@@ -213,7 +213,7 @@ static int shell_ws_init(struct shell_websocket *ctx, int ws_socket)
                goto error;
        }
 
-       log_backend_ws_register(ws_socket);
+       // log_backend_ws_register(ws_socket);
 
        return 0;
```

The `log_backend_ws` is not needed here, since logs are sent via the shell backend anyway.

What I'm not sure about though is if the issue is purely down to two log backends trying to access the same socket, or whether there's a bug in the `log_backend_ws.c` which would cause problems even if used on its own without the shell.

May be worth making a PR with these fixes at least and then investigate whether there are any further issues with the (non-shell) websocket log backend later, what do you think?

mrodgers-witekio, 2025-01-06:
At the very least I think there is an issue with the return value of `line_out` in `log_backend_ws.c`. This should be returning the number of bytes processed, but actually returns the return value from `ws_console_out`, which is the number of TX retries (usually zero).

But since there isn't an in-tree example of using the websocket logging backend without the websocket shell, it'll take some time to debug any further.

rlubos, 2025-01-09:
Just FYI, I'm back on it, you were right that the root cause for the busy-looping was `line_out` returning invalid value (which was ignored in case of immediate logging). I've identified some other issues while testing, I'll send a PR when I wrap everything up.

rlubos, 2025-01-09:
Ok, I've finally managed to get the websocket console to work reasonably, opened a PR with fixes: https://github.com/zephyrproject-rtos/zephyr/pull/83755
From what I've tested I didn't see any crashes/hangups/duplicated logs in various configurations, but some further testing would be appreciated.

mrodgers-witekio, 2025-01-09:
Great, I'll give it a try when I get a chance (likely tomorrow)

PR #83755 (2025-01-13): shell: websocket: Fix several issues with websocket shell/logger
With this set of fixes there're no more crashes/hangups/duplicated logs when running websocket console.

Fixes #83480
closed 2025-01-13 by kartben