zephyrproject-rtos/zephyr #83494: Move all `DT_IRQ_*` macros from `devicetree.h` to `interrupt_controller.h`.

rruuaanng, 2025-01-02:
## Introduction

Currently, file `deivcetree.h` mainly contains general macros for device trees, such as for-each macros and other node-related macros. It also includes interrupt-related macros, such as `DT_IRQ_*`. I believe these should be moved to their appropriate location in file `interrupt_controller.h`, which would make the semantics more intuitive. This change would make it easier for users to know they should include specific hardware-related device tree headers during usage.

### Problem description

This makes the division of the device tree API unclear. IRQ related macros should reside in files specific to the corresponding system.

### Proposed change

Move all `DT_IRQ_*` macros from file `devicetree.h` to file `interrupt_controller.h`.

## Detailed RFC

### Proposed change (Detailed)

Move all `DT_IRQ_*` macros from file `devicetree.h` to file `interrupt_controller.h`.

### Dependencies

None

### Concerns and Unresolved Questions

This might force users to change their header file import paths, so I believe this should be negotiated as much as possible.

## Alternatives

rruuaanng, 2025-01-02:
At the very least, file `devicetree.h` does not contain any other system-specific macros, such as those for `GPIO`, `SPI`, or others.

henrikbrixandersen, 2025-01-02:
> At the very least, file `devicetree.h` does not contain any other system-specific macros, such as those for `GPIO`, `SPI`, or others.

The interrupt properties are described in the devicetree specification, the others are not. 

The reason for putting e.g. the GPIO_DT_SPEC macros in the GPIO driver header file is, that these depend on the `struct gpio_dt_spec` defined there. Only macros without such dependencies can go in the `include/zephyr/devicetree*` hierarchy.

rruuaanng, 2025-01-02:
Yes, interrupts are defined in the ePAPR specification and are specifically mentioned. However, an important point to note is that the macro name in `driver/gpio.h` is `GPIO_DT_*`, while the macro name in `devicetree/gpio.h` is `DT_GPIO_*`. If the purpose of `devicetree.h` is to describe the generic interfaces defined in the standard, shouldn't the contents of `devicetree/interrupt_controller.h` be moved into `devicetree.h`?

rruuaanng, 2025-01-02:
After careful consideration, I believe that making users undertake a large-scale migration of a stable API for the sake of surface-level consistency is the wrong decision, even though there is currently no stability maintenance for the device tree API. Therefore, I am closing this issue, even though I really wanted to proceed with it.

@henrikbrixandersen Thank you for your reply :)

closed 2025-01-02 by rruuaanng