zephyrproject-rtos/zephyr #83519: Add ThorVG (lottie) support to LVGL

thomasstenersen, 2025-01-03:
**Is your feature request related to a problem? Please describe.**
PR #68168 updates LVGL to version 9+ which now has built-in support for handling lottie animations through ThorVG. However, to limit the scope of the PR, it does not include the necessary changes to make ThorVG work in Zephyr.

To get the added ThorVG support there are a few minor things that needs to be fixed:

1. Fix some minor issues related to cross compilation (at least with GNU Arm toolchain) in LVGL/ThorVG: 
   - https://github.com/thomasstenersen/lvgl/commit/e94f308886a4b8504b7813b0d308097c8b2b65dd
   - https://github.com/thomasstenersen/lvgl/commit/0b70f62ed8c56a7ae85fbbc765749f1d007fb84e
   - https://github.com/thomasstenersen/lvgl/commit/cb054e5bc13f88daf5da3f47a4ae00d2ce7dda1f
2. [Expose `strdup()`](https://github.com/thomasstenersen/zephyr/commit/0e1f95f94c3ad7b3bf2f8419ddd99b9435e2254c) in the libc
3. Add the ThorVG sources to the LVGL module: https://github.com/thomasstenersen/zephyr/commit/e4665f58dcf41e55e2002199d7c49915fb75e28c

**Describe the solution you'd like**
It looks like ThorVG is copied directly from https://github.com/thorvg/thorvg into https://github.com/lvgl/lvgl, so fixing the first issue *should* be done with a PR to ThorVG first before updating https://github.com/lvgl/lvgl and pulling in the update into Zephyr. Exposing `strdup()` in the libc can be done independently though.

**Additional context**
I've created a proof of context branch [in my Zephyr fork](https://github.com/thomasstenersen/zephyr/tree/feature/lottie). I've also added a sample ([`samples/subsys/display/lvgl_lottie`](https://github.com/zephyrproject-rtos/zephyr/commit/f1e5934b08c315b83d58be9d44e4c56699bf8f44)) to showcase the feature. The sample has been tested on `native_sim` and `stm32f746g_disco`.

faxe1008, 2025-01-07:
I started checking those:

Usage of alloca should be fine, issue seems to be that they are including the wrong header:
https://github.com/thomasstenersen/lvgl/commit/0b70f62ed8c56a7ae85fbbc765749f1d007fb84e#diff-92a7e772b0481b241d2ff321660a14464695a54de5dcd80976a9f79eb9f511acL29

So the fix is possibly to include `alloca.h` if `__ZEPHYR__` is defined or is there more of an issue to alloca? I see that its used in some places intree.

faxe1008, 2025-01-08:
`strdup` is available if ` _POSIX_C_SOURCE=200809L` is defined.

faxe1008, 2025-01-08:
Trying to fix the thorvg issues at the source: https://github.com/thorvg/thorvg/pull/3107

faxe1008, 2025-02-26:
So the fixes were upstreamed to ThorVG, waiting for the update of the library within LVGL.
See the related discussion: https://github.com/lvgl/lvgl/issues/7834
ThorVG v1.0 will be released somewhere around March.

nguterresn, 2025-03-27:
This is still an issue with Zephyr 4.1.0. The following functions have caused compilation errors due to missing reference:

- strdup
- alloca
- strcasecmp

> strdup is available if  _POSIX_C_SOURCE=200809L is defined.

This has fixed strdup for me. 

> So the fix is possibly to include alloca.h if __ZEPHYR__ is defined or is there more of an issue to alloca? I see that it's used in some places intree.

Yes, it fixes the issue:

```c
#ifdef _WIN32
    #include <malloc.h>
#elif defined(__linux__) || defined(__ZEPHYR__)
    #include <alloca.h>
#else
    #include <stdlib.h>
#endif
```

However, for `strcasecmp`, I had to include a function as such:

```c
static int strcasecmp(const char* s1, const char* s2) {
    while (*s1 && *s2) {
        int diff = tolower((unsigned char)*s1) - tolower((unsigned char)*s2);
        if (diff != 0)
            return diff;
        s1++;
        s2++;
    }
    return tolower((unsigned char)*s1) - tolower((unsigned char)*s2);
}
```

Is there a fix for this?

phb98, 2025-10-03:
Hello, what is the status of this feature ? Look like the main HEAD does not support thorVG, right ?

Pavllick, 2025-10-04:
@phb98 Doesn't look like it is supported by main yet, but patching LVGL's CMake with @thomasstenersen's [commit](https://github.com/thomasstenersen/zephyr/commit/e4665f58dcf41e55e2002199d7c49915fb75e28c) fixed it for me.

faxe1008, 2025-10-04:
The issue is that it currently has lots of compile warnings meaning having it built and pass CI is currently not doable. 

I submitted the fixes to ThorVG itsself, but LVGL has to update their intree version first. Bit since their version has a bunch of modifications, its not as easy as just copying it seems.