zephyrproject-rtos/zephyr #83543: drivers: clock: stm32: stm32c03 fail to build

nandojve, 2025-01-03:
**Describe the bug**

The stm32c0 fail to build because of missing `apb2_prescaler`.

**Expected behavior**
All stm32 SoC should build with a default value for apb to avoid this issue.

**Impact**
Not possible to build when using timers3.

```
include/generated/zephyr/devicetree_generated.h:931:36: error: 'DT_N_S_soc_S_rcc_40021000_P_apb2_prescaler' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_rcc_40021000_P_apb1_prescaler'?
  931 | #define DT_N_NODELABEL_rcc         DT_N_S_soc_S_rcc_40021000
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
include/zephyr/devicetree.h:5231:29: note: in definition of macro 'DT_CAT3'
 5231 | #define DT_CAT3(a1, a2, a3) a1 ## a2 ## a3
      |                             ^~
include/zephyr/drivers/clock_control/stm32_clock_control.h:73:33: note: in expansion of macro 'DT_PROP'
   73 | #define STM32_APB2_PRESCALER    DT_PROP(DT_NODELABEL(rcc), apb2_prescaler)
      |                                 ^~~~~~~
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 0.17
 - Commit mainline

mathieuchopstm, 2025-01-06:
I have not been able to reproduce this error by building both `samples/drivers/counter/alarm/` and `tests/drivers/pwm/pwm_api/` for board `nucleo_c031c6`, and the error message you provided does not indicate in which place the macro is being expanded.

As far as I can tell, there are only three places where `STM32_APB2_PRESCALER` is used, all of which make sure the property exists before using the macro, so I'm a bit confused as to where it comes from.

Can you please give instructions on how to reproduce the issue?

---

1. Should not cause the error: gated behind property existance check
https://github.com/zephyrproject-rtos/zephyr/blob/72694a9d096d7c649045758331637f1b240de2e4/drivers/clock_control/clock_stm32_ll_common.c#L342-L344

2. Should not cause the error: gated behind `!CONFIG_SOC_SERIES_STM32C0X`
https://github.com/zephyrproject-rtos/zephyr/blob/72694a9d096d7c649045758331637f1b240de2e4/drivers/pwm/pwm_stm32.c#L248-L257

3. Should not cause the error: gated behind `!HAS_COMPAT(st,stm32f0-rcc)`
https://github.com/zephyrproject-rtos/zephyr/blob/72694a9d096d7c649045758331637f1b240de2e4/drivers/counter/counter_ll_stm32_timer.c#L400-L408
which is the compatible used by STM32C0 RCC:
https://github.com/zephyrproject-rtos/zephyr/blob/72694a9d096d7c649045758331637f1b240de2e4/dts/arm/st/c0/stm32c0.dtsi#L103-L104

nandojve, 2025-01-06:
Hi @mathieuchopstm ,

You are right! This was found initially when the #70951 was open. I was searching on mainline about STM32_APB2_PRESCALER, after your report, and found that current PWM driver have a different implementation we have on #70951. https://github.com/zephyrproject-rtos/zephyr/blob/88b9cb6efcb18cd91bbcbb440f455724a872c6b9/drivers/pwm/pwm_stm32.c#L248-L258

With this it is not necessary the #83544 patch since the problem was found inside the freq-meter.

closed 2025-01-06 by nandojve