zephyrproject-rtos/zephyr #83551: samples\bluetooth\central_ht fault on frdm_mcxw71

JingsaiLu, 2025-01-04:
**Describe the bug**

samples\bluetooth\central_ht fault on frdm_mcxw71
**To Reproduce**

follow this guide to flash nbu fw, https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/nxp/frdm_mcxw71/doc/index.rst#bluetooth

west build -p always -b frdm_mcxw71 zephyr/samples/bluetooth/central_ht
west flash -r jlink

will see the fault print on the console

**Expected behavior**

**Impact**

**Logs and console output**

[image]

**Environment (please complete the following information):**

 - OS: ( Windows)
 - Toolchain (Zephyr SDK)
 - Commit SHA (commit ad1657f04b3)

**Additional context**

if checkout to v4.0 branch ([image]
this app works well on frdm_mcxw71
)

JingsaiLu, 2025-01-06:
try to add increase stack size, CONFIG_MAIN_STACK_SIZE=2048
app will work well, 

not sure if this issue happens on all zephyr ble platforms, 
and why require large stack size on main branch.

decsny, 2025-01-06:
@axelnxp

JingsaiLu, 2025-01-07:
more info, 

one difference between main branch and v4.0 branch for the bt, is 
the security lib from https://github.com/intel/tinycrypt/tree/master to mbedtls, 

the root cause is from mbedtls_psa_random_seed(), which make stack overflow, with default stack size setting,

no specific define for MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG on w71,
so app is using the default mbedtls rng, which consume more memory.

`
    static psa_status_t mbedtls_psa_random_seed(mbedtls_psa_random_context_t *rng)
    {
    #if defined(MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG)
        /* Do nothing: the external RNG seeds itself. */
        (void) rng;
        return PSA_SUCCESS;
    #else /* MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG */
        const unsigned char drbg_seed[] = "PSA";
        int ret = mbedtls_psa_drbg_seed(&rng->drbg, &rng->entropy,
                                        drbg_seed, sizeof(drbg_seed) - 1);
        return mbedtls_to_psa_error(ret);
    #endif /* MBEDTLS_PSA_CRYPTO_EXTERNAL_RNG */
    }

`

dleach02, 2025-01-10:
yes, the deprecation of tinycrypt has had a ripple affect.

@JingsaiLu can you submit a PR to increase the stack size?

JingsaiLu, 2025-01-13:
hi @dleach02 , checked with Yassine, 
alex is doing this for both mcxw71 and rw612.

PR #83884 (2025-01-15): [NXP][MCXW][RW] Fix stack overflow in BLE samples after migration to mbedtls
mbedtls is now used in BLE samples, increasing the stack depth needed of the calling threads. This was causing stack overflows in several BLE samples.
Increasing the main stack size for common samples, but also the shell stack size for samples calling bt API from the shell thread like the bt shell.

Fixes #83551
closed 2025-01-15 by kartben