zephyrproject-rtos/zephyr #83567: LVGL: Multiple Displays

towen, 2025-01-05:
**Is your feature request related to a problem? Please describe.**
It does not appear to be possible to use multiple displays with LVGL in Zephyr, even though LVGL itself supports multiple displays.
In [`modules/lvgl/lvgl.c`](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/lvgl.c#L29) zephyr is hardcoded to reference one display device: 
`#define DISPLAY_NODE          DT_CHOSEN(zephyr_display)`

**Describe the solution you'd like**
If not using CONFIG_LV_Z_AUTO_INIT, then LVGL could be initialised manually with two separate contexts, using different display devices.
Instead of simply calling `lvgl_init()`, we could call a new function `lvgl_init_display(const struct device *display_dev)`

**Describe alternatives you've considered**
lvgl.c uses unencapsulated variables (display and disp_data) and would need to be rewritten.. There are various KConfigs to define display settings (colour depth, etc) so supporting multiple *different* displays may be difficult, but multiple identical displays should be possible with less effort.

towen, 2025-01-24:
Hey @JarmouniA, thanks a lot!
For me at least, a Kconfig with the number of displays, and an alias zephyr_display_n would be fine. But for others, maybe there is a cleaner way to avoid changing existing devicetrees when the number of displays is 1. 
If it is >1 then an order should be defined somewhere, ideally in devicetree.. 
"zephyr_display" perhaps could still be used to indicate the "primary" display in some multi-display applications.

But, "Inferring" the number of displays directly from devicetree sounds complicated and could be addressed in future if needed? I can't think of an app that would need to ask devicetree how many displays it has. A "minimum number of connected displays" for a particular app, defined in Kconfig, sounds OK to me.

faxe1008, 2025-01-24:
@JarmouniA thanks for your effort in attempting this. I see these options primarily:

- Go down the route for a RFC and change devicetree so that the `chosen` node supports multiple displays. This would be best because the selection of which displays are used stays in the devicetree, which is arguably the right place. This would also allow to create the buffers statically for all displays in the same fashion.
- Leave the devicetree as is but expose some of the methods for allocating dynamic buffers and setting up the handlers to the public. If someone wants more then one display, leave it to them to initialize it.

About the Kconfig usage: I forsee quite a lot of resistance for that idea. Putting hardware related things which are handled in devicetree historically has been disliked (rightfully so imo :^) )

faxe1008, 2025-02-14:
> Is there an alternative way to initialise multiple displays, given that the devicetree PR above has been closed?

Yes there is nothing preventing you from doing this manually, it's just not as plug and play as the single-screen use case.
You have to create a new `lv_display_t`, create drabuffer(s) for it, setup the flush callback and the properties of the display (like pixelformat etc.). This should work just fine.

PR #86815 (2025-05-08): LVGL: multi-display support
LVGL multi-display support
Closes #83567

```
west build -p always -b native_sim/native/64 samples/modules/lvgl/multi_display
./build/zephyr/zephyr.exe
```
[image]

Tested for potential regressions with `samples/modules/lvgl/demos`, `samples/subsys/display/lvgl`, & `tests/lib/gui/lvgl` on native_sim_64 and STM32H750B_DK.
closed 2025-05-08 by kartben