zephyrproject-rtos/zephyr #83574: Add a new mesh sample as a client role

caogule, 2025-01-06:
**Is your feature request related to a problem? Please describe.**
I hope that the client and the server implement data exchange through the light model.

**Describe the solution you'd like**

There are currently two samples.
third_party\bluetooth\samples\mesh
third_party\bluetooth\samples\mesh_provisioner

I need three samples: provisioner, client, server. Both the client and the server implement network configuration through provisoner. The client and the server implement data exchange through the light model.
**Describe alternatives you've considered**

**Additional context**

ShahinHaque97, 2025-01-06:
@caogule I'd have a look at ncs (https://github.com/nrfconnect/sdk-nrf/tree/main/samples/bluetooth/mesh)

PavelVPV, 2025-03-21:
> > There is no any "light model". Neither Meh Protocol nor Mesh Model specification has it. Furthermore, Zephyr does not have Mesh Model specification implementation if "light model" was assumed as model from this specification.
> 
> BT_MESH_MODEL_ID_GEN_ONOFF_CLI BT_MESH_MODEL_ID_GEN_ONOFF_SRV

Please have a look at nRF Connect SDK, these models as many others are implemented there: https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/libraries/bluetooth/mesh/models.html

alxelax, 2025-03-21:
@caogule, It already exists. Please read documentation.
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/boards/nordic/mesh/onoff-app