zephyrproject-rtos/zephyr #83582: Unable to obtain the link state using phy_get_link_state()

VineetaNarkhede-eaton, 2025-01-06:
**Describe the bug**

I need to obtain the PHY speed. I am using phy_get_link_state() to retrieve the speed, but this function calls eth_iface_init(), which has already been invoked during boot-up without any problem. When eth_iface_init() is called from phy_get_link_state() again. the dev pointer in eth_iface_init() points to null, causing a fault. It appears that get_link() is not pointing to the correct function.

 - Target: STM32H753VIT6

**To Reproduce**

```
    iface = net_if_get_default();
    if (iface == NULL) 
    {
        printk("\n No default network interface found");
        return;
    }

	dev = net_if_get_device(iface);
    if (dev == NULL) 
    {
        printk("\nFailed to get device from network interface");
        return;
    }
	printk( "Device name: %s", dev->name);

	dev_data = (struct eth_stm32_hal_dev_data *)dev->data;
    if (dev_data == NULL) 
    {
        printk("\nFailed to get device data");
        return;
    }
	
    ret = phy_get_link_state(dev, &state);
```

**Expected behavior**
phy_get_link_state() should return phy_link_speed

**Impact**
Device resets when calling phy_get_link_state().

**Environment:**
 - OS:  Windows
 - zephyr version: 3.2.0

erwango, 2025-01-15:
@VineetaNarkhede-eaton Please note that V3.2.0 is not supported anymore.
Would you mind testing on more recent release (starting v3.7.0) ?

maass-hamburg, 2025-02-24:
``phy_get_link_state()`` is from the ethphy driver api. so the dev should be the ethernet phy dev, not the ethernet  dev.
``net_if_get_device(iface)`` will give you the ethernet dev. You should use ``net_eth_get_phy(iface)`` instead to get the phy dev.
This requires that the according ethernet driver implements the ``get_phy`` function, which the current st ethernet driver does not, but there a two PRs pending implementing it, but they will only be merged after the next release, so it will probably work in version 4.2. You can also get the phy dev from the devicetree.

closed 2025-02-24 by maass-hamburg