zephyrproject-rtos/zephyr #83584: Bluetooth: Controller: ISO: CIS MIC errors commonly occur

Thalley, 2025-01-06:
**Describe the bug**
It has become a common occurrence to see MIC errors causing CIS disconnects (reason 0x3d (`BT_HCI_ERR_TERM_DUE_TO_MIC_FAIL`)) in various configurations. These are often found in the LE Audio tests. They occur with both unidirectional and bidirectional configurations and with both a single peripheral and multiple peripherals. 

**To Reproduce**
1. Apply https://github.com/zephyrproject-rtos/zephyr/pull/83539
2. Run https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/bsim/bluetooth/audio/compile.sh
3. Run any of the tests that disabled in that PR marked with "reason 0x3d"

**Expected behavior**
Expect to see no MIC errors when sending audio on a CIS on an encrypted ACL. 

**Impact**
Significant as it occurs often and seemingly not something that is fixable by the application. Causes CI issues. 

**Logs and console output**
Common log for these tests (look for `bt_iso_chan_disconnected: 0xXXXXXXXX, reason 0x3d`):
```
 d_00: @00:00:12.995916  Started stream 0x84ce7c0
d_00: @00:00:12.995916  [00:00:12.995,910] <inf> bap_stream_tx: Registered 0x84ce7c0 for TX
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_cap_initiator: bt_cap_initiator_started: cap_stream 0x84ce7c0
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_cap_initiator: update_proc_done_cnt: proc 1 subproc 5: 2/4
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_cap_initiator: bt_cap_initiator_started: Stream 0x84ce7c0 started (2/4 streams done)
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_bap_stream: bt_bap_stream_start: stream 0x84ce920 ep 0x8537724
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_bap_unicast_client: bt_bap_unicast_client_start: stream 0x84ce920
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_bap_unicast_client: unicast_client_ep_start: ep 0x8537724 buf 0x8537d5c
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_bap_unicast_client: unicast_client_ep_start: id 0x03
d_00: @00:00:12.995916  [00:00:12.995,910] <dbg> bt_bap_unicast_client: bt_bap_unicast_client_ep_send: conn 0x84fd340 ep 0x8537724 buf 0x8537d5c len 3
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_disconnected: 0x850dee0
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_chan_disconnected: 0x8535e60, reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x8535e60 iso 0x850dee0 connected -> disconnected
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_cleanup_acl: 0x850dee0
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_iso_disconnected: stream 0x84cf160 ep 0x852e448 state streaming reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_iso_disconnected: stream 0x84cf1b8 ep 0x852e7d8 state enabling reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_get_status: ep 0x852e448 id 0x01 state qos-configured
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_get_status_qos: dir sink codec id 0x06 interval 7500 framing 0x00 phy 0x02 rtn 1 latency 15 pd 10000
d_01: @00:00:13.006947  [00:00:13.006,927] <wrn> bt_conn: bt_conn_create_pdu_timeout: Timeout discarded. No blocking in syswq.
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: state_transition_work_handler: ase 0x852e444 ep 0x852e448 id 0x01 streaming -> qos-configured
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_disconnected: 0x850dfd0
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_chan_disconnected: 0x8535ef0, reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x8535ef0 iso 0x850dfd0 connected -> disconnected
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_iso: bt_iso_cleanup_acl: 0x850dfd0
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_iso_disconnected: stream 0x84cf18c ep 0x852e610 state streaming reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_iso_disconnected: stream 0x84cf1e4 ep 0x852e9a0 state enabling reason 0x3d
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_get_status: ep 0x852e610 id 0x02 state qos-configured
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: ascs_ep_get_status_qos: dir sink codec id 0x06 interval 7500 framing 0x00 phy 0x02 rtn 1 latency 15 pd 10000
d_01: @00:00:13.006947  [00:00:13.006,927] <wrn> bt_conn: bt_conn_create_pdu_timeout: Timeout discarded. No blocking in syswq.
d_01: @00:00:13.006947  [00:00:13.006,927] <dbg> bt_ascs: state_transition_work_handler: ase 0x852e60c ep 0x852e610 id 0x02 streaming -> qos-configured
d_00: @00:00:13.013547  [00:00:13.013,519] <inf> bap_stream_tx: Stream 0x84ce7c0 sent 0 SDUs
d_01: @00:00:13.065698  [00:00:13.065,673] <dbg> bt_ascs: ascs_cp_write: conn 0x84fd340 attr 0x8490c28 buf 0x85acf0f len 3 op Receiver Start Ready (0x04)
d_01: @00:00:13.065698  [00:00:13.065,673] <dbg> bt_ascs: ascs_start: num_ases 1
d_01: @00:00:13.065698  [00:00:13.065,673] <dbg> bt_ascs: ascs_start: ase 0x03
d_01: @00:00:13.065698  [00:00:13.065,673] <dbg> bt_ascs: ase_start: ase 0x852e7d4
d_01: @00:00:13.065698  [00:00:13.065,673] <wrn> bt_ascs: ase_start: Start failed: CIS not connected: 0
d_01: @00:00:13.065698  [00:00:13.065,673] <dbg> bt_ascs: ascs_cp_rsp_add: id 0x03 code Invalid ASE State (0x04) reason None (0x00)
d_00: @00:00:13.066004  [00:00:13.065,979] <dbg> bt_bap_unicast_client: unicast_client_ep_notify: conn 0x84fd340 ep 0x853737c len 17
d_00: @00:00:13.066004  [00:00:13.065,979] <dbg> bt_cap_common: bt_cap_common_abort_proc: Aborting proc 1 with subproc 5 for 0x84fd340: -74
d_00: @00:00:13.066004  Stream 0x84ce660 stopped with reason 0x13
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/83539 (but also occurs for https://github.com/zephyrproject-rtos/zephyr/pull/81093). Both running on `b7b3449f482cd9fb736e3a79363d390fd648051d`

**Additional context**
See also https://github.com/zephyrproject-rtos/zephyr/actions/runs/12581046107/job/35064082709?pr=81093

cvinayak, 2025-01-06:
Is this reproducible on target devices or it is just issue with bsim testing? Which sample can I use on target to reproduce and analyze the MIC failure?

Thalley, 2025-01-06:
I haven't tried to replicate on target. Given the semi-randomness of this issue, it's likely timing related, but I can try to replicate it with the shell. I don't think any of the existing samples run with the configurations that is causing issues

mringwal, 2025-01-07:
FWIW, I'm getting the same error (0x3d - Connection terminated due to MIC failure) when trying to stream audio from an Android Pixel 7a to the PTS dongle with new firmware 13.33137.0.0.

Thalley, 2025-01-24:
@cvinayak It can easily be reproduced on target using https://github.com/zephyrproject-rtos/zephyr/pull/84507

Thalley, 2025-03-13:
Reopening as we see this happening again in https://github.com/zephyrproject-rtos/zephyr/pull/87030

PR #84640 (2025-02-05): Bluetooth: Controller: Fix incorrect event_count when CIG overlaps
Fix incorrect event_count use in CIG events when the next
CIG interval's prepare overlaps with the current CIG event.
Use separate event_count_prepare variable in ULL and copy
the value in LLL event.

Fixes #83584.

Other miscellaneous improvements:
- Fix Connected ISO to use accumulated latency to update the
payload count.
- Fix incorrect elapsed events value when event prepare are
aborted in the pipeline. This can caused premature
supervision timeouts.
Relates to commit https://github.com/zephyrproject-rtos/zephyr/commit/2470 [... cut]
PR #87538 (2025-03-29): Bluetooth: Controller: Fix incorrect CIS payload count at CIS Estab
Fix incorrect payload count at CIS Establish due to existing
CIG event overlapping the ACL event at the instant when the
CIS gets the active flag set.

The overlapping CIG event picked up the new CIS that had its
active flag set in the current CIG event instead of at the
actual CIS offset which is in the next CIG event.

Fix incorrect CIS offset in use if instant is picked from
the peer sent CIS RSP PDU. Instead, keep the instant that
was sent in the CIS REQ PDU as the instant to send in the
CIS IND PDU.

This fixes CIS failed [... cut]
closed 2025-03-29 by kartben