zephyrproject-rtos/zephyr #83591: drivers: stepper: API: Do not limit velocity to integer

andre-stefanov, 2025-01-06:
## Introduction

Current step/dir stepper API is allowing to set velocity as a `uint32_t` representing steps per second. The API has to allow higher velocity resolution.

### Problem description

There are projects which require high **precision** and **resolution** of stepper speed. While the timer counter allows resolutions as high as MCU frequency (prescaler = 0), the stepper API limits it only to integer of steps per second:

```C
__syscall int stepper_set_max_velocity(
    const struct device *dev, 
    uint32_t micro_steps_per_second // steps per second
);
```
```C
__syscall int stepper_run(
    const struct device * dev, 
    enum stepper_direction direction, 
    uint32_t velocity // steps per second
)
```

This limits usable resolution especially with slow velocities. For example it is not possible to define a velocity of 10.52 steps per second even if the counter resolution would make it possible otherwise. This is a very crucial limitation for projects like astrophotography tracker which require low stepper speeds with very precise velocities which also allow compensation of mechanical imperfections.

Additionally such an API is hiding actual limitations of the hardware and devicetree configuration. The timing source (counter or system tick) leads to only limited set of resulting step frequencies but the stepper api allows all the values in between and rounds it up to the next valid timing value (see [code](https://github.com/zephyrproject-rtos/zephyr/blob/411f9d8fcb84e46297290b17e006896f9f219ab8/drivers/stepper/step_dir/step_dir_stepper_counter_timing.c#L31)).

### Proposed change

**Context**

The actual issue results from discussion #68547 where it was recommended to use integers for stepper control to improve performance on MCUs without an FPU. While it is reasonable to use steps (`int`) instead of e.g. angles (`float`) for actual position parameters (relative and positive), the speed is something completely different and is not related to the nature of a motor with discrete positions (as contrast to DC motors like discussed in the #68547). The actual transformation calculation from speed (steps per second) to timer counter interval (integer ticks) is infrequent (only during speed changes), can be precalculated (lookup tables or constants) and thus does not cause performance issues in most projects even on chips without FPU. This can be observed in the Arduino ecosystem with libraries like [AccelStepper](https://github.com/waspinator/AccelStepper) or [FastAccelStepper](https://github.com/gin66/FastAccelStepper). These are mostly used on 8bit AVR MCUs which have a horrible floating point performance.

**Requirements:**
- Hardware capabilities should not be limited by the API in regard to velocity resolution
- Solution should not introduce performance losses
- API should be easy to use from application code

**Change**

Existing API should be extended by several functions:
1. Run stepper by specifying stepping interval. The interval is based on the tick of the timing source. This should be the main function and be used by others. It contains logic for timing source configuration.
2. Run stepper by specifying steps per second as floating point number. This function simplifies the API usage by hiding specifics of used timing source. This comes at a cost of using floating number arithmetic once during the interval calculation.

The existing function (which accepts velocity as `uint32_t` steps per second) can be kept while delegating the call to function 1 analogous to floating point velocity function.

## Detailed RFC

Main change enabling the solution is to extend the api of `include/drivers/stepper.h` by following functions:

```C
/**
 * @brief Set the minimum interval between steps
 * 
 * @details This function sets the minimum interval between steps in the stepper motor controller.
 * This function is non-blocking.
 * @note Setting min interval does not set the motor into motion, a combination of set_min_interval
 * and move is required to set the motor into motion.
 * 
 * @param dev pointer to the stepper motor controller instance
 * @param ticks The number of timing source ticks between single steps.
 * 
 * @retval -EIO General input / output error
 * @retval -ENOSYS If not implemented by device driver
 * @retval 0 Success
 */
__syscall int stepper_set_min_interval(const struct device *dev, uint32_t ticks);
```
```C
/**
 * @brief Run the stepper motor for a specified interval in a given direction.
 *
 * @details If interval is > 0, motor shall be set into motion and run incessantly at the specified stepping 
 * interval until and unless stalled or stopped using some other command, for instance, motor_enable(false).
 * This function is non-blocking.
 * 
 *
 * @param dev Pointer to the device structure for the driver instance.
 * @param direction The direction in which to run the stepper motor.
 * @param ticks The number of timing source ticks between single steps.
 * 				- > 0: Run the stepper with the given interval in a given direction
 * 				- 0: Stop the stepper
 *
 * @retval -ECANCELED If the stepper is disabled
 * @retval -EIO General input / output error
 * @retval -ENOSYS If not implemented by device driver
 * @retval 0 Success
 */
__syscall int stepper_run_interval(const struct device *dev, enum stepper_direction direction, uint32_t ticks);
```
These functions should behave the same as the already existing `stepper_set_max_velocity` and `stepper_run` functions but without the transformation of steps per second to tick interval. Application code calling this function will be responsible for interval calculation. This slightly increases the complexity because now application has to know the timing source frequency to convert steps/s to interval. On the other side this allows the application to decide the rounding strategy of steps/s to interval (round up/down/closest).

Already existing functions `set_max_velocity` and `stepper_run` can be left to avoid breaking API changes.

In addition two convenience function overloads of `stepper_set_max_velocity` and `stepper_run` can be created to allow hight stepping resolution without bothering with timing source frequencies in application code:

```C
__syscall int stepper_set_max_velocity(const struct device *dev, float steps_per_sec);
```
```C
__syscall int stepper_run(const struct device *dev, float steps_per_sec);
```
Both functions do not need an extra `stepper_direction` parameter since the sign of the velocity parameter will determine the rotation direction.

In case the usage of floats here is problematic on soft-float MCUs, one can guard them by an appropriate Kconfig flag which could default to value of CONFIG_FPU.

### Proposed change (Detailed)

1. Add before mentioned new function declarations to stepper.h
2. Change `stepper_driver_api`
    1. replace `stepper_set_max_velocity_t` with `stepper_set_min_interval_t` and it in stepper.h
    2. change `stepper_run_t` to accept interval in ticks instead of velocity in integer steps per second
3. Adjust driver/stepper accordingly to deal correctly with new api for SPI or step/dir type of drivers
4. Adapt and fix tests

### Dependencies

At the moment I do not see any required changes in other components since stepper component is new and isolated.

### Concerns and Unresolved Questions

Stepper functions providing float support wont be able to properly check against velocity = 0.0f probably due to the floating-point error. Old functions stopped the stepper in case velocity 0 was provided. Should this case be handled here somehow as well? We could check if the absolute velocity value is lower than some minimum threshold.

## Alternatives

At the moment no alternatives discovered.

andre-stefanov, 2025-01-06:
Meanwhile i started to prepare proposed changes and will link the PR later here.

andre-stefanov, 2025-01-06:
> I understand the idea, the only thing I see is that the proposal of using ticks only works smoothly imo. in the context of a step-dir interface. If you control the driver IC via other means (SPI, etc.) this means that the ticks are in the context of the driver IC. But the second you swap the IC for another one this means you need to adjust the tickcount possibly. So it kinda breaks the abstraction that the API provides.
> 

You are absolutely right. This is probably the better way of exposing these functions. Should they then guarded by the `CONFIG_STEP_DIR_STEPPER` flag in stepper.h or rather an extra header file under `include/drivers/stepper/step_dir.h`?

> What about dealing with this by having the step-dir code expose two "escape-hatch" APIs which offer ticks as a parameter instead of microsteps/second?
> 
> ```
> int step_dir_stepper_run_ticks(const struct device* dev, enum step_direction, uint32_t ticks);
> int step_dir_stepper_set_min_step_ticks(const struct device* dev, uint32_t ticks);
> ```

Sure this sounds good. Do you know an example of other escape-hatch APIs in zephyr code which i could use as reference?

jilaypandya, 2025-01-06:
> You are absolutely right. This is probably the better way of exposing these functions. Should they then guarded by the `CONFIG_STEP_DIR_STEPPER` flag in stepper.h or rather an extra header file [...]

`CONFIG_STEP_DIR_STEPPER` +1

faxe1008, 2025-01-07:
> You are absolutely right. This is probably the better way of exposing these functions. Should they then guarded by the CONFIG_STEP_DIR_STEPPER flag in stepper.h or rather an extra header file under include/drivers/stepper/step_dir.h?

yes I think `include/drivers/stepper/step_dir.h` is the way to go here since this is a public header, which is what we want.

> Sure this sounds good. Do you know an example of other escape-hatch APIs in zephyr code which i could use as reference?

Many of the sensor have some additional APIs exposed which are awkward to achieve using the sensor API. For example:
https://github.com/zephyrproject-rtos/zephyr/blob/bc42004d1be40d9b5bec2d3e8c600780b644ff6e/include/zephyr/drivers/sensor/scd4x.h#L53

Some more comments on this? @bjarki-andreasen maybe :) ?

faxe1008, 2025-01-08:
> I'm all for setting stepper interval in some higher resolution time unit:
> 
> ```
> int stepper_set_step_interval(const struct device *dev, uint64_t step_interval_us);
> ```

Maybe I have it backwards but does this really increase the resolution? 
Those two would be the same, right?
```
stepper_set_step_interval(stepper, 1);
```

and with the current velocity API:
```
stepper_set_max_velocity(stepper, USEC_PER_SECOND);
```

with `stepper_set_step_interval` you can not go finer, right?

@jilaypandya @andre-stefanov  Aside from that I drafted up the version where we leave the stepper API as is but provide the "escape-hatch" for the step-dir interface:
https://github.com/faxe1008/zephyr/compare/d9a9f9a15cefcb8c311c120163a1ade13be14bec%5E..bcaae12bd652358f600deb1756f17c7cdfb7dc92

PR #83726 (2025-01-15): RFC: Change Stepper API to use step interval
As discussed in the original RFC issue: Change the stepper API to accept the interval to wait at a minimum in between executing consequtive steps. This allows for a more granular control then the previous interface. 

Resolves #83591
closed 2025-01-15 by kartben