zephyrproject-rtos/zephyr #83599: NXP drivers: flash: soc_flash_mcux: CMD_MARGIN_CHECK fails infrequently

DerekSnell, 2025-01-06:
On LPC5500 devices, using `CMD_MARGIN_CHECK` in the `is_area_readable()` can return with an error code, even when the flash is readable and operating as expected.  The `CMD_MARGIN_CHECK` should be removed from `is_area_readable()` in this driver.

From the LPC55S69 datasheet: "As cells age and lose charge, a correctly programmed address will fail this check, while still being able to be read successfully for the remaining duration of the data retention time."

Replicating the issue can be challenging, as the `CMD_MARGIN_CHECK`  returns errors on a small percentage of devices, or after a flash sector has been erased/programmed many times.  But these errors have been seen on custom boards with the LPC55S69.

closed 2025-01-06 by fabiobaltieri

bravl, 2025-02-21:
Hi @DerekSnell,

Could you share the part / page of the LPC55S69 datasheet you mention? I'm currently using an LPC55S69 device where I see the issue described here and would love to have a better understanding of why this occurs.

Thanks!

DerekSnell, 2025-02-21:
Hi @bravl ,
I see this statement was removed from the datasheet.  I also see that NXP has defeatured the `CMD_MARGIN_CHECK`.  That command is not intended for the application level, it is an internal test used by NXP.  The references to `CMD_MARGIN_CHECK` will also be removed from the User Manuals.

Best regards

DerekSnell, 2025-02-21:
Hi @bravl ,
No concern needed.  This command can return false errors on perfectly good flash.  That is the reason the command is defeatured from applications, and removed from this driver.

You can sleep easy, my friend :)  Best regards

bravl, 2025-02-21:
Hi @DerekSnell 

Awesome, thank you for the explanation!

Bram

PR #83603 (2025-01-18): [Backport v4.0-branch] NXP drivers: flash: soc_flash_mcux: remove CMD_MARGIN_CHECK
Backport 88b9cb6efcb18cd91bbcbb440f455724a872c6b9 from #83472.

Fixes #83599
PR #83602 (2025-01-22): [Backport v3.7-branch] NXP drivers: flash: soc_flash_mcux: remove CMD_MARGIN_CHECK
Backport 88b9cb6efcb18cd91bbcbb440f455724a872c6b9 from #83472.

Fixes #83599