zephyrproject-rtos/zephyr #83618: dhcp client is unable to acquire an address.

ashutoshpandey-eaton, 2025-01-07:
**Describe the bug**
dhcp client is unable to acquire an address.

**1**.Our client is able to get the IP from mobile hotspot (acting as WIFI-AP) 
[working_log.log](https://github.com/user-attachments/files/18327552/working_log.log)

**2**.Our client is **not able** to get the IP from any router (acting as WIFI-AP): NOTE: dhcp DISCOVER message is not seen on Wireshark, in this case.
[not_working_log.log](https://github.com/user-attachments/files/18327555/not_working_log.log)

In the non- working case it is seen in the logs [not_working_log.log] dhcp client tries to resend Discover over time however same DISCOVER message in not seen in the Wireshark, only dhcp clinet received ARP packet with print log: "net_core: process_data: (0x24003028): Packet 0x2402d028 discarded by L2"

similar type of issue observed earlier by someone. https://github.com/zephyrproject-rtos/zephyr/issues/66543

**Expected behavior**
dhcp client should get the IP every time.

**Impact**
dhcp client

**Environment **
zephyr OS : 3.6.0
SDK: zephyr-sdk-0.16.5-1
Host board(client): nucleo-h753zi

**What could be the cause (by looking at the logs) , in one case dhcp client is working while in another case its not working??**
I'd appreciate input on how to further debug the issue.

nordicjm, 2025-01-07:
>  zephyr OS : 3.6.0

This is no longer support, please update to 3.7 or 4.0 and check that the issue is still present

carlescufi, 2025-03-05:
Closing due to inactivity. @ashutoshpandey-eaton please reopen if you think this is still an issue.

closed 2025-03-05 by carlescufi