zephyrproject-rtos/zephyr #83619: System Crash in k_mem_map_phys_guard Function

Rrooach, 2025-01-07:
**Describe the bug**

When testing Zephyr with our fuzzing tools, a crash occurs in the k_mem_map_phys_guard function due to an invalid destination (DST) pointer passed to memset.

**To Reproduce**

Modify samples/hello_world to include the following code:

```c
void syz_mem_map_and_subtract_atomic(int mem_size, uint32_t flags, int subtract_value) {
    void *mapped_mem = k_mem_map(mem_size, flags);
    if (mapped_mem != NULL) {
        atomic_t my_atomic = ATOMIC_INIT(subtract_value);
        atomic_sub(&my_atomic, subtract_value);
    }
}
```

Compile and run:

```bash
west build -p always -b qemu_x86 samples/hello_world
qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae -machine q35 -kernel zephyr.elf -nograpic
```

**Observed Behavior:** 
Crash occurs at mmu.c:651 in the k_mem_map_phys_guard function due to an invalid destination pointer for memset.

**Expected Behavior:** 
The function should validate the destination pointer before passing it to memset.

**Logs and console output**

The k_mem_map() finally calls the k_mem_map_phys_guard, and at mmu.c:651, it crashed, so I think there might need a simple check to check if DST is valid before the ```memset```.
```bash
Breakpoint 4, k_mem_map_phys_guard (phys=<optimized out>, size=<optimized out>, flags=<optimized out>,
    is_anon=<optimized out>) at /root/zephyr/zephyrproject/zephyr/kernel/mmu.c:647
647		if (!uninit) {
(gdb) n
651			memset(dst, 0, size);
```

**Environment (please complete the following information):**

OS: Ubuntu 22.04 LTS
Toolchain Zephyr SDK
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/143b14bb4fb779437fdeb8369e42fbb81db4a5fd

**Additional context**

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

Rrooach, 2025-01-07:
After modify samples/hello_world, we should call ```syz_mem_map_and_subtract_atomic(0x180000, 0x0, 0x72e).```

dcpleung, 2025-03-10:
Already fixed in commit 6b3fff3a2f28.

closed 2025-03-10 by dcpleung