zephyrproject-rtos/zephyr #83621: System Crash in cipher_block_op Function

Rrooach, 2025-01-07:
**Describe the bug**

A crash occurs during the encryption operation in the cipher_block_op function due to an uninitialized handler in the ctx->ops structure.

**To Reproduce**

Modify samples/hello_world to include the following code:
```c
void syz_encrypt_then_hash_with_dynamic_memory(uint8_t msg_len, uint8_t key_len, uint8_t hash_len) {
    uint8_t *msg = k_malloc(msg_len);
    uint8_t key[key_len];
    struct cipher_ctx ctx = { .keylen = key_len, .key.bit_stream = key };
    struct cipher_pkt cipher_pkt = { .in_buf = msg, .in_len = msg_len, .out_buf = msg, .out_len = msg_len };

    cipher_block_op(&ctx, &cipher_pkt);
    k_free(msg);
}
```
Compile and run:
```bash
west build -p always -b qemu_x86 samples/hello_world
qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae -machine q35 -kernel zephyr.elf -nographic
```
Call
```c
syz_encrypt_then_hash_with_dynamic_memory(0x101, 0x0, 0x9).
```
**Observed Behavior:**
Crash occurs because ctx->ops.block_crypt_hndlr is NULL.

**Expected Behavior:**
The encryption context should initialize ctx->ops.block_crypt_hndlr before calling the function.

**Logs and console output**

The crash ends up here, but I am not quite sure what going on here :(
```bash
cipher_block_op (pkt=0x11ff80 <z_main_stack+856>, ctx=0x11ff98 <z_main_stack+880>)
    at /root/zephyr/zephyrproject/zephyr/include/zephyr/crypto/crypto.h:259
259		return ctx->ops.block_crypt_hndlr(ctx, pkt);
(gdb) p ctx
$1 = (struct cipher_ctx *) 0x11ff98 <z_main_stack+880>
(gdb) p ctx->ops
$2 = {cipher_mode = 0, {block_crypt_hndlr = 0x0, cbc_crypt_hndlr = 0x0, ctr_crypt_hndlr = 0x0,
    ccm_crypt_hndlr = 0x0, gcm_crypt_hndlr = 0x0}}
(gdb) p pkt
$3 = (struct cipher_pkt *) 0x11ff80 <z_main_stack+856>
```

**Environment (please complete the following information):**

OS: Ubuntu 22.04 LTS
Toolchain Zephyr SDK
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/143b14bb4fb779437fdeb8369e42fbb81db4a5fd

**Additional context**

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

fabiobaltieri, 2025-01-07:
@ceolin can you take a look? unclear to me if the trigger of the issue make sense but worth looking into it

ceolin, 2025-01-08:
Hi,

That is the whole code ? You have to call `cipher_begin_session()` before doing an operation. See https://github.com/zephyrproject-rtos/zephyr/blob/cc042a6a708dfaf84a847d3fc5cb6ff9a09f4492/include/zephyr/crypto/cipher.h#L111-L115

I recommend you to take a look at https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/crypto to see how to use this API.

Please, just ping if you don't find how to do it or if you find any problems :)

Rrooach, 2025-01-08:
Hi, @ceolin; yes, that is the whole code; basically, we are fuzzing the Zephyr, and the code here is our generated "fuzzing driver".

Based on your hint, I understand that we need to call begin_session first during our usage, but to provide such __unsafe usage__, I am wondering if there we need certain checkers to ensure the vitality of  ```cipher_ctx ctx ``` here.

Furthermore, I tried to add cipher_begin_session before cipher_block_op, but seems there are some problems when initializing the ```device *dev.```, below are the code:
```c
    const struct device *dev;

	dev = device_get_binding("CRYPTO_0");
	if (dev == NULL) {
		return;
	}

    if (cipher_begin_session(dev, &ctx, CRYPTO_CIPHER_ALGO_AES,
				 CRYPTO_CIPHER_MODE_ECB,
				 CRYPTO_CIPHER_OP_DECRYPT)) {
		return;
	}
    if (cipher_block_op(&ctx, &cipher_pkt) == 0) {
        hash_pkt.in_buf = msg;
        hash_pkt.in_len = msg_len;
        hash_pkt.out_buf = k_malloc(hash_len);

        hash_compute(&hash_ctx, &hash_pkt);
        k_free(hash_pkt.out_buf);
    }
```

ceolin, 2025-01-15:
> Hi, [@ceolin](https://github.com/ceolin); yes, that is the whole code; basically, we are fuzzing the Zephyr, and the code here is our generated "fuzzing driver".
> 
> Based on your hint, I understand that we need to call begin_session first during our usage, but to provide such **unsafe usage**, I am wondering if there we need certain checkers to ensure the vitality of `cipher_ctx ctx ` here.
> 
> Furthermore, I tried to add cipher_begin_session before cipher_block_op, but seems there are some problems when initializing the `device *dev.`, below are the code:
> 
>     const struct device *dev;
> 
> 	dev = device_get_binding("CRYPTO_0");
> 	if (dev == NULL) {
> 		return;
> 	}
> 
>     if (cipher_begin_session(dev, &ctx, CRYPTO_CIPHER_ALGO_AES,
> 				 CRYPTO_CIPHER_MODE_ECB,
> 				 CRYPTO_CIPHER_OP_DECRYPT)) {
> 		return;
> 	}
>     if (cipher_block_op(&ctx, &cipher_pkt) == 0) {
>         hash_pkt.in_buf = msg;
>         hash_pkt.in_len = msg_len;
>         hash_pkt.out_buf = k_malloc(hash_len);
> 
>         hash_compute(&hash_ctx, &hash_pkt);
>         k_free(hash_pkt.out_buf);
>     }

Hi @Rrooach can you provide me which build options are you enabling ? Regarding the **usafe usage** there are some historical context why several APIs on Zephyr does not check parameters. The main reason is that the application is assumed to be trusted, the exception is user threads. Obviously parameters that come from untreated sources (external data) must be checked.

ceolin, 2025-01-15:
@Rrooach btw, I would love to know more about the fuzzing and possibly help you with that.

Rrooach, 2025-01-16:
Hi, @ceolin, thanks for your kind reply!
> [@Rrooach](https://github.com/Rrooach) can you provide me which build options are you enabling ? 

This is my build configurations, I try to enable comment ones, but still failed to build the project
``` c
CONFIG_OSDP_SC_ENABLED=y
CONFIG_IPC_SERVICE=y
CONFIG_IPC_SERVICE_BACKEND_RPMSG=y
CONFIG_HEAP_MEM_POOL_SIZE=1024
CONFIG_JSON_LIBRARY=y
CONFIG_PIPES=y
CONFIG_SYS_HEAP_RUNTIME_STATS=y # Enable runtime stats for heap
CONFIG_SYS_HEAP_STRESS=y

# CONFIG_OSDP=y
# CONFIG_CRYPTO=y
# CONFIG_CRYPTO_AES=y
# CONFIG_CRYPTO_MODE_ECB=y
# CONFIG_CRYPTO_DRIVERS=y
# CONFIG_OSDP_CRYPTO_DRV_NAME="CRYPTO_0"
```
> [@Rrooach](https://github.com/Rrooach) btw, I would love to know more about the fuzzing and possibly help you with that.

We adapt Syzkaller to support embedded OS fuzzing on diverse architecture, including Zephyr; as for Syzkaller, we use the pseudo function as test input, where the pseudo function contains a set of operations as of the reproduce code above.
As you might notice here, we do have trouble in preparing high-quality input cases, which may contain such API misuse, and we are, as of now, trying our best to improve such issues.

You can ask me any detailed questions about the fuzzing! 

best regrads,

ceolin, 2025-01-29:
> Hi, [@ceolin](https://github.com/ceolin), thanks for your kind reply!
> 
> > [@Rrooach](https://github.com/Rrooach) can you provide me which build options are you enabling ?
> 
> This is my build configurations, I try to enable comment ones, but still failed to build the project
> 
> CONFIG_OSDP_SC_ENABLED=y
> CONFIG_IPC_SERVICE=y
> CONFIG_IPC_SERVICE_BACKEND_RPMSG=y
> CONFIG_HEAP_MEM_POOL_SIZE=1024
> CONFIG_JSON_LIBRARY=y
> CONFIG_PIPES=y
> CONFIG_SYS_HEAP_RUNTIME_STATS=y # Enable runtime stats for heap
> CONFIG_SYS_HEAP_STRESS=y
> 
> # CONFIG_OSDP=y
> # CONFIG_CRYPTO=y
> # CONFIG_CRYPTO_AES=y
> # CONFIG_CRYPTO_MODE_ECB=y
> # CONFIG_CRYPTO_DRIVERS=y
> # CONFIG_OSDP_CRYPTO_DRV_NAME="CRYPTO_0"
>  > [@Rrooach](https://github.com/Rrooach) btw, I would love to know more about the fuzzing and possibly help you with that.
> 
> We adapt Syzkaller to support embedded OS fuzzing on diverse architecture, including Zephyr; as for Syzkaller, we use the pseudo function as test input, where the pseudo function contains a set of operations as of the reproduce code above. As you might notice here, we do have trouble in preparing high-quality input cases, which may contain such API misuse, and we are, as of now, trying our best to improve such issues.

That is awesome, what is your environment ? how are you feeding input into zephyr ? AFAIR syzkaller has an agent that is feed by the fuzzer with input that to execute test cases. How are you doing on Zephyr ?

Are you using which hypervisor ?

I think to have something productive you should target zephyr userspace / syzcall or APIs that handle external / untrusted data because a lot of internal APIs have no checks.

> 
> You can ask me any detailed questions about the fuzzing!
> 
> best regrads,

Thanks a lot for doing it :)

closed 2025-04-14 by github-actions[bot]