zephyrproject-rtos/zephyr #83623: System Crash in k_msgq_get Function

Rrooach, 2025-01-07:
**Describe the bug**

A crash occurs in the k_msgq_get function due to invalid or uninitialized msgq->wait_q.

**To Reproduce**

Modify samples/hello_world to include the following code:
```c
void syz_msgq_send_receive_loop(size_t msg_size, uint32_t max_msgs, int num_iterations) {
    struct k_msgq msgq;
    char buffer[msg_size * max_msgs];
    k_msgq_init(&msgq, buffer, msg_size, max_msgs);

    for (int i = 0; i < num_iterations; i++) {
        int msg = i;
        k_msgq_put(&msgq, &msg, K_NO_WAIT);

        int received_msg;
        k_msgq_get(&msgq, &received_msg, K_NO_WAIT);
    }
}
```
Compile and run:
```bash
west build -p always -b qemu_x86 samples/hello_world
qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae -machine q35 -kernel zephyr.elf -nographic
```
Call
```c
syz_msgq_send_receive_loop(0x7ff, 0x2, 0x10001);
```
**Observed Behavior:**
Crash occurs when attempting to access msgq->wait_q.
**Expected Behavior:**
The function should validate msgq->wait_q before dereferencing.

**Logs and console output**

It ends up in z_impl_k_msgq_get, where failed to check the validity of msgq->wait_q, and crash the program.
```bash
0x0010468d in z_impl_k_msgq_get (msgq=<optimized out>, data=<optimized out>, timeout=...)
    at /root/zephyr/zephyrproject/zephyr/kernel/msg_q.c:217
217	{
(gdb) n
224		key = k_spin_lock(&msgq->lock);
(gdb) n
228		if (msgq->used_msgs > 0U) {
(gdb) n
230			(void)memcpy((char *)data, msgq->read_ptr, msgq->msg_size);
(gdb) n
231			msgq->read_ptr += msgq->msg_size;
(gdb) n
232			if (msgq->read_ptr == msgq->buffer_end) {
(gdb) n
235			msgq->used_msgs--;
(gdb) n
238			pending_thread = z_unpend_first_thread(&msgq->wait_q);
(gdb) n
```

**Environment (please complete the following information):**

OS: Ubuntu 22.04 LTS
Toolchain Zephyr SDK
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/143b14bb4fb779437fdeb8369e42fbb81db4a5fd

**Additional context**

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

andyross, 2025-01-07:
Agreed, this smells like a routine stack overflow absent more data.  Can you post the fuzzer/asan/msan output?

As far as variable length arrays: we actually do cheat in a handful of places and hide alloca()-style allocations behind a CONFIG_MISRA_SANE kconfig for situations where the value is very high and the code is pretentiously considered high quality and safe.  But yes, in general they aren't allowed.  But they do work in app code as long as you don't blow up your stack.  (And that risk, obviously, is precisely **why** they aren't generally allowed)

Rrooach, 2025-01-07:
> Can you post the fuzzer/asan/msan output?

Hi, unfortunately, the program simply reboots itself, so there is not much valuable output, but here is the complete gdb log I just dump: https://pastebin.com/tXkYwnFf.

As we can find, it seems to have crashed at: ```rb.h:134```.

andyross, 2025-01-07:
Again, can you please try this with a static buffer (or a k_malloc'd one) and see if you can reproduce?

peter-mitsis, 2025-01-07:
One other data point to support the semll of a smashed stack--building the hello_world sample application, I observe that 4094 bytes are getting carved for the buffer out of a stack size that is configured for only 1024 bytes.

kartben, 2025-01-07:
> One other data point to support the smell of a smashed stack--building the hello_world sample application, I observe that 4094 bytes are getting carved for the buffer out of a stack size that is configured for only 1024 bytes.

Yep but like I mentioned in my reply from earlier it looks like once the stack is set to be large enough, the snippet in https://github.com/zephyrproject-rtos/zephyr/issues/83623#issuecomment-2575565328 works while the VLA equivalent from the original comment doesn't seem seem to.

Rrooach, 2025-01-08:
> Again, can you please try this with a static buffer (or a k_malloc'd one) and see if you can reproduce?

Hi, I change the code into ```static char buffer[SIZE]```, where SIZE is ```#define SIZE (0x7ff * 0x2)```, the error keep happening: https://pastebin.com/tNHdihCP

andyross, 2025-01-08:
OK, thanks.  Indeed, I'm able to reproduce too, and was almost fooled[1].  But it's user error.  :)

What's happening here is that you initialized a msgq with a message size of 2047 bytes.  But then you call `k_msgq_get()` on your `received_msg`, **which is a 4-byte integer**.  The renaming 2043 bytes get dropped onto your stack frame and that of your callers, blowing up the world.

So I guess we were right about the stack overflow smell, but not about the proximate cause.  I'm going to close this, feel free to reopen if I've missed something.

[1] OK, fine, I was totally hoodwinked.  I went as far as getting it to fail in native_sim (thanks @aescobar), where I can use CONFIG_ASAN=y, and started digging through the resulting stack overflow before I realized what the problem was.

closed 2025-01-08 by andyross