zephyrproject-rtos/zephyr #83624: System Crash in json_obj_encode Function

Rrooach, 2025-01-07:
**Describe the bug**

A crash occurs in the json_obj_encode function due to invalid memory access when encoding JSON objects.

**To Reproduce**

Modify samples/hello_world to include the following code:
```c
void syz_json_conditional_object_operations(uint32_t json_len, uint32_t descr_len) {
    char json[json_len] = "{\"value\": 100}";
    struct json_obj json_obj;
    struct my_struct { int value; } obj;
    struct json_obj_descr descr[descr_len];
    descr[0].field_name = "value";
    descr[0].type = JSON_TOK_NUMBER;
    descr[0].offset = offsetof(struct my_struct, value);

    json_obj_parse(json, json_len, descr, descr_len, &obj);
    json_obj_encode(descr, descr_len, &obj, NULL, NULL);
}
```

Compile and run:

```bash
west build -p always -b qemu_x86 samples/hello_world
qemu-system-i386 -m 32 -cpu qemu32,+nx,+pae -machine q35 -kernel zephyr.elf -nographic
```
Call
```c
syz_json_conditional_object_operations(0x5, 0x0);
```
**Observed Behavior:**
Crash occurs due to an invalid memory access in the json_obj_encode function.
**Expected Behavior:**
The function should validate all pointers and ensure proper memory handling before encoding.

**Logs and console output**

There seems also to has a illegal memory access at
```c
int json_obj_encode(const struct json_obj_descr *descr, size_t descr_len,
		    const void *val, json_append_bytes_t append_bytes,
		    void *data)
{
	size_t i;
	int ret;

	ret = append_bytes("{", 1, data);
	if (ret < 0) {
		return ret;
	}

	for (i = 0; i < descr_len; i++) {
		ret = str_encode((const char **)&descr[i].field_name,
				 append_bytes, data);
		if (ret < 0) {
			return ret;
		}

		ret = append_bytes(":", 1, data);
		if (ret < 0) {
			return ret;
		}
		
```

**Environment (please complete the following information):**

OS: Ubuntu 22.04 LTS
Toolchain Zephyr SDK
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/143b14bb4fb779437fdeb8369e42fbb81db4a5fd

**Additional context**

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

fabiobaltieri, 2025-01-07:
@Rrooach is this a regression or an existing bug? if it's a regression, have you been able to bisect to a specific commit?

Rrooach, 2025-01-08:
@fabiobaltieri, it appears to me that this is not a regression but is an existing bug, with git blame:
d66069d051ee lib/json/json.c  (Leandro Pereira       2017-09-05 16:24:47 -0700 1113) int json_obj_encode(
Seems the entire function was introduced in 2017 and never changed

fabiobaltieri, 2025-01-08:
Ok so this is trying to call `append_bytes` that you explicitly passed as `NULL`, your assessment is:

> The function should validate all pointers and ensure proper memory handling before encoding.

I mean, sure, feel free to open a pull request to check that the argument is not NULL and error out and then this specific case would not crash, but I don't really see much value in that, you could still pass an invalid pointer and the system would happily dereference it and crash anyway, wouldn't really be too different than passing an invalid data length, your test scenario obviously misusing the API, there's only so much you can really validate in this context.

closed 2025-03-24 by github-actions[bot]