zephyrproject-rtos/zephyr #83662: net/ethernet/ethernet.c: Handle ieee 802.1q (vlan) headers with id 0 (even if CONFIG_VLAN is not enabled)

clamattia, 2025-01-07:
**Is your enhancement proposal related to a problem? Please describe.**
Handle ieee 802.1q headers with vlan id 0 correctly, by forwarding them to the native default interface. This is as far as I could gather the intended use for the vlan-id 0. This is also required for some industrial protocols.

**Describe the solution you'd like**
Ethernet Frames tagged with id `0` should be stripped or in other words forwarded to the native interface, even if the `CONFIG_VLAN` is not enabled.
I am experimenting with something like:
```diff
+       if (type == NET_ETH_PTYPE_VLAN) {
+               struct net_eth_vlan_hdr *hdr_vlan = (struct net_eth_vlan_hdr *)NET_ETH_HDR(pkt);
+               if (ntohs(hdr_vlan->vlan.tci) & 0x0fff == 0) {
+                       type = ntohs(hdr_vlan->type);
+                       hdr_len = sizeof(struct net_eth_vlan_hdr);
+               }
+       }
```

**Describe alternatives you've considered**
None, as this is as far as I know the standardized behavior. If someone can confirm, this would be great.

clamattia, 2025-01-08:
Please also note, that this will have to be adjusted:
`ethernet.h`
```C
#if defined(CONFIG_NET_VLAN)
#define _NET_ETH_MAX_HDR_SIZE		(sizeof(struct net_eth_vlan_hdr))
#else
#define _NET_ETH_MAX_HDR_SIZE		(sizeof(struct net_eth_hdr))
#endif
```

It assumes, that the drivers need not be able to receive vlan-tagged frames except, when `CONFIG_NET_VLAN` is enabled.
See related issue in stm32 driver: https://github.com/zephyrproject-rtos/zephyr/issues/83277

jukkar, 2025-01-09:
> It assumes, that the drivers need not be able to receive vlan-tagged frames except, when `CONFIG_NET_VLAN` is enabled.

I think this is a good assumption, after all if VLAN is not enabled the driver can just drop VLAN frames.

jukkar, 2025-01-09:
> Handle ieee 802.1q headers with vlan id 0 correctly, by forwarding them to the native default interface. This is as far as I could gather the intended use for the vlan-id 0. This is also required for some industrial protocols.

This is a good point, the VLAN 0 should be passed to native interface. Would you be able to send a PR for it?

clamattia, 2025-01-09:
> This is a good point, the VLAN 0 should be passed to native interface. Would you be able to send a PR for it?

Yes, I have prepared a PR. Please have a look.

clamattia, 2025-01-09:
> > It assumes, that the drivers need not be able to receive vlan-tagged frames except, when `CONFIG_NET_VLAN` is enabled.
> 
> I think this is a good assumption, after all if VLAN is not enabled the driver can just drop VLAN frames.

This assumption is no longer valid, if the proposal of this issue is implemented. See proposal in PR.

clamattia, 2025-01-23:
Completed by: https://github.com/zephyrproject-rtos/zephyr/pull/83734
Enhancement tracked by: https://github.com/zephyrproject-rtos/zephyr/issues/84023