zephyrproject-rtos/zephyr #83663: [RFC] Bouffalolab Vendor Support

VynDragon, 2025-01-07:
# Introduction

There is community interest to add support for Bouffalolab SoCs in Zephyr. This RFC was opened to serve as an umbrella to help community and Zephyr members know the initiatives.

## Problem description

There is a pull request for the initial support: https://github.com/zephyrproject-rtos/zephyr/pull/78795 .
Work is ongoing for more.

## Proposed changes

Give an overview about all pieces required to add Bouffalolab SoCs in Zephyr.

# Detailed RFC

In this section of the document the target audience is the dev team. Upon reading this section each engineer should have a rather clear picture of what needs to be done in order to implement the described feature.

## Vendor Documentation and SDK

Bouffalolab provides a open source SDK which can be used for reference, especially for peripherals for which the documentation is incomplete or absent.
There is some significant amount of documentation available in https://github.com/bouffalolab/bl_docs .

## HAL

There is no HAL per se for Bouffalolab devices, the only materials from the vendor that are directly used are the headers providing peripheral register lists. These headers are Apache 2 and we are awaiting Bouffalolab confirmation for their usage in a zephyr HAL repository.

The current expected HAL repository, which is located currently at https://github.com/nandojve/hal_bouffalolab , is as follow:

```
|---include
	|---bouffalolab
		|---bl60x (headers specific to SoC serie)
		|---bl61x
		|---bl70x
		|---...
		|---common (common peripheral headers)
	|---zephyr
		|---dt-bindings
			|---pinctrl (pinctrl headers for zephyr)
				|---bflb-common-pinctrl.h (common pinctrl header)
				|---bl602x-pinctrl.h (SoC package specific pinctrl header)
				|---bl604x-pinctrl.h
				|---bl60x-pinctrl.h (SoC specific pinctrl header)
				|---...
|---blob (folder for blobs for RF peripherals)
	|---bl60x
		|---hal (blob HAL)
		|---include (blob header)
		|---lib (blob libraries)
	|---bl61x
	|---...
|---pinconfigs (YAML pin data from which pinctrl.h files are generated)
	|---README.md
	|---bl60x.yml
	|---bl61x.yml
	|---bl70x.yml
	|---...
|---scripts (scripts and scripts-related files, notably pinctrl.h generator)
	|---bflbpinctrl.py
	|---...
|---zephyr
	|---module.yml
|---LICENSE
|---README.md
```

## CPUs

Bouffalolab SoCs use various CPU designs: from SiFive (SiFive E24 on BL60x, BL70x and BL70xL), and from Xuantie T-HEAD (E907 on BL61x, and C906, E907, E902 on BL808 and BL606P).
SiFive E24 Reference Manual is available from sources such as starfive or alibaba. Xuantie CPUs reference manuals are available directly on Xuantie's website after registration.

### Caches

Xuantie CPUs provide instruction caches and branch prediction, configured via the XTHeadCMO RISCV extension and various CSR.
The SiFive E24 used in BL60x and BL70x do not provide a cache, but Bouffalolab uses a peripheral, called L1C (Level 1 Cache) to use part of the ITCM from E24 as a fast instruction cache.
BL60x, BL70x, and BL61x rely heavily on these caching mechanisms due to slow flash, with 16KB to 48KB dedicated to instruction caching on the fastest memory in the device, it is recommended to pay close attention to it.
Examples of using those mechanisms and their pitfalls within zephyr are available in the initialization code, the flash driver, and the BL61x USB driver. It is very critical in drivers interacting with the flash or DMA.

## Availability for developers

Devboards and modules for BL60x and BL61x are easily available from AI-Thinker, Pine64, and Sipeed via typical channels (Amazon, Aliexpress, LCSC...).
Devboards and modules for BL70x are sometimes available from XT/DOIT. Some noname boards are also available. It is also embbeded in some Sipeed products, such as the RV-debugger.
Devboards and modules for BL808 are available from Sipeed and Pine64.
BL606P is so far only available in the Pine64 PineVox.
TG7100C is a alternative packaging of BL602, available from AI-Thinker via typical channels.

## Peripherals Drivers

Most peripherals are either the same or very similar between SoCs. It is recommended to implement drivers in a way that make them usable for all SoCs.

# Task List

## Essentials

Essentials are understood to be:
- HAL repository
- CPU support
BL60x, BL70x, and BL61x can use riscv-privileged arch in CLIC vectored mode, and so all the basic features are supported out the box by Zephyr once the right settings are set.
- SoC initialization
- USART driver
- Interrupts
- GPIO driver
- PinCtrl driver
- System Timer driver

This pull request contains all the essentials for BL60x serie:
https://github.com/zephyrproject-rtos/zephyr/pull/78795

### Future

- [ ] [72102](https://github.com/zephyrproject-rtos/zephyr/pull/72102) Clock driver for BL60x
- [ ] [72102](https://github.com/zephyrproject-rtos/zephyr/pull/72102) Clock driver for BL61x
- [ ] [72102](https://github.com/zephyrproject-rtos/zephyr/pull/72102) Clock driver for BL70x

## SoCs

- [x] BL60x: BL602, BL604
https://github.com/zephyrproject-rtos/zephyr/pull/78795
- [ ] BL70x: BL702, BL706
Complete: https://github.com/VynDragon/zephyr/tree/bflb_upstreaming_efuse
- [ ] BL61x: BL616, BL618
Mostly Complete (no AUPLL init): https://github.com/VynDragon/zephyr/tree/bflb/bl61x
https://github.com/VynDragon/zephyr/tree/bflb/bl61x_usb
- [ ] BL70xL: BL702L, BL704L
- [ ] BL808
- [ ] BL606P

## Boards

- [x] bl604e_iot_dvk (BL604)
https://github.com/zephyrproject-rtos/zephyr/pull/78795
- [ ] dt_bl10_devkit (BL602)
- [ ] dt_XT-ZB1-DevKit (BL702)
- [ ] ai_WB2-12F-Kit (BL602)
- [ ] ai_WB2-32S-Kit (BL602)
- [ ] ai_M62-12F-Kit (BL616)
- [ ] ai_M62-32S-Kit (BL616)
- [ ] ai_M61-32S-Kit (BL618)
- [ ] ai_TG-12F-Kit (TG7100C)
- [ ] sipeed_M0P-Dock (BL618)
- [ ] sipeed_M0s-Dock (BL616)
- [ ] sipeed_M0sense (BL702)
- [ ] sipeed_M1s-Dock (BL808)
- [ ] pine64_PineVox (BL606P)
- [ ] pine64_Ox64 (BL808)
- [ ] pine64_PineCone (BL602)
- [ ] others_BL702S-Blue (BL702) https://aliexpress.com/item/1005005927910587.html keyword 'BL702S'
- [ ] others_BL616-Blue (BL616) https://aliexpress.com/item/1005006916672346.html keyword 'BL616'

## Drivers

Drivers are 'complete' if they support all currently complete SoCs.

- [ ] ADC
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x
- [ ] Audio ADC
- [ ] CAM (DVP)
- [ ] Audio DAC
- [ ] DAC
https://github.com/VynDragon/zephyr/tree/bflb/bl61x but for BL61x
- [ ] DBI
- [ ] DMA
- [ ] Ethernet EMAC
- [ ] Efuses (dependency)
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x
- [ ] Flash Single Bank
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x
- [ ] Flash Multi Banks
- [ ] GPIO FIFO Mode
- [ ] I2C
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x
- [ ] I2S
- [ ] IR
- [ ] JDEC
- [ ] KeyScan
- [ ] LINbus
- [ ] LPM (Low Power Modes)
Work Started in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x_usb
- [ ] MJPEG
- [ ] PIR Sensor Controller
- [ ] PMP
- [ ] PSRAM
- [ ] PWM
- [ ] QDEC (rotary encoders)
- [ ] Security Engine/Accelerator
- [ ] SDH
- [ ] SDIO
- [ ] SPI
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x
- [ ] Timers (!= riscv system timer)
- [ ] USB Device BL61x
Complete in: https://github.com/VynDragon/zephyr/tree/bflb/bl61x_usb
- [ ] USB Host BL61x
- [ ] USB Device/Host BL808/606P
- [ ] USB Device BL70x
- [ ] VENC
- [ ] BLE BL60x
- [ ] BLE BL70x
- [ ] BLE BL70xL
- [ ] Bluetooth/BLE BL61x
- [ ] Bluetooth/BLE BL808/606P
- [ ] Wifi Bl60x 
- [ ] Wifi BL61x
- [ ] Wifi BL808/BL606P
- [ ] 802.15.4 BL61x
- [ ] 802.15.4 BL70x
- [ ] 802.15.4 BL70xL
- [ ] 802.15.4 BL808/BL606P

kristoftjonck, 2025-02-14:
Great to see this moving forward! I noticed that the Iso11898 (CAN) driver is missing from the peripheral list. I’m not sure which Bouffalolab modules support CAN, but I know the BL61x does. I also spotted a clock initialization for CAN in the Bouffalo SDK for the BL808. It would be great to see support for this added!

gamelaster, 2025-02-14:
@kristoftjonck BL61X for sure supports CAN (but as far as I remember, external transceiver is required). I think @VynDragon will add it into the list.

VynDragon, 2025-02-14:
> Great to see this moving forward! I noticed that the Iso11898 (CAN) driver is missing from the peripheral list. I’m not sure which Bouffalolab modules support CAN, but I know the BL61x does. I also [...]

Hi, the thing with the CAN peripheral is that it's been removed from the docs intentionally, and so has the code. It is still available from alternative sources or Qualcomm. It was a subject of discussion recently but I haven't re-added it to the list, which I will do now.
Regardless, I can say now already it is unlikely I will produce a driver for it, due to lack of test hardware and personal interest, but someone else might do it. There is also the option of contracting my company (or someone else familiar with bflb hardware like gamelaster who is an expert) and providing test hardware if you are very interested in this feature.

kristoftjonck, 2025-02-17:
@VynDragon @gamelaster Thanks for the insight! Do you have any additional details on the intentional removal of the CAN peripheral in the documentation? Were there any specific issues, and do you have direct contact with Buffalo Labs? We'd love to gain more detailed insights. Can I reach out using the email address listed on your website?

VynDragon, 2025-02-17:
>  intentional removal of the CAN peripheral in the documentation
Not much, we suspect a patent or copyright issue.
> direct contact with Buffalo Labs
Not me personally, but we (the 'working group' I guess?) hope to strengthen communications once initial support is merged.
> Can I reach out using the email address listed on your website?
Idk if you mean me or gamelaster but mine is also used for commits and on my website so you are fully free to contact me for any relevant reasons.

gamelaster, 2025-02-17:
@kristoftjonck as VynDragon said 😁 

> Can I reach out using the email address listed on your website?

If that was targeted to me, then yeah, for sure, feel free to use my email from my personal website, or the one used in git commits.

VynDragon, 2025-02-25:
@profound1987 Continuing from PR thread

> LHAL
Current LHAL is not suitable for use with Zephyr, unless there is a very different version than the one at https://github.com/bouffalolab/bouffalo_sdk/tree/master that you can share with us. Please understand we have studied the question of using SDK/LHAL directly before making the decision to not use it.

> release_v1.4.5
This is a temporary version (please see the request at https://github.com/zephyrproject-rtos/zephyr/issues/84174), future updates remove the SDK entirely and rely only on the header files for each platforms.

We would be happy to setup a meeting with you, please be aware multiple people are working on this project.

Hiihtaja77, 2026-04-15:
Hi @VynDragon !

What kind of memory model expected to use with BL618 Socs ?
1. PSRAM have own address space and network and display buffers ( and whatever ) can be allocated in it by using special tag.
2. Code from Flash memory can be shadowed to PSRAM to execution and remain part is used as native SRAM memory ( PSRAM is wasted a bit but it might be better performance).

Does both methods might be supported at some point ?

VynDragon, 2026-04-15:
Both methods should work with configuration yes.
The issue with PSRAM relocation is that on zephyr it is not possible to initialize the PSRAM before doing the relocation (this has been a subject of dicussion and if you look you will see i have interacted with the relevant PRs because of similar ideas), but we are planning on having MCUBoot as some point (and it would allow working around the header generation issues for QCC74x) and that will be able to do it. I have succeeded in getting PoC MCUBoot ram boot on bl70x already, but for a different usage. However XIP might be faster or as fast as running from PSRAM in most cases, especially with properly configured flash (everything defaults to DO SPI mode at XCLK speeds which is 40Mhz for BL61x and QCC74x usually), i have continuous read QIO working at 80Mhz for M61 ALL (bl618m65) in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/107223 (it also allows you to add more PSRAM with 4 pins (BL618 PSRAM uses 8 pins)... if you can manage to make it play nice, i have only managed to get MRAM to play nice on the test setups so far)

Also i've not gotten QCC744m to drive its flash faster than QO at 40MHz, so if you want to investigate that, this PR is what you need.

VynDragon, 2026-06-06:
@Hiihtaja77 https://github.com/zephyrproject-rtos/zephyr/pull/110672 Do check this out if you have hardware to test