zephyrproject-rtos/zephyr #83666: Unable to build MCUBoot sample with nucleo_h753zi board

warlock31415, 2025-01-07:
**Describe the bug**

Build the `with_mcuboot` sample with the nucleoh753zi board leads to a `'rom_start' will not fit in region 'FLASH'` error
- Host platform : Ubuntu 22.04.4 LTS
- Target platform: Nucleo H753ZI

**To Reproduce**

```
west init && west update
west build -p -b nucleo_h753zi --sysbuild zephyr/samples/sysbuild/with_mcuboot 
```

**Expected behavior**

**Impact**

- Simple sample should not take up 256KBit of space that the slot1/slot2 device tree specifies. Unsure if what is happening with the linker behind the scenes

**Logs and console output**

```-- west build: making build dir /tmp/tmp.evEMcodxno/build pristine
-- west build: generating a build system
-- Found Python3: /tmp/tmp.evEMcodxno/.venv/bin/python (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/pk/.cache/zephyr
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nucleo_h753zi, qualifiers: stm32h753xx
Parsing /tmp/tmp.evEMcodxno/zephyr/share/sysbuild/Kconfig
Loaded configuration '/tmp/tmp.evEMcodxno/build/_sysbuild/empty.conf'
Merged configuration '/tmp/tmp.evEMcodxno/zephyr/samples/sysbuild/with_mcuboot/sysbuild.conf'
Configuration saved to '/tmp/tmp.evEMcodxno/build/zephyr/.config'
Kconfig header saved to '/tmp/tmp.evEMcodxno/build/_sysbuild/autoconf.h'
-- 
   *****************************
   * Running CMake for mcuboot *
   *****************************

-- Application: /tmp/tmp.evEMcodxno/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.22.1
-- Found Python3: /tmp/tmp.evEMcodxno/.venv/bin/python (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/pk/.cache/zephyr
-- Zephyr version: 4.0.99 (/tmp/tmp.evEMcodxno/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nucleo_h753zi, qualifiers: stm32h753xx
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0)
-- Found Dtc: /tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /tmp/tmp.evEMcodxno/zephyr/boards/st/nucleo_h753zi/nucleo_h753zi.dts
-- Found devicetree overlay: /tmp/tmp.evEMcodxno/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: /tmp/tmp.evEMcodxno/build/mcuboot/zephyr/zephyr.dts
-- Generated pickled edt: /tmp/tmp.evEMcodxno/build/mcuboot/zephyr/edt.pickle
-- Generated zephyr.dts: /tmp/tmp.evEMcodxno/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /tmp/tmp.evEMcodxno/build/mcuboot/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /tmp/tmp.evEMcodxno/build/mcuboot/zephyr/dts.cmake
Parsing /tmp/tmp.evEMcodxno/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/tmp/tmp.evEMcodxno/zephyr/boards/st/nucleo_h753zi/nucleo_h753zi_defconfig'
Merged configuration '/tmp/tmp.evEMcodxno/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration '/tmp/tmp.evEMcodxno/zephyr/samples/sysbuild/with_mcuboot/sysbuild/mcuboot.conf'
Merged configuration '/tmp/tmp.evEMcodxno/build/mcuboot/zephyr/.config.sysbuild'
Configuration saved to '/tmp/tmp.evEMcodxno/build/mcuboot/zephyr/.config'
Kconfig header saved to '/tmp/tmp.evEMcodxno/build/mcuboot/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.evEMcodxno/build/mcuboot
[... 485 lines cut ...]
[131/139] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work.c.obj
[132/139] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[133/139] Linking C static library zephyr/kernel/libkernel.a
[134/139] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /tmp/tmp.evEMcodxno/build/with_mcuboot/zephyr/zephyr_pre0.map 
: && /tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/tmp/tmp.evEMcodxno/build/with_mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/tmp/tmp.evEMcodxno/build/with_mcuboot/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF -L"/tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp" -lc -lgcc && cd /tmp/tmp.evEMcodxno/build/with_mcuboot/zephyr && /usr/bin/cmake -E true
/tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `rom_start' will not fit in region `FLASH'
/tmp/tmp.evEMcodxno/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 17534 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[13/16] Completed 'mcuboot'
FAILED: _sysbuild/sysbuild/images/with_mcuboot-prefix/src/with_mcuboot-stamp/with_mcuboot-build /tmp/tmp.evEMcodxno/build/_sysbuild/sysbuild/images/with_mcuboot-prefix/src/with_mcuboot-stamp/with_mcuboot-build 
cd /tmp/tmp.evEMcodxno/build/with_mcuboot && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Ubuntu 22.04.4 LTS
 - Toolchain: Zephyr SDK
 - Commit SHA: main - `9d0627a4bcbfd16f6f44d2bfb597057d725ee13e`

**Additional context**

This is was noticed while trying to build for a stm32h755xixh (custom board)

FRASTM, 2025-01-10:
@warlock31415 did you try to change the partition of the flash0 ?
When increasing the size of the `slot0_partition` for example

warlock31415, 2025-01-10:
@FRASTM I'll have to delete both the flash and the storage partitions to give the image slots more sectors. Both the image slots are already at 256K
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* 128KB for bootloader */
		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(128)>;
			read-only;
		};

		/* storage: 128KB for settings */
		storage_partition: partition@20000 {
			label = "storage";
			reg = <0x00020000 DT_SIZE_K(128)>;
		};

		/* application image slot: 256KB */
		slot0_partition: partition@40000 {
			label = "image-0";
			reg = <0x00040000 DT_SIZE_K(256)>;
		};

		/* backup slot: 256KB */
		slot1_partition: partition@80000 {
			label = "image-1";
			reg = <0x00080000 DT_SIZE_K(256)>;
		};

		/* swap slot: 128KB */
		scratch_partition: partition@c0000 {
			label = "image-scratch";
			reg = <0x000c0000 DT_SIZE_K(128)>;
		};

	};
```

This is the default dts for the board too

FRASTM, 2025-01-13:
However, setting the size of the slot0_partition to DT_SIZE_K(384)  (and slot1_partition too) makes the build successful.

erwango, 2025-01-15:
> But now i don't have a scratch partition...

Scratch partition is not mandatory

closed 2025-03-31 by github-actions[bot]