zephyrproject-rtos/zephyr #83671: Potential crashes or undefined behavior due to unchecked slen value of cbprintf_package

PwnVerse, 2025-01-08:
## Description

It is possible for `cbprintf_package` to return a malformed package if the return value is not checked and `cbprintf_package_convert` is called with this malformed input. Failing to validate the return value could result in crashes or undefined behavior due to invalid memory access when attempting to process the malformed package.

```c
const char *fmt = *(const char **)(buf + sizeof(void *));
```

This may lead to an invalid memory access if `in_packaged` is malformed or points to a malformed location returned by `cbprintf_package` (if it's returned length is not validated by caller).

## Steps to reproduce:

1. Pass a malformed or corrupted `in_packaged` pointer to the `cbprintf_package_convert` function.
2. Trigger the execution of the line that accesses `*(const char **)(buf + sizeof(void *))`.

## Suggested Fix

Adding checks [here](https://github.com/zephyrproject-rtos/zephyr/blob/307eca452e5f7d4678f6bc821928f75656186d5f/lib/os/cbprintf_packaged.c#L931C11-L931C17) in the library call to ensure that `slen` is positive can certainly help.

kartben, 2025-01-14:
@PwnVerse feel free to submit a PR with your suggested fix, it would be much appreciated!

nordic-krch, 2025-01-15:
Returned values are there to be checked.  Maybe assert would be enough?

closed 2025-03-31 by github-actions[bot]