zephyrproject-rtos/zephyr #83689: Adding support of googletest / googlemock

anobli, 2025-01-08:
## Introduction

### Problem description

### Proposed change

### Proposed change (Detailed)

### Dependencies

## Alternatives

JeffWelder, 2025-04-20:
I flipped you a DM on discord about my attempts to support showing these tests in the VSCode C++ Testmate extension. 😀

kartben, 2025-04-20:
@anobli Twister already has a gTest harness, have you tried it?

anobli, 2025-04-23:
@kartben Yes, I have been using the gTest harness, and it was working. The only issue I had was with the timings printed by gTest (the harness doesn't handle them).
I should have precised it, but, the the issue not really with gTest but really with gmok.
IIRC, gmock only build for native target, but, it doesn't build for any qemu or real target.
I have opened a PR for it, but, it take a while:
https://github.com/google/googletest/pull/4692

kartben, 2025-04-23:
@yperess might be able to help

royyandzakiy, 2026-02-08:
Hi @anobli, Sorry this might be out of topic. I'm currently trying to run gtest on my zephyr setup for native_sim. Do you have any examples with this or similair setup?