zephyrproject-rtos/zephyr #83701: Add support for building and running samples and tests in CI that use deprecated features

carlescufi, 2025-01-08:
**Is your enhancement proposal related to a problem? Please describe.**

The problem is that, when we deprecate a module or subsystem, today we have no way of ensuring that it keeps working during the two release deprecation period. This is because CI fails whenever `select DEPRECATED` is used in Kconfig (as it should, since we don't want to allow deprecated code to be used in-tree).

**Describe the solution you'd like**
Extend the `{sample,testcase}.yaml` format so certain samples and tests can be built and executed even when enabling deprecated features.

**Describe alternatives you've considered**
Have a special CI build (i.e. `twister` invocation) that does not fail on deprecation.

**Additional context**
USB: https://github.com/zephyrproject-rtos/zephyr/issues/42066
TinyCrypt: https://github.com/zephyrproject-rtos/zephyr/issues/82323

carlescufi, 2025-01-08:
CC @jfischer-no @PavelVPV

PavelVPV, 2025-01-21:
@carlescufi , @nashif , bsim tests pass without requiring any changes to CI: https://github.com/zephyrproject-rtos/zephyr/pull/84237. Should we go ahead with merging that PR then?

alxelax, 2025-01-22:
Twister doesn't require any changes to restore BLE Mesh BabbleSim testing with Tinycrypt. Since Mesh bsim tests are run over bash script, the deprecation message is not trapped.