zephyrproject-rtos/zephyr #83714: ARM_MPU_REGION_MIN_ALIGN_AND_SIZE has invalid value when FPU_SHARING && MPU_STACK_GUARD

rysiof, 2025-01-08:
**Describe the bug**
With our builds for Arm M7 we are selecting `FPU_SHARING` and `MPU_STACK_GUARD`. For that setup, we have `ARM_MPU_REGION_MIN_ALIGN_AND_SIZE=64` and `MPU_STACK_GUARD_MIN_SIZE_FLOAT=128`. So worst case scenario, we are going to have stack that is 64-byte aligned and 128-byte guard at the bottom of the buffer.

https://developer.arm.com/documentation/101407/0541/Debugging/Debug-Windows-and-Dialogs/Core-Peripherals/Armv7-M-cores/Armv7-M--Memory-Protection-Unit tells us that `The base address is aligned to the size of the region`.

Therefore `ARM_MPU_REGION_MIN_ALIGN_AND_SIZE` must be 128 when `FPU_SHARING && MPU_STACK_GUARD`

**To Reproduce**

**Expected behavior**

**Impact**
security and debugability

**Logs and console output**

**Environment (please complete the following information):**
All

**Additional context**

rysiof, 2025-01-20:
Hi @wearyzen 
Issue is a bit different. Here is an example:
data:   0x10000000 - 0x10000480
stack1: 0x10000480 - 0x10001080

When we switch to stack1, zephyr will try to program MPU guard to
address=0x10000480 and size=128 (due to FPU_SHARING) in my case. This won't work properly as MPU address and size must be aligned. The way MPU will behave is that it will align down address (confirmed with Arm Fast Models) and we will protect address=0x10000400 and size=128 (so not as expected).

So top part of data won't be accessible, when executing thread on stack1, causing MPU fault exception.

wearyzen, 2025-01-20:
Ok so the issue could be because of [this](https://github.com/zephyrproject-rtos/zephyr/blob/8b987d9e0293b1616ae738cdbee0bac33b71c271/cmake/linker_script/arm/linker.cmake#L7) or [this](https://github.com/zephyrproject-rtos/zephyr/blob/8b987d9e0293b1616ae738cdbee0bac33b71c271/include/zephyr/arch/arm/cortex_m/scripts/linker.ld#L64).

In that case the PR looks good.

rysiof, 2025-01-20:
Tyty! I really appreciate that!

closed 2025-01-22 by kartben

gabbla, 2025-03-26:
Hello, I'm writing under this issue as a starting point. 
I'm updating my project from Zephyr 3.6.0 to 4.1.0 and I noticed that https://github.com/zephyrproject-rtos/zephyr/pull/83716 broke my firmware on STM32H573 (arm v8). In my configuration I have `CONFIG_FPU=y` and `CONFIG_MPU_STACK_GUARD=y`.
During the first context switch I'm getting a BUS_FAULT with reason 2  (`K_ERR_STACK_CHK_FAIL`).
To make the firmware run I either need to set `CONFIG_FPU_SHARING=n` or revert back `CONFIG_ARM_MPU_REGION_MIN_ALIGN_AND_SIZE` to 32 bytes.
I'm not an expert on how the MPU works but it seems that my issue is related to the above PR.

rysiof, 2025-03-26:
> Hello, I'm writing under this issue as a starting point. I'm updating my project from Zephyr 3.6.0 to 4.1.0 and I noticed that [#83716](https://github.com/zephyrproject-rtos/zephyr/pull/83716) broke [...]

Hi, `K_ERR_STACK_CHK_FAIL` indicates stack overflow, so i would increase stack size of a thread that crashed.

gabbla, 2025-03-26:
> Hi, K_ERR_STACK_CHK_FAIL indicates stack overflow, so i would increase stack size of a thread that crashed.

Hi rysiof, you are right, however I'm pretty sure I'm not overflowing any stack. 

[image]

The BUS_FAULT happens at the beginning of main and not during a context switching. The trigger seems to be a call to `std::make_shared<...>` of an object with a size of 40 bytes (but even allocating `uint8_t` triggers the exception). Using `new` seems to work. My heap is 8KiB

rysiof, 2025-03-26:
> Hi rysiof, you are right, however I'm pretty sure I'm not overflowing any stack.
>
> [image]
>
> The BUS_FAULT happens at the beginning of main and not during a context switching. The trigger seems to be [...]

Hm, that is interesting. not sure why make shared fails while new/malloc works tbh - i would expect both of them fail.
This might be some alignment issue. Could you double check if your main stack is aligned to 128 bytes? If stack is not aligned then MPU will guard bottom part od stack and some region in memory before stack, causing problems - this is what we were seeing on our ARM M7 project.

If main stack is aligned, then maybe there some more differences between M33 and M7 in MPU/FPU. If possible narrow down to exact load/store instruction and address that causes issues

gabbla, 2025-03-26:
> Hm, that is interesting. not sure why make shared fails while new/malloc works tbh - i would expect both of them fail.
This might be some alignment issue. Could you double check if your main stack is aligned to 128 bytes? If stack is not aligned then MPU will guard bottom part od stack and some region in memory before stack, causing problems - this is what we were seeing on our ARM M7 project.
> If main stack is aligned, then maybe there some more differences between M33 and M7 in MPU/FPU. If possible narrow down to exact load/store instruction and address that causes issues

So while checking the stack alignment I noticed that my main stack size was not aligned (not even to 32 bytes). Aligning the main stack size to 128 fixed the issue.

Thank you very much for your time!
Gab

rysiof, 2025-03-26:
> So while checking the stack alignment I noticed that my main stack size was not aligned (not even to 32 bytes). Aligning the main stack size to 128 fixed the issue.
>
> Thank you very much for your [...]

Yay!

PR #83716 (2025-01-22): Changes MPU region alignment to 128 when FPU_SHARING and MPU_STACK_GUARD
Changes ARM_MPU_REGION_MIN_ALIGN_AND_SIZE to 128 if ARM_MPU && FPU_SHARING && MPU_STACK_GUARD

Fixes: #83714
commit ddff91fe (2025-01-22): arch: Changes MPU alignment to 128 when FPU_SHARING and MPU_STACK_GUARD

With our builds for Arm M7 we are selecting FPU_SHARING and
MPU_STACK_GUARD. For that setup, we have
ARM_MPU_REGION_MIN_ALIGN_AND_SIZE=64 and
MPU_STACK_GUARD_MIN_SIZE_FLOAT=128. So worst case scenario,
we are going to have stack that is 64-byte aligned and
128-byte guard at the bottom of the buffer.

Fixes: #83714