zephyrproject-rtos/zephyr #83717: FRDM-MCXW71tests/bluetooth/mesh_shell Cannot turn on advertising

bschuster1031-nxp, 2025-01-08:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

bschuster1031-nxp, 2025-01-13:
Using nRF connect and nRF mesh I also find that the mesh does not appear to show up

carlescufi, 2025-01-30:
> Using nRF connect and nRF mesh I also find that the mesh does not appear to show up

@PavelVPV could you please confirm if this is reproduceable on Nordic hw?

PavelVPV, 2025-01-30:
Since CDB is enabled, you need to execute `mesh cdb create` command in shell first.

PavelVPV, 2025-01-30:
Sorry, I was wrong. -120 means that the device is already provisioned.

From the logs, it seems something was stored in the flash, so mesh thinks it is provisioned:
```
[00:00:00.215,000] bt_mesh_access: Invalid vendor model index 0
[00:00:00.215,000] bt_mesh_access: Failed to get model for elem_idx 0 mod_idx 0
[00:00:00.215,000] settings: set-value failure. key: bt/mesh/v/0/sub error(-2)
```

One reason why it could happen is that a new firmware with a different Composition Data was flashed to the already provisioned device without unprovisioning.

decsny, 2025-02-19:
@axelnxp

axelnxp, 2025-02-24:
Just checked on the RC2 tag with FRDM-MCXW71, and I can confirm the mesh advertising can start correctly.

```
*** Booting Zephyr OS build v4.0.0-5753-g6504a1e2a635 ***
WARNING: Using an insecure PSA ITS encryption key provider.
[00:00:00.034,000] <wrn> secure_storage: Using an insecure PSA ITS encryption key provider.
[00:00:00.039,000] <inf> fs_nvs: 8 Sectors of 8192 bytes
[00:00:00.039,000] <inf> fs_nvs: alloc wra: 0, 1fd0
[00:00:00.039,000] <inf> fs_nvs: data wra: 0, 0
Initializing...
Press the <Tab> button for supported commands.
Before any Mesh commands you must run "mesh init"
[00:00:00.062,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:01.381,000] <inf> bt_hci_core: Identity: 00:60:37:08:00:19 (public)
[00:00:01.381,000] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x8300, manufacturer 0x0025
[00:00:01.381,000] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0x1400
Mesh initialized
Use "prov pb-adv on" or "prov pb-gatt on" to enable advertising
uart:~$ 
---- Sent utf8 encoded message: "mesh init\r\n" ----
mesh init
Mesh shell initialized
uart:~$ 
---- Sent utf8 encoded message: "mesh prov pb-adv on\r\n" ----
mesh prov pb-adv on
PB-ADV enabled
[00:00:09.938,000] <inf> bt_mesh_provisionee: Device UUID: dddd0000-0000-0000-0000-000000000000
uart:~$ 
```

decsny, 2025-02-24:
@bschuster1031-nxp , Axel says he tried on the latest version of Zephyr and no problem, you can double check and if that's the case, we can close this?

bschuster1031-nxp, 2025-02-26:
> [@bschuster1031-nxp](https://github.com/bschuster1031-nxp) , Axel says he tried on the latest version of Zephyr and no problem, you can double check and if that's the case, we can close this?

Yes I confirmed this on my own board