zephyrproject-rtos/zephyr #83725: Does Nordic have plan to port partition_manager to zephyr main?

swift-tk, 2025-01-09:
**Is your feature request related to a problem? Please describe.**

Fixed-partitions are just not flexible enough and require manaul computation of image sizes.

The partition_manager seems flexible enough and could not only manage flash and also manage RAM.
https://developer.nordicsemi.com/nRF_Connect_SDK/doc-legacy/2.6.99-cs1/nrf/scripts/partition_manager/partition_manager.html#

**Describe the solution you'd like**

Port partition_manager to zephyr main

**Describe alternatives you've considered**

**Additional context**

carlescufi, 2025-01-09:
@swift-tk no, we have no plans, sorry.

carlescufi, 2025-01-09:
@swift-tk instead, what I would suggest is that you open an RFC in upstream Zephyr regarding ROM/RAM partition management and mention NCS' Partition Manager there. This way we can get a discussion started on the future of partitions in upstream Zephyr.