zephyrproject-rtos/zephyr #83735: CoAP server should not silently truncate and ACK too long requests

martinjaeger, 2025-01-09:
**Describe the bug**

The CoAP server reads incoming packets from the UDP socket using `recvfrom`, which silently truncates messages if they don't fit into the buffer (see [Socket API docs here](https://pubs.opengroup.org/onlinepubs/9699919799/functions/recvfrom.html)).

There is an assertion which checks the returned length. However, this assertion will never fail as long as the Socket API is implemented correctly.

https://github.com/zephyrproject-rtos/zephyr/blob/d78952b7ba930e20910c2b6fdac49cc8dc4f709e/subsys/net/lib/coap/coap_server.c#L139-L141

**To Reproduce**

Steps to reproduce the behavior:
1. Build some code that uses the CoAP server, e.g. `samples/net/sockets/coap_server`
2. Send a CoAP request that is larger than `CONFIG_COAP_SERVER_MESSAGE_SIZE`.
3. Observe that the message is silently truncated.

**Expected behavior**

The CoAP server should not further process incomplete CoAP messages. It should also not respond to the client with an ACK, but with [4.13 Request Entity Too Large](https://www.rfc-editor.org/rfc/rfc7252.html#section-5.9.2.9) instead.

**Impact**

Buffers must be oversized to avoid undefined behavior, as incoming request length is usually not fully predictable at compile-time.

**Environment (please complete the following information):**

 - Commit SHA: d78952b7ba930e20910c2b6fdac49cc8dc4f709e

pdgendt, 2025-01-09:
Please test #83758 if this fixes your issue.
One comment though, is that the "4.13 Request Entity Too Large" is a response code which should still be done with an ACK in case of a confirmable message, I think.

martinjaeger, 2025-01-09:
Wow, thank you very much for the quick fix. Yeah, sure, you're right. It's still an ACK, but with an error code.

I will test it tomorrow and let you know the result.

PR #83758 (2025-01-13): net: lib: coap: Handle truncated messages in CoAP server
If the CoAP server receives a message that doesn't fit into the receive buffer, we should stop processing the message and respond to the client with 4.13 "Request Entity too large".

Fixes #83735
closed 2025-01-13 by kartben