zephyrproject-rtos/zephyr #83737: :zephyr:board: role not usable in other projects

gmarull, 2025-01-09:
**Describe the bug**

Zephyr boards are now defined using the custom `.. zephyr:board:: NAME` Sphinx directive, and referenced back using

```
:zephyr:board:`NAME`
```

However, it looks like the role is broken when used in external projects. Note that sample references (`:zephyr:code-sample:`) work just fine using the custom Zephyr domain outside of Zephyr.

**To Reproduce**

- Create a new Sphinx project, e.g. with `sphinx-quickstart`
- Enable the Zephyr domain, `extensions   = ["zephyr.domain", ...]`
- On any page, write e.g.
  ```
  :zephyr:board:`actinius_icarus`
  ```

Observe build errors:

```
Traceback (most recent call last):                                                                                                                                        
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/cmd/build.py", line 514, in build_main
    app.build(args.force_all, args.filenames)                                 
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/application.py", line 381, in build                                                   
    self.builder.build_update()                                                                                                                                           
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/builders/__init__.py", line 358, in build_update                                      
    self.build(                                                                                                                                                           
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/builders/__init__.py", line 437, in build                                             
    self.write(docnames, list(updated_docnames), method)                                                                                                                  
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/builders/__init__.py", line 711, in write                                             
    self.write_documents(docnames)                                                                                                                                        
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/builders/__init__.py", line 723, in write_documents
    self._write_parallel(sorted_docnames, nproc=self.app.parallel - 1)              
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/builders/__init__.py", line 780, in _write_parallel                                   
    doctree = self.env.get_and_resolve_doctree(docname, self)                                                                                                             
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                             
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/environment/__init__.py", line 661, in get_and_resolve_doctree
    self.apply_post_transforms(doctree, docname)                                
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/environment/__init__.py", line 729, in apply_post_transforms
    transformer.apply_transforms()                                                                                                                                        
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/transforms/__init__.py", line 94, in apply_transforms
    super().apply_transforms()  # type: ignore[misc]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^                                                       
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/docutils/transforms/__init__.py", line 182, in apply_transforms
    transform.apply(**kwargs)                                                                                                                                             
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/transforms/post_transforms/__init__.py", line 44, in apply
    self.run(**kwargs)                                                                                                                                                    
  File "/home/gmarull/ws/nordic/sdk-nrf/.venv/lib64/python3.12/site-packages/sphinx/transforms/post_transforms/__init__.py", line 90, in run
    newnode = domain.resolve_xref(                                                                                                                                        
              ^^^^^^^^^^^^^^^^^^^^                                                                                                                                        
  File "/home/gmarull/ws/nordic/sdk-nrf/zephyr/doc/_extensions/zephyr/domain/__init__.py", line 847, in resolve_xref
    elem["docname"],                                                                                                                                                      
    ~~~~^^^^^^^^^^^                                                                                                                                                       
KeyError: 'docname'  
```

**Expected behavior**

Zephyr domain must be usable outside the project, as nowadays, it is the only way to cross-reference to board pages. This used to work fine with plain top-page reference and intersphinx.

**Impact**

Impossible to reference board pages from other Sphinx projects.

**Logs and console output**

N/A

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used d78952b7ba930e20910c2b6fdac49cc8dc4f709e

**Additional context**

N/A

kartben, 2025-01-14:
> @gmarull can you please try with a single-process build (-j 1), as I have a hunch the bug might be in `ZephyrDomain#merge_domaindata`

yeah no, it's not that :)

kartben, 2025-01-14:
btw would it make sense to add basic demo usage of some of the Zephyr domain directives in the example-application?

[image]

gmarull, 2025-01-14:
> [@gmarull](https://github.com/gmarull) btw would it make sense to add basic demo usage of some of the Zephyr domain directives in the example-application?
> 
> <img alt="Image" width="943" src="https://private-user-images.githubusercontent.com/128251/402913083-8a5731bf-3a54-41a3-8baa-f5bc96e524c5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4NTQzNDAsIm5iZiI6MTczNjg1NDA0MCwicGF0aCI6Ii8xMjgyNTEvNDAyOTEzMDgzLThhNTczMWJmLTNhNTQtNDFhMy04YmFhLWY1YmM5NmU1MjRjNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExNFQxMTI3MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MmIxMjBiNDZiYThiNDU3NjA3MjVlY2Y2MDI4Nzk2YjJlZGI0NDlmYjgzMjNmYzY3YmRkNzU2MmNiNzFiNDVhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Vnz5Rk-a4WQsGSqoO-0yL1CiKXFhuOBJUthAd5AB1eM">

ok, feel free to open a PR

henrikbrixandersen, 2025-01-14:
> [@gmarull](https://github.com/gmarull) btw would it make sense to add basic demo usage of some of the Zephyr domain directives in the example-application?

Yes, please! :)