zephyrproject-rtos/zephyr #83741: Use hyphens instead of underscores for property separation

rruuaanng, 2025-01-09:
**Is your enhancement proposal related to a problem? Please describe.**

#53506 #83352 work list

In the current source tree, all properties in device trees and bindings across subsystems that contain underscores need to be renamed using hyphens. The subsystems requiring these changes are listed below. (The scope of work only includes `dts/` directory)

**Describe the solution you'd like**

None

**Describe alternatives you've considered**

None

**Additional context**

A change probably includes such a process:
1. e.g. This property `tick_us` needs to be renamed to `tick-us`.
2. Modify the corresponding attribute name in the corresponding binding.

These two are exceptions:
```
mmc-hs400-1_8v
mmc-hs200-1_8v
```

henrikbrixandersen, 2025-01-10:
@rruuaanng Thank you for opening this.

For the CAN bindings, the only underscores I see, are:
- `dts/bindings/can/bosch,m_can-base.yaml` binding name (aligns with https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/net/can/bosch%2Cm_can.yaml)
- `dts/bindings/can/infineon,xmc4xxx-can-node.yaml`:  `clock_div8`: Renamed here: https://github.com/zephyrproject-rtos/zephyr/pull/83782

rruuaanng, 2025-01-10:
Thank you for the time you spent on this task, Henrik, it was quite difficult to locate them 👍

rruuaanng, 2025-01-11:
cc @jilaypandya Maybe you will be interested in stepper.

PR #83352 (2025-06-17): treewide: Use hyphens instead of underscores for property separation
This inherits from proposal #53502, with the main work as shown in the title. For details, see https://github.com/zephyrproject-rtos/zephyr/issues/53506.

Closes: #83741
closed 2025-06-17 by henrikbrixandersen