zephyrproject-rtos/zephyr #83743: ESP32-S3 with Octal SPI RAM - Freezes During Boot

Piziwate, 2025-01-09:
**Describe the bug**
When I compile the hello_world example with PSRAM enabled in octal mode, the startup process freezes during initialization. This issue is reproducible on the `esp32s3_devkitc/esp32s3/procpu` board. Module used: `ESP32-S3-DevKitC-1-N32R8V`.

The freeze occurs within the HAL at the file `modules\hal\espressif\components\esp_psram\esp32s3\esp_psram_impl_octal.c`. Specifically, it happens during a call to the method `esp_opiflash_set_required_regs()`, where the s_chip_func function is NULL and the s_chip_id variable is set to 0.

I don't quite understand the connection between flash initialization and SPI RAM in this case.

**To Reproduce**

1. west build -b esp32s3_devkitc/esp32s3/procpu ./samples/hello_world/ --pristine -DCONFIG_ESP_SPIRAM=y -DCONFIG_SPIRAM_TYPE_ESPPSRAM64=y -DCONFIG_SPIRAM_MODE_OCT=y -DCONFIG_SPIRAM_SPEED_80M=y -DCONFIG_ESPTOOLPY_FLASHMODE_OPI=y
2. west flash
3. Boot process hangs after outputing `(276) MSPI Timing: PSRAM timing tuning index: 5`

**Expected behavior**
The system should boot and display the message:
Hello World! esp32s3_devkitc/esp32s3/procpu

**Impact**
The good news is that you have a work-around by commenting out the line  ((*s_chip_func)->regs_set(s_chip_id)) in the HAL.
But it would be ideal to optimize this case to prevent issues for other users. 

**Logs and console output**
Full boot log :  [bootlog.txt](https://github.com/user-attachments/files/18361730/bootlog.txt)

**Environment (please complete the following information):**

 - OS: Windows 11
 - Zephyr SDK : zephyr-sdk-0.16.8 & zephyr-sdk-0.17.0
 - Commit SHA or Version used : main branch

sylvioalves, 2025-01-10:
Fixed, will submit it soon.

sylvioalves, 2025-01-10:
@Piziwate @tyalie Would you be able to check #83817 ?

Piziwate, 2025-01-13:
@sylvioalves  I just tested it, and I’m pleased to confirm that everything is in order on my end!

Piziwate, 2025-01-13:
@sylvioalves However, I am still encountering a minor issue with the flash in octal mode. When I compile the following example:
`west build -b esp32s3_devkitc/esp32s3/procpu samples/subsys/fs/littlefs/ --pristine -DCONFIG_ESP_SPIRAM=y -DCONFIG_SPIRAM_TYPE_ESPPSRAM64=y -DCONFIG_SPIRAM_MODE_OCT=y -DCONFIG_SPIRAM_SPEED_80M=y -DCONFIG_ESPTOOLPY_FLASHMODE_OPI=y`
The SPI RAM functions properly and is correctly tested, but it is not possible to access the flash:
```
W (770) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
E (774) spi_flash: failed to get chip size
E: esp_flash_init_default_chip failed 24581
*** Booting Zephyr OS build v4.0.0-3020-g81563c12a25d ***
Sample program to r/w files on littlefs
Area 5 at 0x3c0000 on flash-controller@60002000 for 131072 bytes
I: LittleFS version 2.9, disk version 2.1
I: FS at flash-controller@60002000:0x3c0000 is 32 0x1000-byte blocks with 512 cycle
I: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: esp_flash_read failed 258
W: can't mount (LFS -5); formatting
E: esp_flash_read failed 258
```
Should I open an issue for this, or could it be related to something else?

sylvioalves, 2025-01-15:
@Piziwate can you, for now, comment the line below before I submit a proper fix to this issue?
https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/components/spi_flash/esp_flash_spi_init.c#L404

Piziwate, 2025-01-15:
Thank you @sylvioalves, the use of the flash is indeed functional with this line commented out!

PR #85625 (2025-02-14): soc: espressif: rework code startup to improve flash and spiram loader
This PR reworks all Espressif's SOC initialization code to fix booting issues regaring OCTAL FLASH/SPIRAM.
This also updates bootloader (either Simple Boot and MCUBoot) to use ROM functions during flash read/write operations, ensuring no dependencies of the flash driver initialization.
Runner handler is now common to all SoCs and fixes flash mode and frequency configuration, required for some flash memories.
Esptool is also updated to v4.8.1.

Fixes #64757
Fixes #85246
Fixes #83743
Fixes #82244
closed 2025-02-14 by kartben