zephyrproject-rtos/zephyr #83745: samples: boards: st: serial_wakeup doesn't work with stm32l562e_dk.

gbarkadiusz, 2025-01-09:
**Describe the bug**
I'd like to use the sample `samples.boards.stm32.power_mgmt.serial_wakeup` as an example for using the LPUART and the wakeup-source ability of the UART on the `STM32L562E-DK`. I noticed that this board was removed from the `platform allow` property in the `sample.yaml` file at https://github.com/zephyrproject-rtos/zephyr/pull/49798. The reason for its removal was that it cannot be tested under CI. So, as I understand it, this sample should work with the `STM32L562E-DK` board locally, but it doesn't. This could be a potential regression. I don't know how long this test has not worked with the `STM32L562E-DK`, so performing a bisect is difficult.

**To Reproduce**
Under twister.
- Add stm32l562e_dk to the sample.yaml file as platform allow.
- Run: `twister --device-testing --device-serial /dev/ttyACM1 -p stm32l562e_dk -T samples/boards/st/power_mgmt/serial_wakeup/`

**Logs and console output**
Log:
```
INFO    - Using Ninja..
INFO    - Zephyr version: v4.0.0-3084-gd78952b7ba93
DEBUG   - Running cmake script <path_to>/zephyrproject/zephyr/cmake/verify-toolchain.cmake
DEBUG   - Calling cmake: /usr/bin/cmake -DFORMAT=json -P /zephyrproject/zephyr/cmake/verify-toolchain.cmake
DEBUG   - Finished running <path_to>/zephyrproject/zephyr/cmake/verify-toolchain.cmake
INFO    - Using 'zephyr' toolchain.
DEBUG   - Reading testsuite configuration files under <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup...
DEBUG   - Found possible testsuite in <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup
DEBUG   - No duplicates found.
DEBUG   - platform filter: ['stm32l562e_dk']
DEBUG   -   vendor filter: []
DEBUG   -     arch_filter: None
DEBUG   -      tag_filter: None
DEBUG   -     exclude_tag: None
DEBUG   - Checking platform filter: ['stm32l562e_dk']
INFO    - Building initial testsuite list...
INFO    - Writing JSON report <path_to>/zephyrproject/zephyr/twister-out/testplan.json

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|

INFO    - JOBS: 4
INFO    - Adding tasks to the queue...
DEBUG   - adding stm32l562e_dk/stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup
INFO    - Added initial list of jobs to queue
DEBUG   - Launched 4 jobs
DEBUG   - Running cmake on<path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx
DEBUG   - Calling cmake: /usr/bin/cmake -B<path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup -DTC_RUNID=e9036b81f0a050c56505061ab80347d9 -DTC_NAME=samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y -DEXTRA_GEN_EDT_ARGS=--edtlib-Werror -GNinja -DPython3_EXECUTABLE=/usr/bin/python3 -DZEPHYR_TOOLCHAIN_VARIANT=zephyr -S<path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup -DCONFIG_DEBUG=y -DBOARD=stm32l562e_dk/stm32l562xx -DMODULES=dts -P<path_to>/zephyrproject/zephyr/cmake/package_helper.cmake
DEBUG   - Finished running cmake <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx in 4.85 seconds
DEBUG   - Running cmake on <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx
DEBUG   - Calling cmake: /usr/bin/cmake -B<path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup -DTC_RUNID=e9036b81f0a050c56505061ab80347d9 -DTC_NAME=samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y -DEXTRA_GEN_EDT_ARGS=--edtlib-Werror -GNinja -DPython3_EXECUTABLE=/usr/bin/python3 -DZEPHYR_TOOLCHAIN_VARIANT=zephyr -S<path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup -DCONFIG_DEBUG=y -DBOARD=stm32l562e_dk/stm32l562xx
DEBUG   - Finished running cmake <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx in 8.77 seconds
DEBUG   - build test: stm32l562e_dk/stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup
DEBUG   - Building <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx
DEBUG   - Running /usr/bin/cmake --build <path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup
DEBUG   - Finished building <path_to>/zephyrproject/zephyr/samples/boards/st/power_mgmt/serial_wakeup for stm32l562e_dk/stm32l562xx in 14.77 seconds
DEBUG   - run test: stm32l562e_dk/stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup
DEBUG   - Reset instance status from 'passed' to None before run.
DEBUG   - Retain DUT:stm32l562e_dk/stm32l562xx, Id:None, counter:1, failures:0
DEBUG   - Using serial device /dev/ttyACM1 @ 115200 baud
DEBUG   - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '<path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup', '--runner', 'pyocd', '--', '--dev-id=004800483137510D33333639']
DEBUG   - -- west flash: using runner pyocd
-- runners.pyocd: Flashing file: <path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup/zephyr/zephyr.hex
[---|---|---|---|---|---|---|---|---|----]
[========================================]

DEBUG   - Timed out while monitoring serial output on stm32l562e_dk/stm32l562xx
DEBUG   - Release DUT:stm32l562e_dk/stm32l562xx, Id:None, counter:1, failures:1
DEBUG   - run status: stm32l562e_dk/stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup failed
ERROR   - stm32l562e_dk/stm32l562xx samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup FAILED: Timeout
ERROR   - see: <path_to>/zephyrproject/zephyr/twister-out/stm32l562e_dk_stm32l562xx/zephyr/samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup/device.log
INFO    - Total complete:    1/   1  100%  built (not run):    0, filtered:    0, failed:    1, error:    0
INFO    - 1 test scenarios (1 configurations) selected, 0 configurations filtered (0 by static filter, 0 at runtime).
INFO    - 0 of 1 executed test configurations passed (0.00%), 0 built (not run), 1 failed, 0 errored, with no warnings in 96.00 seconds.
INFO    - 0 of 1 executed test cases passed (0.00%), 1 blocked on 1 out of total 902 platforms (0.11%).
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.

Hardware distribution summary:

| Board                     | ID   |   Counter |   Failures |
|---------------------------|------|-----------|------------|
| stm32l562e_dk/stm32l562xx |      |         1 |          1 |
INFO    - Saving reports...
INFO    - Writing JSON report <path_to>/zephyrproject/zephyr/twister-out/twister.json
INFO    - Writing xunit report <path_to>/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report <path_to>/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup on stm32l562e_dk/stm32l562xx failed (Timeout)
INFO    - 
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    - 
INFO    - west twister -p stm32l562e_dk/stm32l562xx -s samples/boards/st/power_mgmt/serial_wakeup/sample.boards.stm32.power_mgmt.serial_wakeup
INFO    - or with west:
INFO    - west build -p -b stm32l562e_dk/stm32l562xx samples/boards/st/power_mgmt/serial_wakeup -T sample.boards.stm32.power_mgmt.serial_wakeup
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - Run completed
```
**Environment:**

 - OS: Linux
 - Toolchain: Zephyr SDK 17.0
 - Commit SHA or Version used: v4.0.0-3084-gd78952b7ba9

gbarkadiusz, 2025-01-09:
@erwango FYI.

gautierg-st, 2025-07-16:
Hello,
I've tested the sample using west build/west flash and it works as expected.
Setup-wise, I connected the LPUART1 of the DK (PB10/PB11) to the serial line (TX/RX) of an ST-Link. And I received the expected payload on the serial line of the ST-Link.
With Twister, I think that the problem is that the serial line used as a console is not the same as the one used for flashing, and I'm not sure if there is a mean to differentiate the two of them.