zephyrproject-rtos/zephyr #83764: Twister generated code coverage report missing files outside the zephyr repository.

solsbarry, 2025-01-09:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux (ubuntu 22.04)
 - Toolchain 0.16.8
 - Commit SHA or Version used 4.0.0

**Additional context**

nashif, 2025-01-20:
This could be due to the fact we call the gcovr with `-r <zephyr base>` with the purpose of getting coverage data from the zephyr tree. Please try to generate the reports manually (https://docs.zephyrproject.org/latest/develop/test/coverage.html) and see if changing the root will give you what you need. If that is the problem, we probably want to change how this is done in twister.

@golowanow  can you please look into this?

solsbarry, 2025-01-21:
Yes I I manually run gcovr and I specify my project folder as the source base instead of the zephyr base then that works. If twister could accommodate coverage outside the zephyr base that would be great. But I can manually generate the reports myself in the meantime

golowanow, 2025-01-23:
@solsbarry, I'm working this.

golowanow, 2025-01-26:
@solsbarry, could you please try to run Twister on your setup with explicit `--coverage-basedir` to the workspace directory i.e.:

assuming a workspace application [directory layout](https://docs.zephyrproject.org/latest/develop/application/index.html#zephyr-workspace-application), and `testcase.yaml` in `applications/app`
```
cd zephyrproject
./zephyr/scripts/twister -p native_sim --coverage --coverage-tool gcovr --coverage-basedir $ZEPHYR_BASE/.. -T applications/app/
```
On my setup (some recent main 4ed85eec, gcovr 8.2, gcov from Zephyr SDK 0.17.0) the gcovr report includes both zephyr's code and the workspace application's coverage, BUT **what is important** - only when my ZEPHYR_BASE env variable has real paths to the zephyrproject/zephyr directory without any symlinks used. 
Looks like gcovr miss source files when it does its filtering/matching and the symlink-ed paths are different from what is in .gcno files. The same problem happens even for a in-tree test application.  

Anyway, I think lcovr should work better than gcovr in this case (`twister --coverage-tool lcov`).

solsbarry, 2025-01-27:
Yes using `--coverage-basedir` worked, and included the code coverage for functions outside Zephyr. thank you.

commit 9228bb20 (2025-01-30): doc: coverage: Add Twister `--coverage-basedir` use case

Add an example of using Twister for an 'out-of-tree' application's
code coverage report generation with `--coverage-basedir` option,
and a note about `gcovr` tool possible problems with symlinks.

Fixes: #83764
closed 2025-01-30 by kartben