zephyrproject-rtos/zephyr #83766: Twister ignores ZEPHYR_TOOLCHAIN_VARIANT environment variable for posix architecture

keith-zephyr, 2025-01-10:
**Describe the bug**

Twister ignores ZEPHYR_TOOLCHAIN_VARIANT environment variable for posix architecture. 

This was introduced by https://github.com/zephyrproject-rtos/zephyr/pull/82515

**To Reproduce**

Run the following:

```
ZEPHYR_TOOLCHAIN_VARIANT=llvm VERBOSE=1 ./scripts/twister -c -p native_sim -s cpp.libcxx.host
```

Twister uses the "host" toolchain, usually GCC, instead of the selected toolchain variant.

**Expected behavior**
Twister should honor the toolchain set in the environment.

**Impact**
Breaks downstream repos that run tests using a toolchain variant.

I attempted to fix with https://github.com/zephyrproject-rtos/zephyr/pull/83762, but that PR hits CI failures.

keith-zephyr, 2025-01-10:
Considering the fix looks quite a bit more involved, can we revert   https://github.com/zephyrproject-rtos/zephyr/pull/82515?

nashif, 2025-01-10:
> Considering the fix looks quite a bit more involved, can we revert #82515?

I have a workaround for now, this all is caused by the how messy toolchain variants setup work now and hopefully will be improved once we introduce llvm sdk. 

a vartiant names 'host' (signifying gcc, though llvm is also a host toolchain) which is automagically set for native_sim and others is a pain to get right in terms of logic.

keith-zephyr, 2025-01-10:
> I have a workaround for now, this all is caused by the how messy toolchain variants setup work now and hopefully will be improved once we introduce llvm sdk.
> 
Do you have  a PR you can share?  This issue is blocking our daily process to pull Zephyr changes into our downstream repo.

nashif, 2025-01-10:
> Do you have a PR you can share? This issue is blocking our daily process to pull Zephyr changes into our downstream repo.

working on it right now, will push after some tests

PR #83772 (2025-01-10): twister: deal with variants set on command line
not force host variant if environment is setting something different in
the case of posix arch.

Fixes #83766
commit 5769966d (2025-01-10): twister: deal with variants set on command line

not force host variant if environment is setting something different in
the case of posix arch.

Fixes #83766
closed 2025-01-10 by nashif