zephyrproject-rtos/zephyr #83767: ICM42670 Constantly prints "Fetch from reg" error to log

aburt2, 2025-01-10:
**Describe the bug**

ICM42670 constantly prints "Fetch from reg" error to log even if no error has occured.

The `LOG_ERR` call in `icm42670_fetch_from_registers`
https://github.com/zephyrproject-rtos/zephyr/blob/7a189ae0fade5ec4280cd4e917772e31eac55c07/drivers/sensor/tdk/icm42670/icm42670.c#L782-L790

Should be moved further down in the function to here:

https://github.com/zephyrproject-rtos/zephyr/blob/7a189ae0fade5ec4280cd4e917772e31eac55c07/drivers/sensor/tdk/icm42670/icm42670.c#L825-L828

**To Reproduce**

- Add icm42670p to your project
- call `sample_fetch` in your application
- Compile without using `CONFIG_ICM42670_TRIGGER`

**Expected behavior**

Only print error if an error has actually occured, 

**Impact**

Slows down application due to multiple frequent logging calls

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: ZEPHYR SDK
 - Commit SHA or Version used: `HEAD` of `main`

**Additional context**

kartben, 2025-01-14:
@aburt2 you may want to open a pull request to change this to a `LOG_DBG` statement. But it might make sense to even remove this. 
cc @afontaine-invn

afontaine-invn, 2025-01-16:
Hi @aburt2 , this statement is not expected to be kept. Thanks for pointing this bug.

afontaine-invn, 2025-02-10:
This has been fixed with https://github.com/zephyrproject-rtos/zephyr/pull/83498 
We can close this issue and the PR