zephyrproject-rtos/zephyr #83794: lis2de12 sensor driver giving 0 values

narangmayank, 2025-01-10:
**Describe the bug**

**To Reproduce**

Build the sample attached below and run

**Expected behavior**

IMU values should print on the console

**Impact**

showstopper

**Logs and console output**

DTS

```
&i2c0 {	status = "disabled";};
&spi0 {	status = "disabled";};
&i2c1 {	status = "disabled";};
	
&spi1 {
    compatible = "nordic,nrf-spim";
	status = "okay";
	pinctrl-0 = <&spi1_default>;
	pinctrl-1 = <&spi1_sleep>;
	pinctrl-names = "default", "sleep";
	cs-gpios = <&gpio0 30 GPIO_ACTIVE_LOW>;
	lis2de12: lis2de12@0 {
		compatible = "st,lis2de12";
		reg = <0>;
		accel-range = <0>;
		accel-odr = <6>;
		spi-max-frequency = <1000000>; 
	};
};

&pinctrl {
	spi1_default: spi1_default {
		group1 {
				psels = <NRF_PSEL(SPIM_SCK, 0, 28)>,
						<NRF_PSEL(SPIM_MOSI, 0, 29)>,
						<NRF_PSEL(SPIM_MISO, 0, 31)>;
		};
	};

	spi1_sleep: spi1_sleep {
		group1 {
				psels = <NRF_PSEL(SPIM_SCK, 0, 28)>,
						<NRF_PSEL(SPIM_MOSI, 0, 29)>,
						<NRF_PSEL(SPIM_MISO, 0, 31)>;
				low-power-enable;
		};
	};
};
```

Code 

```
/*
 * Copyright (c) 2016 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>
#include <zephyr/kernel.h>
[... 302 lines cut ...]
[00:00:01.006,591] <inf> MAIN: lis2de12_xl[0]: 0, lis2de12_xl[1]: 0, lis2de12_xl[2]:0
[00:00:01.006,622] <inf> MAIN: LIS2DE12: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000

[00:00:02.002,593] <inf> MAIN: LED state: OFF
[00:00:02.009,307] <inf> MAIN: Calling sensor_sample_fetch()
[00:00:02.009,643] <inf> MAIN: lis2de12_xl[0]: 0, lis2de12_xl[1]: 0, lis2de12_xl[2]:0
[00:00:02.009,674] <inf> MAIN: LIS2DE12: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000

LIS2DE12: Temperature: 25.0 C
[00:00:03.002,685] <inf> MAIN: LED state: ON
[00:00:03.012,359] <inf> MAIN: Calling sensor_sample_fetch()
[00:00:03.012,725LIS2DE12: Temperature: 25.0 C
] <inf> MAIN: lis2de12_xl[0]: 0, lis2de12_xl[1]: 0, lis2de12_xl[2]:0
[00:00:03.012,756] <inf> MAIN: LIS2DE12: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows): Windows 11
 - Toolchain (e.g Zephyr SDK, ...): nrf Connect SDK v2.8.0
 - Commit SHA or Version used: Not sure how to check

**Additional context**

I tried reading CNTR_REG_1 and CNTR_REG_4 for checking ODR & g-value respectively that is getting set correctly as you can see in the logs. The next was to just read IMU values which is not working.

narangmayank, 2025-01-10:
Update: 

When I try to read XYZ registers manually (i.e 6 bytes from FIFO_READ_START register) it gives me value.

Code Snippet Added 

```
void read_imu_xyz_register()
{
	uint8_t tx_buffer[8] = {0x00};
	uint8_t rx_buffer[8] = {0x00};
	
	struct spi_buf tx_spi_buf		= {.buf = tx_buffer, .len = 8};
	struct spi_buf_set tx_spi_buf_set 	= {.buffers = &tx_spi_buf, .count = 1};
	struct spi_buf rx_spi_bufs 		= {.buf = rx_buffer, .len = 8};
	struct spi_buf_set rx_spi_buf_set	= {.buffers = &rx_spi_bufs, .count = 1};

	LOG_INF("Reading IMU registers");

	tx_buffer[0] = 0x28 | 0xC0; // multiple spi read command
	int err = spi_transceive_dt(&spispec, &tx_spi_buf_set, &rx_spi_buf_set);
	if (err < 0) {
		LOG_ERR("spi_transceive_dt() failed, err: %d", err);
		return;
	}

	LOG_INF("tx_buffer[0]: %X, tx_buffer[1]: %X", tx_buffer[0], tx_buffer[1]);

	printk("<------------ rx_buffer: ");
	for(int i=0; i<8; i++)
	{
		printk("%X ", rx_buffer[i]);
	}
	printk("------------>\n");
}

// In the main loop added the call to manual read fun as well
while (1) {
		// LOG_INF("Hello World");
		get_imu_data();
		read_imu_xyz_register();

		#if 0
		int status = lis2de12_status_check();

		if(status & 0x08 == 0x08)
		{
			LOG_INF("Reading IMU data");
			get_imu_data();
		}
		#endif

		k_msleep(1000);
	}
```

```
[00:01:55.518,737] <inf> MAIN: Reading IMU registers
[00:01:55.519,317] <inf> MAIN: tx_buffer[0]: E8, tx_buffer[1]: 0
<------------ rx_buffer: FF 0 FC 0 C7 0 E6 0 ------------>
LIS2DE12: Temperature: -102.5 C
[00:01:56.013,031] <inf> MAIN: LED state: OFF
[00:01:56.519,561] <inf> MAIN: Calling sensor_sample_fetch()
[00:01:56.520,355] <inf> MAIN: lis2de12_xl[0]: 0, lis2de12_xl[1]: 0, lis2de12_xl[2]:0
[00:01:56.520,385] <inf> MAIN: LIS2DE12: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000

[00:01:56.523,193] <inf> MAIN: Reading IMU registers
[00:01:56.523,742] <inf> MAIN: tx_buffer[0]: E8, tx_buffer[1]: 0
<------------ rx_buffer: FF 0 FC 0 C7 0 E6 0 ------------>
```

narangmayank, 2025-01-10:
@teburd @MaureenHelm Found out the issue. The issue was with SPI multi byte read even if application calls to read out more than 1 bytes the addr auto increment bit was not setting up.

PFA Git diff 

```
diff --git a/drivers/sensor/st/stmemsc/stmemsc_spi.c b/drivers/sensor/st/stmemsc/stmemsc_spi.c
index 3d728c78d8b..4b8c2fd8847 100644
--- a/drivers/sensor/st/stmemsc/stmemsc_spi.c
+++ b/drivers/sensor/st/stmemsc/stmemsc_spi.c
@@ -63,7 +63,10 @@ int stmemsc_spi_write(const struct spi_dt_spec *stmemsc,
 int stmemsc_spi_read_incr(const struct spi_dt_spec *stmemsc,
                          uint8_t reg_addr, uint8_t *value, uint8_t len)
 {
-       reg_addr |= STMEMSC_SPI_ADDR_AUTO_INCR;
+       if(len > 1) {
+               reg_addr |= STMEMSC_SPI_ADDR_AUTO_INCR;
+       }
+
        return stmemsc_spi_read(stmemsc, reg_addr, value, len);
 }

diff --git a/drivers/sensor/st/lis2de12/lis2de12.c b/drivers/sensor/st/lis2de12/lis2de12.c
index 9cceccdeb20..b417bccda39 100644
--- a/drivers/sensor/st/lis2de12/lis2de12.c
+++ b/drivers/sensor/st/lis2de12/lis2de12.c
@@ -435,7 +435,7 @@ static int lis2de12_init(const struct device *dev)

 #define LIS2DE12_CONFIG_SPI(inst)                                              \
        {                                                                       \
-               STMEMSC_CTX_SPI(&lis2de12_config_##inst.stmemsc_cfg),           \
+               STMEMSC_CTX_SPI_INCR(&lis2de12_config_##inst.stmemsc_cfg),              \
                .stmemsc_cfg = {                                                \
                        .spi = SPI_DT_SPEC_INST_GET(inst, LIS2DE12_SPI_OP, 0),  \
                },                                                              \
```

Getting IMU values after changing above 2 files.
Should I raise against the issue?

kartben, 2025-01-14:
> [...] Should I raise a PR against the issue?

@narangmayank yes please! Thanks :)

PR #84239 (2025-01-24): drivers: sensor: lis2de12: fix read accel via spi
Fixes: #83794 

Added a fix for reading acceleration data from LIS2DE12 sensor over SPI.
PR #84514 (2025-01-29): [Backport v3.7-branch] drivers: sensor: lis2de12: fix read accel via spi
Backport cb608812cfae2702bf244a77d945af07424cef2f~2..cb608812cfae2702bf244a77d945af07424cef2f from #84239.

Fixes #83794
PR #84515 (2025-01-31): [Backport v4.0-branch] drivers: sensor: lis2de12: fix read accel via spi
Backport cb608812cfae2702bf244a77d945af07424cef2f~2..cb608812cfae2702bf244a77d945af07424cef2f from #84239.

Fixes #83794
closed 2025-01-24 by kartben