zephyrproject-rtos/zephyr #83798: Bluetooth: LE: Audio: kernel panic with Bap Unicast Server sample when music stream is started (board nRF-5340-Audio-DK)

dev-pac, 2025-01-10:
**Describe the bug**

When my phone ( samsung zflip 4 ) start the music with the "bap unicast server" a kernel panic is trigger after 1/2 sec 

**To Reproduce**

Steps to reproduce the behavior:
1. edit samples/bluetooth/bap_unicast_server/prj.conf and add:
```
#### Custom cfg ####
CONFIG_LOG=y
# CONFIG_BT_LOG_LEVEL_DBG=y
CONFIG_CONSOLE=y
CONFIG_RTT_CONSOLE=y
CONFIG_LOG_PRINTK=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_LOG_BUFFER_SIZE=4096

# samsung zflip 4 send metadata supperior to default value (4)
## <wrn> bt_ascs: Not enough space for Codec Config Metadata: 7 > 4
## <err> bt_ascs: Enable rejected: err -12, code 13, reason 0
CONFIG_BT_AUDIO_CODEC_CFG_MAX_METADATA_SIZE=15
CONFIG_BT_AUDIO_CODEC_CAP_MAX_METADATA_SIZE=15
```
2. west build -b nrf5340dk/nrf5340/cpuapp --sysbuild samples/bluetooth/bap_unicast_server/
3. west flash
4. use a phone with BLE audio support then bound + stream a music

**Expected behavior**

No kernel panic and stream proceeded

**Impact**

**Logs and console output**

Logs:
```
SEGGER J-Link V7.94b - Real time terminal output
SEGGER J-Link (unknown) V1.0, SN=1050177497
Process: JLinkGDBServerCLExe
*** Booting Zephyr OS build v3.7.0 ***
[00:00:00.382,904] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.382,934] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.382,965] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.7 Build 0
[00:00:00.384,765] <inf> bt_hci_core: Identity: E8:19:DF:C5:36:52 (random)
[00:00:00.384,796] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.384,826] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
Location successfully set
Supported contexts successfully set
Available contexts successfully set
Advertising successfully started
Connected: 7E:E0:A3:C8:1F:4F (random)
ASE Codec Config: conn 0x200022e8 ep 0x200029ac dir 1
codec_cfg 0x06 cid 0x0000 vid 0x0000 count 19
data: type 0x01 value_len 1
08
data: type 0x02 value_len 1
01
data: type 0x03 value_len 4
00000000
data: type 0x04 value_len 2
7800
data: type 0x05 value_len 1
01
  Frequency: 48000 Hz
  Frame Duration: 10000 us
  Channel allocation: 0x0
  Octets per frame: 120 (negative means value not pressent)
  Frames per SDU: 1
ASE Codec Config stream 0x20005498
QoS: stream 0x20005498 qos 0x20007988
QoS: interval 10000 framing 0x00 phy 0x02 sdu 120 rtn 2 latency 10 pd 40000
Enable: stream 0x20005498 meta_len 7
RX stream 0x20005498 len 0
RX stream 0x20005498 len 0
RX stream 0x20005498 len 0
RX stream 0x20005498 len 0
Release: stream 0x20005498
Audio Stream 0x20005498 stopped with reason 0x13
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:410
        Controller unresponsive, command opcode 0x0406 timeout with err -11
[00:01:17.525,970] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:01:17.526,000] <err> os: r3/a4:  0x00000003 r12/ip:  0x00000010 r14/lr:  0x00005d9f
[00:01:17.526,000] <err> os:  xpsr:  0x01000000
[00:01:17.526,000] <err> os: s[ 0]:  0x200098d0  s[ 1]:  0x00000003  s[ 2]:  0x20009880  s[ 3]:  0x20009880
[00:01:17.526,031] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000001  s[ 6]:  0x20009890  s[ 7]:  0x20009890
[00:01:17.526,031] <err> os: s[ 8]:  0x00000006  s[ 9]:  0x20002608  s[10]:  0x20005498  s[11]:  0x20005cb0
[00:01:17.526,062] <err> os: s[12]:  0x200098d0  s[13]:  0xffffffff  s[14]:  0xffffffff  s[15]:  0x00000000
[00:01:17.526,062] <err> os: fpscr:  0x00000000
[00:01:17.526,062] <err> os: Faulting instruction address (r15/pc): 0x00005dae
[00:01:17.526,092] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:01:17.526,123] <err> os: Current thread: 0x20002f28 (unknown)
[00:01:18.045,745] <err> os: Halting system
```

[advertising debug logs](https://github.com/user-attachments/files/18375715/message.1.txt)
[bounding debug logs](https://github.com/user-attachments/files/18375725/message.2.txt)
[stream started debug logs](https://github.com/user-attachments/files/18375728/message.3.txt)

**Environment (please complete the following information):**

- Zephyr 3.7 (LTS)
- board nRF-5340-Audio-DK

**Additional context**

hermabe, 2025-01-10:
As the assertion message says, the controller is not responding to the disconnect command. Can you enable logging on the netcore and see what happens there? Most likely there is an assert or fault there that prevents it from responding.

dev-pac, 2025-01-10:
> As the assertion message says, the controller is not responding to the disconnect command. Can you enable logging on the netcore and see what happens there? Most likely there is an assert or fault there that prevents it from responding.

How can i get logs from cpunet ?

tried with RTT added in hci_ipc prj.conf but nothing received:
```
#### Custom cfg ####
CONFIG_LOG=y
CONFIG_CONSOLE=y
CONFIG_RTT_CONSOLE=y
CONFIG_LOG_PRINTK=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_LOG_BUFFER_SIZE=4096
```

cvinayak, 2025-01-10:
@dev-pac The conf file for hci_ipc for ISO features is `nrf5340_cpunet_iso-bt_ll_sw_split.conf`, by default boot banner and assertions should be output... do you see the boot banner?

cvinayak, 2025-01-10:
@dev-pac  are you building for nrf5340dk and flashing to nrf5340_audio_dk?

> west build -b nrf5340dk/nrf5340/cpuapp --sysbuild samples/bluetooth/bap_unicast_server/

dev-pac, 2025-01-10:
> @dev-pac are you building for nrf5340dk and flashing to nrf5340_audio_dk?
> 
> > west build -b nrf5340dk/nrf5340/cpuapp --sysbuild samples/bluetooth/bap_unicast_server/

I don't understand your question , west build is  `-b nrf5340dk/nrf5340/cpuapp`so it's for nrf5340dk and yes board is nrf5340_audio_dk ( specify in title issue  and environment description )

cvinayak, 2025-01-10:
Do you have problem building for audio dk and flashing on audio dk?

```
west build -b nrf5340_audio_dk/nrf5340/cpuapp --sysbuild samples/bluetooth/bap_unicast_server/
```

dev-pac, 2025-01-10:
> Do you have problem building for audio dk and flashing on audio dk?
> 
> ```
> west build -b nrf5340_audio_dk/nrf5340/cpuapp --sysbuild samples/bluetooth/bap_unicast_server/
> ```

No problem with the build or with the flash of the sample.

cvinayak, 2025-01-10:
Please update the steps to reproduce in the description, as it is misleading.

dev-pac, 2025-01-10:
> Please update the steps to reproduce in the description, as it is misleading.

I don't understand, which steps are missing for you? It's just the build and flash example of zephyr on the nrf5340_audio_dk board ?

dev-pac, 2025-01-10:
my bad, I just noticed that it was nrf5340_audio_dk in your comment  and nrf5340dk in my build command.

dev-pac, 2025-01-10:
Logs with correct board :

```
*** Booting Zephyr OS build v3.7.0 ***
[00:00:00.284,362] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.284,393] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.284,423] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.7 Build 0
[00:00:00.286,193] <inf> bt_hci_core: Identity: E8:19:DF:C5:36:52 (random)
[00:00:00.286,224] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.286,254] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
Location successfully set
Supported contexts successfully set
Available contexts successfully set
Advertising successfully started
Connected: 40:CD:7E:7B:5A:EB (random)
ASE Codec Config: conn 0x200022e8 ep 0x200029ac dir 1
codec_cfg 0x06 cid 0x0000 vid 0x0000 count 19
data: type 0x01 value_len 1
08
data: type 0x02 value_len 1
01
data: type 0x03 value_len 4
00000000
data: type 0x04 value_len 2
7800
data: type 0x05 value_len 1
01
  Frequency: 48000 Hz
  Frame Duration: 10000 us
  Channel allocation: 0x0
  Octets per frame: 120 (negative means value not pressent)
  Frames per SDU: 1
ASE Codec Config stream 0x20005498
QoS: stream 0x20005498 qos 0x20007988
QoS: interval 10000 framing 0x00 phy 0x02 sdu 120 rtn 2 latency 10 pd 40000
Enable: stream 0x20005498 meta_len 7
RX stream 0x20005498 len 0
RX stream 0x20005498 len 0
Release: stream 0x20005498
Audio Stream 0x20005498 stopped with reason 0x13
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:410
        Controller unresponsive, command opcode 0x0406 timeout with err -11
[00:00:37.642,791] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:00:37.642,822] <err> os: r3/a4:  0x00000003 r12/ip:  0x00000010 r14/lr:  0x00005db3
[00:00:37.642,822] <err> os:  xpsr:  0x01000000
[00:00:37.642,822] <err> os: s[ 0]:  0x20008cd0  s[ 1]:  0x00000003  s[ 2]:  0x20008c80  s[ 3]:  0x20008c80
[00:00:37.642,852] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000001  s[ 6]:  0x20008c90  s[ 7]:  0x20008c90
[00:00:37.642,852] <err> os: s[ 8]:  0x00000006  s[ 9]:  0x20002608  s[10]:  0x20005498  s[11]:  0x20005cb0
[00:00:37.642,883] <err> os: s[12]:  0x20008cd0  s[13]:  0xffffffff  s[14]:  0xffffffff  s[15]:  0x00000000
[00:00:37.642,883] <err> os: fpscr:  0x00000000
[00:00:37.642,883] <err> os: Faulting instruction address (r15/pc): 0x00005dc2
[00:00:37.642,913] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:37.642,944] <err> os: Current thread: 0x20002f28 (unknown)
[00:00:38.161,285] <err> os: Halting system
```

cvinayak, 2025-01-10:
> @dev-pac The conf file for hci_ipc for ISO features is `nrf5340_cpunet_iso-bt_ll_sw_split.conf`, by default boot banner and assertions should be output... do you see the boot banner?

did you check this?

dev-pac, 2025-01-10:
For hci_ipc build i have :
`Merged configuration '/workspaces/bolt-simple-app/zephyr/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf'`

What i should add to get logs of cpunet, i'm using the nrfconnect vscode extension.
With the extension RTT from cpuapp are displayed but nothing on cpunet RTT

cvinayak, 2025-01-10:
By default it has `CONFIG_UART_CONSOLE=y`, `CONFIG_BOOT_BANNER=y`, `CONFIG_BT_ASSERT=y` and `CONFIG_BT_ASSERT_VERBOSE=y`, that should give the boot banner and assertion failure output in the UART. Use your favorite terminal application (i am not using VSCode now, you can use one in VSCode?).

dev-pac, 2025-01-10:
logs from ttyACM0 :
```
---- Opened the serial port /dev/ttyACM0 ----
*** Booting Zephyr OS build v3.7.0 ***
[00:00:00.285,278] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.285,308] <inf> bt_hci_core: HW Variant: nRF53x (0x0003)
[00:00:00.285,339] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.7 Build 0
[00:00:00.287,780] <inf> bt_hci_core: Identity: E8:19:DF:C5:36:52 (random)
[00:00:00.287,811] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.287,841] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
Location successfully set
Supported contexts successfully set
Available contexts successfully set
Advertising successfully started
Connected: 4C:70:A8:DF:65:C1 (random)
[00:01:01.076,721] <wrn> bt_conn: conn 0x200022e8 failed to establish. RF noise?
Disconnected: 4C:70:A8:DF:65:C1 (random) (reason 0x3e)
Advertising successfully started
Connected: 4C:70:A8:DF:65:C1 (random)
ASE Codec Config: conn 0x200022e8 ep 0x200029ac dir 1
codec_cfg 0x06 cid 0x0000 vid 0x0000 count 19
data: type 0x01 value_len 1
08
data: type 0x02 value_len 1
01
data: type 0x03 value_len 4
00000000
data: type 0x04 value_len 2
7800
data: type 0x05 value_len 1
01
  Frequency: 48000 Hz
  Frame Duration: 10000 us
  Channel allocation: 0x0
  Octets per frame: 120 (negative means value not pressent)
  Frames per SDU: 1
ASE Codec Config stream 0x20005498
QoS: stream 0x20005498 qos 0x20007988
QoS: interval 10000 framing 0x00 phy 0x02 sdu 120 rtn 2 latency 10 pd 40000
Enable: stream 0x20005498 meta_len 7
```
logs from ttyACMC1
```
---- Opened the serial port /dev/ttyACM1 ----
*** Booting Zephyr OS build v3.7.0 ***
[00:00:00.005,737] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:00.006,378] <inf> hci_ipc: Received message of 4 bytes.
--- 31 messages dropped ---
[00:00:00.019,714] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.019,927] <inf> hci_ipc: Sent message of 15 bytes.
[00:00:00.020,141] <inf> hci_ipc: Received message of 8 bytes.
[00:00:00.020,355] <inf> hci_ipc: Sent message of 7 bytes.
[00:00:00.020,568] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.020,782] <inf> hci_ipc: Sent message of 8 bytes.
[00:00:00.020,996] <inf> hci_ipc: Received message of 12 bytes.
[00:00:00.021,209] <inf> hci_ipc: Sent message of 7 bytes.
[00:00:00.021,423] <inf> hci_ipc: Received message of 12 bytes.
[00:00:00.021,636] <inf> hci_ipc: Sent message of 7 bytes.
[00:00:00.021,850] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.022,064] <inf> hci_ipc: Sent message of 19 bytes.
[00:00:00.022,369] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.022,613] <inf> hci_ipc: Sent message of 71 bytes.
[00:00:00.022,827] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.023,040] <inf> hci_ipc: Sent message of 15 bytes.
[00:00:00.023,254] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.023,468] <inf> hci_ipc: Sent message of 13 bytes.
[00:00:00.023,681] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.023,895] <inf> hci_ipc: Sent message of 30 bytes.
[00:00:00.024,139] <inf> hci_ipc: Received message of 10 bytes.
[... 309 lines cut ...]
[00:01:12.356,475] <inf> hci_ipc: Received message of 29 bytes.
[00:01:12.357,055] <inf> hci_ipc: Sent message [00:01:12.357,543] <inf> hci_ipc: Sent message of 8 bytes.
[00:01:12.385,528] <inf> hci_ipc: Sent message of 23 bytes.
[00:01:12.386,718] <inf> hci_ipc: Received message of 9 bytes.
[00:01:12.387,023] <inf> hci_ipc: Received message of 17 bytes.
[00:01:12.387,237] <inf> hci_ipc: Received message of 24 bytes.
[00:01:12.387,817] <inf> hci_ipc: Sent message of 8 bytes.
[00:01:12.388,275] <inf> hci_ipc: Sent message of 8 bytes.
[00:01:12.394,012] <inf> hci_ipc: Sent message of 10 bytes.
[00:01:12.394,256] <inf> hci_ipc: Received message of 6 bytes.
[00:01:12.394,531] <inf> hci_ipc: Sent message of 7 bytes.
[00:01:12.428,070] <inf> hci_ipc: Sent message of 32 bytes.
[00:01:12.428,283] <inf> hci_ipc: Received message of 17 bytes.
[00:01:12.428,527] <inf> hci_ipc: Sent message of 9 bytes.
[00:01:12.428,833] <inf> hci_ipc: Received message of 24 bytes.
[00:01:12.445,922] <inf> hci_ipc: Sent message of 8 bytes.
ASSERTION FAIL [net_buf_simple_headroom(buf) >= len] @ WEST_TOPDIR/zephyr/subsys/net/buf_simple.c:341
[00:01:12.447,967] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000000  r2/a3:  0x00000010
[00:01:12.447,967] <err> os: r3/a4:  0x00000004 r12/ip:  0x2100b940 r14/lr:  0x01026f85
[00:01:12.447,998] <err> os:  xpsr:  0x29000000
[00:01:12.447,998] <err> os: Faulting instruction address (r15/pc): 0x01036956
[00:01:12.448,059] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:01:12.448,089] <err> os: Current thread: 0x21000e80 (unknown)
[00:01:12.595,428] <err> os: Halting system
```

cvinayak, 2025-01-10:
please cherry-pick e8c5405609e355cf9483debffbd09844e4c92900 and give a try?

`git cherry-pick e8c5405609e355cf9483debffbd09844e4c92900`

cvinayak, 2025-01-10:
Ok, it is bug in the Controller implementation exposed due to a possible race during termination of ACL and CIG. This github issue will need some planning effort to submit a fix.

dev-pac, 2025-01-10:
Ok thanks

closed 2025-03-26 by github-actions[bot]