zephyrproject-rtos/zephyr #83806: drivers: usb: device: raspberrypi_pico_usbd - use of uninitialized semaphore

FlorianWeber1018, 2025-01-10:
**Describe the bug**

In some cases the write semaphore is used before it was initialized. This leads to a fatal error.
In my case this happens just after
irq_enable(USB_IRQ); (https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/device/usb_dc_rpi_pico.c#L1076)

**To Reproduce**

configure a cdc-acm-uart on the usb device and restart the device (via gdb) till it just happens during boot.  

**Expected behavior**

nothing should use a semaphore before it was initialized. 

**Environment (please complete the following information):**

 - OS: Linux in docker container on macOS host

**Additional context**

I use the usb interface for a cdc-acm-uart on a raspberryPi pico (rp2040). With the following configured options:
CONFIG_DEBUG=y
CONFIG_DEBUG_OPTIMIZATIONS=y
CONFIG_NO_OPTIMIZATIONS=y
CONFIG_LOG_SPEED=y
and it disappears or seems to happen less frequently if the compiler optimization is enabled.

kartben, 2025-01-14:
@FlorianWeber1018 would you be able to send a PR, if you've identified where the missing initialization should happen, maybe?

FlorianWeber1018, 2025-01-15:
@kartben  No i am not that used to USB stuff. Thats why i have submitted this issue, hopefully that someone with a more detailed knowledge is looking on that.

jfischer-no, 2025-02-20:
https://github.com/zephyrproject-rtos/zephyr/pull/86102

> **To Reproduce**
> 
> configure a cdc-acm-uart on the usb device and restart the device (via gdb) till it just happens during boot.

I am not sure how to reproduce it. At least the backtrace would be very helpful. I see the irq_enable() is called too early, can you please try https://github.com/zephyrproject-rtos/zephyr/pull/86102 if it fixes your problem.

FlorianWeber1018, 2025-02-21:
@jfischer-no i have tested https://github.com/zephyrproject-rtos/zephyr/pull/86102. It seems to fix this issue. 
Test procedure:
With the PR have rebooted 150 times without running in the fatal error in the backtrace above. 
Without the PR the error happens after 29 reboots 2 times.

I think there is a high chance that it is fixed now.

PR #86102 (2025-02-21): drivers: usb_dc_rpi_pico: do not enable interrupt in driver init
Move irq_enable() to the place where the controller is initialized.

Fixes: #83806
closed 2025-02-21 by fabiobaltieri