zephyrproject-rtos/zephyr #83813: CI twister tests: target has not sufficient memory

lfelten, 2025-01-10:
**Describe the bug**
When twister runs automated tests, some (at least one) tests result in binaries that need more memory than the target provides. This will then cause the test to fail.

**To Reproduce**
`west build -p -b m5stack_core2/esp32/appcpu tests/subsys/secure_storage/psa/crypto -T secure_storage.psa.crypto.secure_storage`

**Error:**
```
 /opt/toolchains/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section `.flash.text' will not fit in region `iram0_1_seg'
/opt/toolchains/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: IRAM0 segment data does not fit.
/opt/toolchains/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: region `iram0_1_seg' overflowed by 10072 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Expected behavior**
The test should build and pass.

**Impact**
Blocks PR #82474 

**Logs and console output**

Full output is visible in this CI run:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/12602562689/job/35126072161?pr=82474

**Environment (please complete the following information):**
 - OS: Linux and CI
 - Toolchain Zephyr SDK
 - Commit SHA: current main will fail

**Additional context**
@valeriosetti did the following analysis:

Since I was not able to understand why the changes I made in my PR (https://github.com/zephyrproject-rtos/zephyr/pull/82859) were causing this issue I used git bisect to find the problematic commit. It turns out that it's this one: https://github.com/zephyrproject-rtos/zephyr/commit/322ab2a86e9ca86a1a2380be81218ba4574c91d4 (prior to that commit the sample you linked builds, after it does not).

For the records, the changes I introduced in 82859 reduce the overflow amount from 3296 bytes to 1288 (I tried reverting all the commits manually). From the build failure point of view I understand that this is not interesting, but IMO it proves that 82859 is going in the right direction to reduce RAM/ROM footprint.

sylvioalves, 2025-01-10:
@lfelten CI should not fail when ram overflows. The issue lies in the way CI detected overflow and it is current missing the regex for `iram` section. I submitted an update to overcome this: #83822

lfelten, 2025-01-14:
@sylvioalves  Thanks for fixing this quickly!

closed 2025-01-14 by lfelten