zephyrproject-rtos/zephyr #83823: Code coverage issue on Windows when gcov_tool specified to twister at the command line - Object of type WindowsPath is not JSON serializable

solsbarry, 2025-01-10:
**Describe the bug**
I have a on target unit test that I am executing with twister and asking it to generate a code coverage report for. The build failed with this error:
[image]

Here is a copy of the report being passed to json.dump
[image]

The issue is that I am specifying a gcov_tool at the command line call to twister
`twister....--coverage --gcov-tool my_windows_path...`
And as you can see somehow twister is turning the gcov_tool path into a WindowsPath variable and not a string. And apparently a Windows Path type can't be in a dictionary passed to json.dump. That is the issue. 

In reports.py I updated jason_report() to convert WindowsPath types in the report to strings, and that caused it to build fine. I am not suggesting that is the best answer, but something like it.
[image]
You also have to add this at the top to get it to build
`from pathlib import WindowsPath`
Here is the text of that code if you want to try it.
```
def convert_windows_paths_to_strings(self, d):
      if isinstance(d, dict):
          return {k: self.convert_windows_paths_to_strings(v) for k, v in d.items()}
      elif isinstance(d, list):
          return [self.convert_windows_paths_to_strings(i) for i in d]
      elif isinstance(d, WindowsPath):
          return str(d)
      else:
          return d
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a unit test
2. Run it on Windows with twister, while specifying the path to gcov_tool in the twister command and enabling coverage. 
`twister....--coverage --gcov-tool my_windows_path...`
3. See error

**Expected behavior**
The unit test should build without error

**Impact**
Can't include coverage in my on target unit tests when running on windows.

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA or Version used: 3.7.88-NCS1 ( I know that is a nordic version, but I think this issue is specific to Zephyr more generally.

nashif, 2025-01-20:
@golowanow  here https://github.com/zephyrproject-rtos/zephyr/blob/ebbdb1068d3fa6bfa2aeb78adf384c330994e256/scripts/pylib/twister/twisterlib/reports.py#L297 you already deal with PosixPath, need to make this also work with `WindowsPath`.

golowanow, 2025-01-21:
@solsbarry, could you please check #84334 at your environment ?

solsbarry, 2025-01-21:
@golowanow That fixed the issue. Thank you very much, I appreciate the quick fix.

PR #84334 (2025-01-23): twister: reporting: fix gcov_tool path on Windows
Refactor Reporting to use custom ReportingJSONEncoder and encode all pathlib.Path derived instances there which are possible e.g. in 'environment.options' received from command line.

Fixes: #83823
closed 2025-01-23 by kartben