zephyrproject-rtos/zephyr #83863: LLVM build fails with ASAN for native_sim and sockets/echo

riesentoaster, 2025-01-12:
**Describe the bug**
When using the LLVM toolchain, targeting `native_sim` and enabling ASAN, the example sockets/echo breaks. Building works fine, but the binary crashes during startup.

Possibly related to #71288

**To Reproduce**
`zephyr/samples/net/sockets/echo/prj.conf`: append `CONFIG_ASAN=y`

Then, build using the following command 

```bash
ZEPHYR_TOOLCHAIN_VARIANT=llvm west build -b native_sim samples/net/sockets/echo --pristine
```

**Expected behavior**
The binary shows the same behaviour as without ASAN.

**Impact**
The binary crashes during startup/initialization.

**Logs and console output**

<details><summary>Build logs</summary>

```
-- west build: making build dir /home/ubuntu/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ubuntu/zephyrproject/zephyr/samples/net/sockets/echo
-- CMake version: 3.30.2
-- Found Python3: /home/ubuntu/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/ubuntu/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_sim, qualifiers: native
-- Found toolchain: llvm (clang/ld)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ubuntu/zephyrproject/zephyr/boards/native/native_sim/native_sim.dts
-- Generated zephyr.dts: /home/ubuntu/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/ubuntu/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/ubuntu/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ubuntu/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/ubuntu/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/ubuntu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ubuntu/zephyrproject/zephyr/boards/native/native_sim/native_sim_defconfig'
Merged configuration '/home/ubuntu/zephyrproject/zephyr/samples/net/sockets/echo/prj.conf'
Configuration saved to '/home/ubuntu/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ubuntu/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.38")
-- The C compiler identification is Clang 16.0.6
-- The CXX compiler identification is Clang 16.0.6
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /usr/bin/clang
-- Using ccache: /usr/bin/ccache
-- Looking for xtensa/config/core-isa.h
-- Looking for xtensa/config/core-isa.h - not found
-- Configuring done (22.4s)
-- Generating done (0.3s)
-- Build files have been written to: /home/ubuntu/zephyrproject/zephyr/build
-- west build: building application
[1/1119] Preparing syscall dependency handling

[2/1119] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.99 (/home/ubuntu/zephyrproject/zephyr), build: v4.0.0-3188-gd4deeebff187
[1117/1119] Linking C executable zephyr/zephyr.elf
Generating files from /home/ubuntu/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: native_sim
[1119/1119] Running utility command for native_runner_executable
```

</details>
[... 82 lines cut ...]
-- Build files have been written to: /home/ubuntu/zephyrproject/zephyr/build
-- west build: building application
[50/1119] Preparing syscall dependency handling

[159/1119] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.0.99 (/home/ubuntu/zephyrproject/zephyr), build: v4.0.0-3188-gd4deeebff187
[1117/1119] Linking C executable zephyr/zephyr.elf
Generating files from /home/ubuntu/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: native_sim
[1119/1119] Running utility command for native_runner_executable
```

</details>

<details><summary>Execution logs</summary>

```
WARNING: Using a test - not safe - entropy source
*** Booting Zephyr OS build v4.0.0-3188-gd4deeebff187 ***
IPV6_V6ONLY option is on, turning it off.
Sharing same socket between IPv6 and IPv4
error: bind: 125
exit
```

</details>

riesentoaster, 2025-01-12:
This works with the default toolchain (`zephyr-sdk-0.17.0`) btw.

@aescolar you worked on something similar in #71288, maybe you can take a look?

riesentoaster, 2025-01-12:
For clang 16, I noticed that for some reason registration(?) of `iface`s seems to break: `STRUCT_SECTION_FOREACH` here is invoked twice with ASAN and only once without.

https://github.com/zephyrproject-rtos/zephyr/blob/99a63a776980f855493ceb99036a5974a2b68fb5/subsys/net/ip/net_if.c#L6149-L6165

Without ASAN, `iface` is at `0x8079d58` on my machine. With ASAN, it is invoked with `0x822c5c0` and `0x822c604`.

aescolar, 2025-01-13:
CC @jukkar , @rlubos 
@riesentoaster in your log ASAN  is reporting a segmentation fault there, which would be an indication of an actual issue.
How are you running the zephyr executable?/what parameters are you passing to it?
(This seems unrelated to the other bug you linked)

riesentoaster, 2025-01-13:
I'm not passing any parameters, just calling the executable (`build/zephyr/zephyr.exe`). And without ASAN this works just fine, so the ASAN pass of LLVM seems to change something. If I understand correctly, the segfault happens because for some reason, Zephyr seems to think two interfaces are registered when using ASAN. The stack trace is something like:

1. `net_if_init` in `net_if.c`, specifically in the second evaluation of `STRUCT_SECTION_FOREACH` (see comment above)
2. `init_iface` in `net_if.c`, the first line of which calls
3. `net_if_get_device` in `net_if.h`, specifically in the last line that says `return iface->if_dev->dev;`, where `if_dev` is `NULL`

I have no idea how either the internals of ASAN work nor the internals of how Zephyr interface registration works. So it may be memory layout issue, I don't know.

jukkar, 2025-01-13:
This does not look like a networking issue as it works ok if ASAN is not enabled.

jukkar, 2025-01-13:
Trying this too
```
$ ZEPHYR_TOOLCHAIN_VARIANT=llvm west build -p -b native_sim samples/net/sockets/echo -d ../build/echo -- -DCONFIG_ASAN=y
$ /.../zephyrproject/build/echo/zephyr/zephyr.exe 
WARNING: Using a test - not safe - entropy source
AddressSanitizer:DEADLYSIGNAL
=================================================================
==47835==ERROR: AddressSanitizer: SEGV on unknown address 0x00000000 (pc 0x081427f0 bp 0xe6bff0d8 sp 0xe6bff020 T2)
==47835==The signal is caused by a READ memory access.
==47835==Hint: address points to the zero page.
    #0 0x81427f0 in net_if_get_device /.../zephyrproject/zephyr/include/zephyr/net/net_if.h:970:24
    #1 0x81427f0 in init_iface /.../zephyrproject/zephyr/subsys/net/ip/net_if.c:418:33
    #2 0x81427f0 in net_if_init /.../zephyrproject/zephyr/subsys/net/ip/net_if.c:6165:3
    #3 0x8136bcb in init_rx_queues /.../zephyrproject/zephyr/subsys/net/ip/net_core.c:585:2
    #4 0x8136bcb in net_init /.../zephyrproject/zephyr/subsys/net/ip/net_core.c:641:2

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /.../zephyrproject/zephyr/include/zephyr/net/net_if.h:970:24 in net_if_get_device
Thread T2 created by T1 here:
    #0 0x80cda0c in pthread_create (/.../zephyrproject/build/echo/zephyr/zephyr.exe+0x80cda0c) (BuildId: 6715b3f0c292b40e34b122da705dd831095c5288)
    #1 0x8191973 in nct_new_thread /.../zephyrproject/zephyr/scripts/native_simulator/common/src/nct.c:476:2
    #2 0x811cf3a in posix_new_thread /.../zephyrproject/zephyr/arch/posix/core/posix_core_nsi.c:48:9

Thread T1 created by T0 here:
    #0 0x80cda0c in pthread_create (/.../zephyrproject/build/echo/zephyr/zephyr.exe+0x80cda0c) (BuildId: 6715b3f0c292b40e34b122da705dd831095c5288)
    #1 0x8190ae7 in nce_boot_cpu /.../zephyrproject/zephyr/scripts/native_simulator/common/src/nce.c:209:2
    #2 0x812158b in posix_boot_cpu /.../zephyrproject/zephyr/soc/native/inf_clock/soc.c:126:2
    #3 0x8192d43 in nsi_cpu_auto_boot /.../zephyrproject/zephyr/scripts/native_simulator/common/src/nsi_cpu_ctrl.c:38:4
    #4 0x81904d3 in nsi_init /.../zephyrproject/zephyr/scripts/native_simulator/common/src/main.c:87:2
    #5 0x81905a4 in main /.../zephyrproject/zephyr/scripts/native_simulator/common/src/main.c:119:2
    #6 0xea064518 in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16

==47835==ABORTING
```
Tried with gcc and there is no error.

The echo application expects only one network interface which is defined in a linker section. If the linker gives it two, then that is an error.
Who can I re-assign this, any idea?

aescolar, 2025-01-13:
@jukkar I look into it

aescolar, 2025-01-13:
@jukkar this is the bug(fix):
```diff
diff --git a/include/zephyr/net/net_if.h b/include/zephyr/net/net_if.h
index 14981e9200d..aa6a403aa29 100644
--- a/include/zephyr/net/net_if.h
+++ b/include/zephyr/net/net_if.h
@@ -3270,7 +3270,7 @@ extern int net_stats_prometheus_scrape(struct prometheus_collector *collector,
        static Z_DECL_ALIGN(struct net_if)                              \
                       NET_IF_GET_NAME(dev_id, sfx)[_num_configs]       \
                       __used __in_section(_net_if, static,             \
-                                          dev_id) = {                  \
+                                          dev_id) __noasan = {                 \
                [0 ... (_num_configs - 1)] = {                          \
                        .if_dev = &(NET_IF_DEV_GET_NAME(dev_id, sfx)),  \
                        NET_IF_CONFIG_INIT                              \
```

The address sanitizer in llvm adds padding (redzones) after data.
But for those we are re-grouping using the linker script and for which we iterate over we cannot have that extra padding.

ASAN:

0822e7a0 d _net_if_list_start <--
0822e7a0 d __net_if_eth_native_posix_0_0 <--
0822e820 d _net_if_list_end **<-- 0x80**

0822e820 d _net_if_dev_list_start
0822e820 d __net_if_dev_eth_native_posix_0_0 <--
0822e83c d _net_if_dev_list_end     <-- 0x1C

_net_if_list_start[]; _net_if_list_end

no asan:
0806e9f0 d _net_if_list_start <--
0806e9f0 d __net_if_eth_native_posix_0_0 <--
0806ea34 d _net_if_list_end **<-- 0x44**

0806ea34 d _net_if_dev_list_start 
0806ea34 d __net_if_dev_eth_native_posix_0_0
0806ea50 d _net_if_dev_list_end  <-- 0x1C

kartben, 2025-01-14:
triaging as low priority, plus it has a pending PR fixing this

PR #83903 (2025-01-15): net: if: Fix net_if_list iteration issue with llvm & ASAN
The address sanitizer in llvm adds padding (redzones) after data. But for those structures we are re-grouping using the linker script and for which we iterate over we cannot have that extra padding.

Fixes #83863
PR #93884 (2025-08-01): net: if: Fix net_if_list iteration issue for offloaded interfaces
Disable ASAN for net_if config to avoid iteration issues caused by AddressSanitizer padding.

This was previously fixed in #83903 for `NET_IF_INIT` but not for `NET_IF_OFFLOAD_INIT`.

Closes #83863
closed 2025-08-01 by fabiobaltieri